Koenkk/zigbee2mqtt #26738: Some zigbee devices stoped working after update homeassistant

JacekJI, 2025-03-14:
### What happened?

hello,

Since 2 days, propably after HA upate. Some devices stoped working.
They are online but nothig happen after clik on/ off on ha.
On z2m addon I see error and logs like on the example.

Reboting Ha, addon, power nothing changed.

Unfortunately I don't have exact backup.

### Zigbee2MQTT version

2.1.3-1

### Adapter firmware version

/dev/serial/by-id/usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20220718154725-if00

### Adapter

Sonof Dongle E 

### Setup

addon on home assistan os

### Debug log

 2025-03-14 11:30:32z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Gniazdko zigbee wolne', payload '{"child_lock":null,"current":0,"energy":1,"indicator_mode":null,"linkquality":144,"power":0,"power_outage_memory":null,"state":"OFF","update":{"installed_version":-1,"latest_version":-1,"state":null},"voltage":236}'
info 2025-03-14 11:30:32z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Licznik dom', payload '{"current_X":2.133,"current_Y":2.15,"current_Z":2.582,"energy":9897.33,"linkquality":172,"power_X":462,"power_Y":469,"power_Z":462,"produced_energy":8375.97,"voltage_X":234.7,"voltage_Y":238.8,"voltage_Z":234.5}'
info 2025-03-14 11:30:33z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Licznik dom', payload '{"current_X":2.133,"current_Y":2.15,"current_Z":2.584,"energy":9897.33,"linkquality":172,"power_X":462,"power_Y":469,"power_Z":459,"produced_energy":8375.97,"voltage_X":234.7,"voltage_Y":238.8,"voltage_Z":236.3}'
info 2025-03-14 11:30:33z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Gniazdko taras', payload '{"child_lock":"UNLOCK","current":0,"energy":0,"indicator_mode":null,"linkquality":148,"power":0,"power_outage_memory":null,"state":"OFF","update":{"installed_version":-1,"latest_version":-1,"state":null},"voltage":238}'
info 2025-03-14 11:30:35z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Gniazdko suszarka ', payload '{"child_lock":"UNLOCK","current":0.01,"energy":138.09,"indicator_mode":null,"linkquality":124,"power":0,"power_outage_memory":null,"state":"ON","update":{"installed_version":-1,"latest_version":-1,"state":null},"voltage":234}'
info 2025-03-14 11:30:38z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Licznik dom', payload '{"current_X":2.133,"current_Y":2.15,"current_Z":2.584,"energy":9897.33,"linkquality":172,"power_X":462,"power_Y":469,"power_Z":459,"produced_energy":8375.97,"voltage_X":234.7,"voltage_Y":238.8,"voltage_Z":236.3}'
info 2025-03-14 11:30:38z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Licznik dom', payload '{"current_X":2.129,"current_Y":2.15,"current_Z":2.584,"energy":9897.33,"linkquality":172,"power_X":468,"power_Y":469,"power_Z":459,"produced_energy":8375.97,"voltage_X":235.3,"voltage_Y":238.8,"voltage_Z":236.3}'
info 2025-03-14 11:30:38z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Licznik dom', payload '{"current_X":2.129,"current_Y":2.164,"current_Z":2.584,"energy":9897.33,"linkquality":172,"power_X":468,"power_Y":468,"power_Z":459,"produced_energy":8375.97,"voltage_X":235.3,"voltage_Y":238.2,"voltage_Z":236.3}'
info 2025-03-14 11:30:39z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Licznik dom', payload '{"current_X":2.129,"current_Y":2.164,"current_Z":2.617,"energy":9897.33,"linkquality":172,"power_X":468,"power_Y":468,"power_Z":469,"produced_energy":8375.97,"voltage_X":235.3,"voltage_Y":238.2,"voltage_Z":236.3}'
info 2025-03-14 11:30:44z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Licznik dom', payload '{"current_X":2.167,"current_Y":2.164,"current_Z":2.617,"energy":9897.33,"linkquality":172,"power_X":475,"power_Y":468,"power_Z":469,"produced_energy":8375.97,"voltage_X":234.6,"voltage_Y":238.2,"voltage_Z":236.3}'
info 2025-03-14 11:30:44z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Licznik dom', payload '{"current_X":2.167,"current_Y":2.172,"current_Z":2.617,"energy":9897.33,"linkquality":172,"power_X":475,"power_Y":470,"power_Z":469,"produced_energy":8375.97,"voltage_X":234.6,"voltage_Y":238.3,"voltage_Z":236.3}'
info 2025-03-14 11:30:45z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Licznik dom', payload '{"current_X":2.167,"current_Y":2.172,"current_Z":2.633,"energy":9897.33,"linkquality":172,"power_X":475,"power_Y":470,"power_Z":470,"produced_energy":8375.97,"voltage_X":234.6,"voltage_Y":238.3,"voltage_Z":236.8}'
info 2025-03-14 11:30:45z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Gniazdko_zelazko', payload

lukasgierth, 2025-03-14:
Can also report the same! For me, they work again after some time but then they again don't for unknown reasons.

Tee86, 2025-03-14:
did you try to power-cycle them? disconnect power ==> wait about 30s ==> reconnect

JacekJI, 2025-03-14:
> Can also report the same! For me, they work again after some time but then they again don't for unknown reasons.

Exactly the same, for example after reboot adon working a while and then stop.

Switch off devices also work for few times.
But some devices working fine.

lukasgierth, 2025-03-14:
Similar problem here: https://github.com/Koenkk/zigbee2mqtt/issues/26738

chris-1243, 2025-03-14:
@lukasgierth 

> MAC channel access failure' (225))'

Your wireless spectrum might be too occupied

https://www.zigbee2mqtt.io/guide/faq/#common-error-codes

Check wifi and zigbee channel

Nemor213, 2025-03-14:
Same here. Seems to be an 2025.3 thing

chris-1243, 2025-03-14:
@lukasgierth 

> Changing the channel is not really an option, since Europe only uses channel 11.

I do leave in Europe and I may choose the channel I wish to use between 11 to 26 (25 in my case...)
Usually, WIFI are on channel 1,6 or 11.
ZIGBEE channel 15 is between Wifi 1-6, 20 between 6-11 and 25 after 11

https://www.metageek.com/training/resources/zigbee-wifi-coexistence/

lukasgierth, 2025-03-14:
> [@lukasgierth](https://github.com/lukasgierth)
> 
> > Changing the channel is not really an option, since Europe only uses channel 11.
> 
> I do live in Europe and I may choose the channel I wish to use between 11 to 26 (25 in my case...) Usually, WIFI are on channel 1,6 or 11. ZIGBEE channel 15 is between Wifi 1-6, 20 between 6-11 and 25 after 11
> 
> https://www.metageek.com/training/resources/zigbee-wifi-coexistence/

You're not wrong, but me changing the channel wouldn't change anything. I live in a densely populated area, the house where my apartment is in has so many wifi traffic on every channel.
Will not make a difference.

And was working fine until the last couple of days. Without any additional wifi access point appearing.

chris-1243, 2025-03-14:
This is good practices

What computer do you use? I have seen people in trouble due to i/o on a sd-card

JacekJI, 2025-03-14:
It’s interesting topic, because in my case, one week ego I added in my garden extender WiFi… 
But I’m sure after that few days was ok.

Where I can try change this Channel?

Tee86, 2025-03-15:
> Changing the channel is not really an option, since Europe only uses channel 11.

You probably misread something online. Europe doesn't use any WiFi channels, your router does, and it doesn't care whether it's located in Europe or North Pole.
Improper channel assignment can cause issues you're experiencing.
The best 2.4Ghz channels are 1,6, and 11.
Here are some strong recommendations:
- In case you're also running a Zigbee network, then I recommend use any Wifi scanner app and look at channels 1 and 6, choose which one is less crowded. Then assign that channel to your 2.4Ghz band via router's interface, if extender allows it, then there too.
- Once done, assign channel 25 to your Zigbee network in z2m configuration.yaml. The best combo is Wifi ch1 & Zigbee ch 25, but 6 &25 will be fine too.
- You did not mention what firmware is your Sonoff Dongle on... you should also have a look at this.
- Make sure your coordinator is ideally 1m away (or more) from your router, or any other devices that use 2.4Ghz band (eg Bluetooth)
- Make sure your coordinator is plugged in USB 2 port
- Make sure you're using usb 2 extension cable(ideally shielded), don't plug your dongle directly. Don't clip that cable to any other cables (for cable management)

That's about it

Tee86, 2025-03-15:
> > [@lukasgierth](https://github.com/lukasgierth)

> You're not wrong, but me changing the channel wouldn't change anything. I live in a densely populated area, the house where my apartment is in has so many wifi traffic on every channel. Will not [...]

But you're wrong.
Just one example. Long time ago when I was starting with all this IoT madness, I ran my Zigbee network for months without any issues. Then one day it all went to hell. Spent days troubleshooting. After a week I resolved everything with one stupid USB extension cable.

Forget how things were, focus on what you have now

JacekJI, 2025-03-16:
> You probably misread something online. Europe doesn't use any WiFi channels, your router does, and it doesn't care whether it's located in Europe or North Pole. Improper channel assignment can cause [...]

I didn’t have most from this list before and it was running fine more than one year.

But… today i changed USB port (from 3 to 1) for sonoff dongle.  
Now is working fine, we will see how long.

lukasgierth, 2025-03-16:
> You probably misread something online. Europe doesn't use any WiFi channels, your router does, and it doesn't care whether it's located in Europe or North Pole. Improper channel assignment can cause [...]

You are correct that i misread something regarding the zigbee channels! (confusing something with the additional 868mhz band in Europe, but it's additional not an instead)
Thank you for all the information! Most of this things i already do, so only usb2 with an extension cable and so on. The sonoff dongle firmware is the newest coordinator one (20240710).
What i didn't try yet is the other zigbee channel. Will test this and hopefully most of my devices won't need to be repaired...

On another note: I will definitely replace the usb coordinator with a tcp one which i can place more centrally in my apartment. And then use the usb one with router firmware somewhere where i need more coverage! Was my plan for a long time but never did it.
Will report back in 2-3 days.

Tee86, 2025-03-16:
No prob, you're welcome.

> The sonoff dongle firmware is the newest coordinator one (20240710). 

20240710 is buggy, speaking from very recent experience. 
I downgraded to 20221226 last Friday, so far I have no issues.
There is also a newer firmware [99250304](https://github.com/user-attachments/files/19076647/CC2652R_coordinator_99250304.hex.zip) but I haven't tested it yet.

JacekJI, 2025-03-18:
Hello I found solution.

Finally logs showed error about adapter 

I changes ezp to ember and also updated sonoff dongle.

Since 2 days working fine!

[image]

ericvanbeurden, 2025-03-18:
https://github.com/Koenkk/zigbee2mqtt/issues/26778

I have a similar problem, my complete zigbee network crashes. I traced this back to an HA update. 
WIth a backup of an old version of HA I got it working. 
I will check if my non working image of HA can be fixed with above solution.

closed 2025-05-25 by github-actions[bot]