Koenkk/zigbee2mqtt #26743: configuration for Conbee3 docker solution

nsk, 2025-03-15:
### What happened?

Hi, I'm making this post for those who are struggling with conbee3 and docker, I think it can help, maybe it should be put in the documentation, I searched for myself before finding the solution and by trial and error

configuration.yaml

# Home Assistant integration (MQTT discovery)
homeassistant: true

# allow new devices to join
permit_join: true

# MQTT settings
mqtt:
  # MQTT base topic for zigbee2mqtt MQTT messages
  base_topic: zigbee2mqtt
  # MQTT server URL
  server: 'mqtt://"ip":1883'
  # MQTT server authentication, uncomment if required:
  user: your login
  password: your password

# Serial settings
serial:
  port: /dev/ttyUSB0
  adapter: deconz
  baudrate: 115200

advanced:
log_directory: /var/log
experimental: {}
frontend:
 webGUI
  port: 8089
ota:
  update_check_interval: 10
disable_automatic_update_check: false
script to detect usb port:

sudo vim ports.sh

#!/bin/bash

for sysdevpath in $(find /sys/bus/usb/devices/usb*/ -name dev); do
    (
        syspath="${sysdevpath%/dev}"
        devname="$(udevadm info -q name -p $syspath)"
        [[ "$devname" == "bus/"* ]]
        eval "$(udevadm info -q property --export -p $syspath)"
        [[ -z "$ID_SERIAL" ]]
        echo "/dev/$devname - $ID_SERIAL"
    )
done
sudo chmod 777 ports.sh 
./ports.sh
docker start :

sudo docker run    --name=zigbee2mqtt    --restart=unless-stopped    -p 8089:8089    -v $(pwd)/data:/app/data    -v /run/udev:/run/udev:ro    --device=/dev/serial/by-id/usb-dresden_elektronik_ConBee_III_DE03186404-if00-port0:/dev/ttyUSB0    --user 1001:1001    --group-add dialout    -e TZ=Europe/Zurich    koenkk/zigbee2mqtt

### Zigbee2MQTT version

2.1.3

### Adapter firmware version

0000000000

### Adapter

conbee 3 

### Setup

home assistant

Nerivec, 2025-03-15:
The `configuration.yaml` is not valid, and is not for Z2M 2.x.x.

We have this coming soon:
https://github.com/Koenkk/zigbee2mqtt/pull/26608

closed 2025-03-15 by Nerivec

nsk, 2025-03-15:
I struggled a lot but now it works : 

docker run    --name zigbee2mqtt    --restart=unless-stopped    --device=/dev/serial/by-id/usb-dresden_elektronik_ConBee_III_DE03186404-if00-port0:/dev/ttyUSB0    -p 8080:8080    -v $(pwd)/data:/app/data    -v /run/udev:/run/udev:ro    -e TZ=Europe/Zurich    koenkk/zigbee2mqtt

Using '/app/data' as data directory
Starting Zigbee2MQTT without watchdog.
[2025-03-15 20:30:06] info: 	z2m: Logging to console, file (filename: log.log)
[2025-03-15 20:30:06] info: 	z2m: Starting Zigbee2MQTT version 2.1.3 (commit #ba337bd329aeb4ca17735c0cf09b31293c8cff06)
[2025-03-15 20:30:06] info: 	z2m: Starting zigbee-herdsman (3.2.7)
[2025-03-15 20:30:06] info: 	zh:adapter:discovery: Matched adapter: {"path":"/dev/ttyUSB0","manufacturer":"dresden elektronik","serialNumber":"DE03186404","pnpId":"usb-dresden_elektronik_ConBee_III_DE03186404-if00-port0","vendorId":"0403","productId":"6015"} => deconz: 3
[2025-03-15 20:30:06] info: 	z2m: zigbee-herdsman started (resumed)
[2025-03-15 20:30:06] info: 	z2m: Coordinator firmware version: '{"meta":{"maintrel":0,"majorrel":38,"minorrel":83,"product":0,"revision":"0x26530900","transportrev":0},"type":"ConBee3"}'
[2025-03-15 20:30:06] info: 	z2m: 0xa4c13800ab3b29fa (0xa4c13800ab3b29fa): Not supported (Unknown)
[2025-03-15 20:30:06] info: 	z2m: Currently 1 devices are joined.
[2025-03-15 20:30:06] info: 	z2m: Connecting to MQTT server at mqtt://192.168.1.2:1883
[2025-03-15 20:30:06] info: 	z2m: Connected to MQTT server
[2025-03-15 20:30:06] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"online"}'
[2025-03-15 20:30:07] info: 	z2m: Started frontend on port 192.168.1.2:8888
[2025-03-15 20:30:07] info: 	z2m: Zigbee2MQTT started!
[2025-03-15 20:30:12] info: 	z2m:mqtt: MQTT publish: topic 'homeassistant/binary_sensor/1221051039810110150109113116116_0x00212effff0e1414/connection_state/config', payload '{"device":{"hw_version":"ConBee3 0x26530900","identifiers":["zigbee2mqtt_bridge_0x00212effff0e1414"],"manufacturer":"Zigbee2MQTT","model":"Bridge","name":"Zigbee2MQTT Bridge","sw_version":"2.1.3"},"device_class":"connectivity","entity_category":"diagnostic","name":"Connection state","object_id":"zigbee2mqtt_bridge_connection_state","origin":{"name":"Zigbee2MQTT","sw":"2.1.3","url":"https://www.zigbee2mqtt.io"},"payload_off":"offline","payload_on":"online","state_topic":"zigbee2mqtt/bridge/state","unique_id":"bridge_0x00212effff0e1414_connection_state_zigbee2mqtt","value_template":"{{ value_json.state }}"}'
[2025-03-15 20:31:05] info: 	z2m: Removing device '0xa4c13800ab3b29fa' (block: false, force: true)
[2025-03-15 20:31:05] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xa4c13800ab3b29fa', payload ''
[2025-03-15 20:31:05] info: 	z2m: Successfully removed device '0xa4c13800ab3b29fa' (block: false, force: true)
[2025-03-15 20:31:05] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/remove', payload '{"data":{"block":false,"force":true,"id":"0xa4c13800ab3b29fa"},"status":"ok","transaction":"e4mkw-1"}'
[2025-03-15 20:31:47] info: 	z2m: Successfully changed options