Koenkk/zigbee2mqtt #26748: Ubisys S1-R Series2 (5601) stop responding to certain actions from Zigbee2MQTT after a few days

emagnier, 2025-03-15:
### What happened?

I have several Ubisys S1-R (5601) (Series 2). Once installed, after a few days without issues, they stop responding to certain actions from Zigbee2MQTT. Specifically:
- Manually retrieving or updating the state (On/Off).
- Retrieving or updating the "Power-on behavior" option.
- The "Identify" action.

However, they continue to work normally when controlled via physical buttons.

Some actions still work correctly:
- Physical buttons function properly, and the state updates instantly in Zigbee2MQTT.
- Power and Energy values are also reported correctly. Clicking the refresh button (the blue button with two circular arrows) also works fine.

**Workaround**: Deleting and re-pairing these devices solves the issue, but this is far from ideal. Clicking the orange "Reconfigure" button does not resolve the problem.

### What did you expect to happen?

Being able to modify the state (On/Off) from Zigbee2MQTT, as well as update the "Power-on behavior" and "Identify" actions.

### How to reproduce it (minimal and precise)

Wait a few days (or weeks?), and they eventually stop functioning entirely from Zigbee2MQTT.

### Zigbee2MQTT version

2.1.3-1

### Adapter firmware version

0x26720700

### Adapter

ConBee2

### Setup

Add-on on Home Assistant OS, ODROID-N2/N2+

### Debug log

```
error 2025-03-15 17:31:07z2m: Publish 'get' 'power_on_behavior' to 'Pallier lumière' failed: 'Error: ZCL command 0x001fee000000ac1d/0 genOnOff.read(["startUpOnOff"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (no response received (83) Error: waiting for response TIMEOUT)'
error 2025-03-15 17:31:47z2m: Publish 'get' 'state' to 'Pallier lumière' failed: 'Error: ZCL command 0x001fee000000ac1d/0 genOnOff.read(["onOff"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (no response received (89) Error: waiting for response TIMEOUT)'
error 2025-03-15 17:32:03z2m: Publish 'set' 'state' to 'Pallier lumière' failed: 'Error: ZCL command 0x001fee000000ac1d/0 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (no response received (93) Error: waiting for response TIMEOUT)'
error 2025-03-15 17:32:14z2m: Publish 'set' 'identify' to 'Pallier lumière' failed: 'Error: ZCL command 0x001fee000000ac1d/0 genIdentify.identify({"identifytime":3}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (no response received (95) Error: waiting for response TIMEOUT)'
error 2025-03-15 17:32:26z2m: Publish 'set' 'power_on_behavior' to 'Pallier lumière' failed: 'Error: ZCL command 0x001fee000000ac1d/0 genOnOff.write({"startUpOnOff":2}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (no response received (96) Error: waiting for response TIMEOUT)'
```

emagnier, 2025-05-19:
The issue is still present, FYI.

daNick73, 2025-05-21:
> The issue is still present, FYI.

I have the same set up (Conbee II and S1-R Series 2).

I ordered a different coordinator and will indicate if that solves it for me.

daNick73, 2025-05-26:
I switched to an slzb-06 and my connection to My two S1-R Series 2 has been rockstable for 5 days straight.

So, potentially, this is a Conbee issue (which was moved to not recommended status for the project)

closed 2025-08-02 by github-actions[bot]