Koenkk/zigbee2mqtt #26749: unable to start Z2M

don0cameleon, 2025-03-15:
### What happened?

Hi there,
I'm unnable to start z2m in HA.
I'm using a Smlight slzb-06m in LAN mode. My home assistant is running on a raberrrypi4.
MQTT is installed and running sucessfully
Z2M  yaml is as follows
#
data_path: /config/zigbee2mqtt
socat:
  enabled: false
  master: pty,raw,echo=0,link=/tmp/ttyZ2M,mode=777
  slave: tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5
  options: "-d -d"
  log: false
mqtt: {}
serial:
  port: tcp://192.168.1.26:6638
  baudrate: 115200
  adapter: ezsp
#
tcp listen port is 8485.

Unfortunately I keep obtaining this error in z2m log : 
#
[2025-03-15 19:29:30] error: 	zh:ezsp:uart: Port error: Error: read ECONNRESET
Error: Failure to connect
    at SerialDriver.resetForReconnect (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:341:19)
    at SerialDriver.emit (node:events:524:28)
    at SerialDriver.onPortClose (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/uart.ts:385:18)
    at Object.onceWrapper (node:events:639:26)
    at Socket.emit (node:events:524:28)
    at TCP.<anonymous> (node:net:351:12)
# 

any help please :) thank you very much

### Zigbee2MQTT version

2.1.3-1

### Adapter firmware version

20250220

### Adapter

smlight slzb-06

### Setup

add-on on home assistant / PI4

Tee86, 2025-03-15:
I'd recommend googling "z2m configuration.yaml example"
Yours is too off, you don't even have mqtt configured

Nerivec, 2025-03-15:
_The yaml is from HA add-on config page I presume (different from the plain `configuration.yaml`). The HA startup should automatically detect mqtt config if not present._

Make sure your network is stable, ECONNRESET means the adapter closed its connection.
You may want to also move to `ember` driver instead (replaces `ezsp`):
```yaml
 adapter: ember
```

don0cameleon, 2025-03-16:
@Nerivec @Tee86 hi guys thank you very much for your help, here how I fixed the problem.
1/ re-instal HA using the official Rasberry Pi Imager
2/ instal MQTT Borker and create a user specifically to this purpose
3/ instal Z2M with the correct yaml config.
4/ Flash my smlight with 8.0.2 update.
5/ No need to install ZHA, m2z is engouh.
6/ Put some space between my wifi and the smlight 🗡 

Now every think works properly !

closed 2025-03-16 by don0cameleon