Koenkk/zigbee2mqtt #26750: [New device support]: Philips Hue Signe Floor Lamp but model 929003103601?

comdr-chocchip, 2025-03-15:
### Link

https://www.philips-hue.com/en-au/p/hue-white-and-color-ambiance-signe-gradient-floor-lamp/8719514394230

### Database entry

{"id":28,"type":"Router","ieeeAddr":"0x001788010dfe48b7","nwkAddr":28698,"manufId":4107,"manufName":"Philips","powerSource":"Mains (single phase)","modelId":"929003103601","epList":[11,242],"endpoints":{"11":{"profId":260,"epId":11,"devId":269,"inClusterList":[0,3,4,5,6,8,4096,64515,768,64513,64516],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"929003103601","manufacturerName":"Philips","powerSource":1,"zclVersion":8,"appVersion":2,"stackVersion":1,"hwVersion":0,"dateCode":"20240902","swBuildId":"1.122.2"}},"lightingColorCtrl":{"attributes":{"colorCapabilities":31,"colorTempPhysicalMin":153,"colorTempPhysicalMax":500,"colorMode":1,"currentX":16217,"currentY":34839,"enhancedCurrentHue":28101,"currentSaturation":159,"colorTemperature":153}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":2,"stackVersion":1,"hwVersion":0,"dateCode":"20240902","swBuildId":"1.122.2","zclVersion":8,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1742078791063}

### Zigbee2MQTT version

2.1.3

### Comments

I'm not sure if this is just a model variance due to country but after pairing this I have 'device unsupported' and also unable to use the gradient feature of this lamp. 

compared to a Hue gradient lightstrip (looking at Exposes), i'm missing the gradient scene. 

Anyone know how I can use the gradient scene on this? Other 'versions' of this shows supported. 

Thanks!

### External definition

```shell
const philips = require('zigbee-herdsman-converters/lib/philips');

const definition = {
    zigbeeModel: ['929003103601'],
    model: '929003103601',
    vendor: 'Philips',
    description: 'Automatically generated definition',
    extend: [philips.m.light({"colorTemp":{"range":[153,500]},"color":{"modes":["xy","hs"],"enhancedHue":true}})],
    meta: {},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

gradient scene missing

Koenkk, 2025-03-16:
Could you make a PR by clicking [here](https://github.com/Koenkk/zigbee-herdsman-converters/edit/master/src/devices/philips.ts)?

comdr-chocchip, 2025-03-23:
> Could you make a PR by clicking [here](https://github.com/Koenkk/zigbee-herdsman-converters/edit/master/src/devices/philips.ts)?

Sorry not entirely sure if I’m doing it right. I click that, it asks if I want to fork it and then just post it?

Koenkk, 2025-03-23:
I added it now.

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

Could you also submit a picture for the docs? This can be done by:
1. Create a fork by clicking [here](https://github.com/Koenkk/zigbee2mqtt.io/fork)
2. Go to the `public/images/devices` directory, *Add file* -> *Upload files*
3. Upload the files and press *Commit changes*
4. Press *Contribute* -> *Open pull request* -> update title/description -> *Create pull request*

Make sure that:
- The filename is `929003103601.png`
- The size is 512x512
- The background is transparent (use e.g. [Adobe remove background](https://new.express.adobe.com/tools/remove-background))

closed 2025-03-23 by Koenkk