Koenkk/zigbee2mqtt #26755: Unable to pair device

Hjord, 2025-03-16:
### What happened?

Trying to pair this one:
https://www.zigbee2mqtt.io/devices/TS011F_plug_3.html
It says PLUG-001SPB2 on the bottom of the plug.

### Zigbee2MQTT version

2.1.1

### Adapter firmware version

20201026

### Adapter

CC2652RB

### Setup

Home Assistant Add-On

### Debug log

```
info 2025-03-16 15:12:19z2m: Device '0xa4c138ebd380f27f' joined
info 2025-03-16 15:12:19z2m: Starting interview of '0xa4c138ebd380f27f'
info 2025-03-16 15:12:19z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c138ebd380f27f","ieee_address":"0xa4c138ebd380f27f"},"type":"device_joined"}'
info 2025-03-16 15:12:19z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c138ebd380f27f","ieee_address":"0xa4c138ebd380f27f","status":"started"},"type":"device_interview"}'
info 2025-03-16 15:12:19z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c138ebd380f27f","ieee_address":"0xa4c138ebd380f27f"},"type":"device_announce"}'
error 2025-03-16 15:12:32zh:controller: Interview failed for '0xa4c138ebd380f27f with error 'Error: Interview failed because can not get node descriptor ('0xa4c138ebd380f27f')'
error 2025-03-16 15:12:32z2m: Failed to interview '0xa4c138ebd380f27f', device has not successfully been paired
info 2025-03-16 15:12:33z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c138ebd380f27f","ieee_address":"0xa4c138ebd380f27f","status":"failed"},"type":"device_interview"}'
warning 2025-03-16 15:12:40z2m: Device '0xa4c138ebd380f27f' left the network
```

lux73, 2025-03-16:
Update your Coordinator to latest Firmware 20240710

My Slaesh Coordinator works like a charme with it

Hjord, 2025-03-24:
I have had this device working before on the same coordinator.

closed 2025-05-31 by github-actions[bot]