Koenkk/zigbee2mqtt #26760: SPM01-U01 BITUO TECHNIK Energy metter -  z2m: Failed to configure at startup and reconfigure

GiambaFr, 2025-03-16:
### What happened?

Zigbee2mqtt dashboard show error message at sartup and reconfigure for the SPM01-U01 device.
Device seem to work correctly after that.
I have 2 of them and both are giving the same error

### What did you expect to happen?

No error message.

### How to reproduce it (minimal and precise)

Just pair the device.

### Zigbee2MQTT version

2.0 -> 2.1.3

### Adapter firmware version

20230507

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus ZBDongle-P TI CC2652P

### Setup

Systemctl on Ubuntu-server 20.04, VMWare

### Debug log

`z2m: Failed to configure 'Energie Cuisine', attempt 2 (Error: ZCL command 0xbc026efffea7f48d/1 seMetering.configReport([{"attribute":"currentSummDelivered","minimumReportInterval":5,"maximumReportInterval":3600,"reportableChange":0}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'FAILURE') at Endpoint.checkStatus (/opt/zigbee2mqtt/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:349:28) at Endpoint.zclCommand (/opt/zigbee2mqtt/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:954:26) at Endpoint.configureReporting (/opt/zigbee2mqtt/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:686:9) at Object.currentSummDelivered (/opt/zigbee2mqtt/node_modules/.pnpm/zigbee-herdsman-converters@23.2.0/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:127:5) at Object.configure (/opt/zigbee2mqtt/node_modules/.pnpm/zigbee-herdsman-converters@23.2.0/node_modules/zigbee-herdsman-converters/src/devices/bituo_technik.ts:32:13) at Configure.configure (/opt/zigbee2mqtt/lib/extension/configure.ts:124:13) at EventEmitter.wrappedCallback (/opt/zigbee2mqtt/lib/eventBus.ts:204:17))`

GiambaFr, 2025-03-27:
What can I do to fix that problem  ?

Koenkk, 2025-03-28:
@script0803 could you take a look at this?

script0803, 2025-03-29:
Hi @GiambaFr , this PR[https://github.com/Koenkk/zigbee-herdsman-converters/pull/9069] has fixed the issue. Please wait for the next Zigbee2MQTT update.

closed 2025-03-30 by Koenkk