Koenkk/zigbee2mqtt #26763: External Converters Loading Stopping at errors

ivanfmartinez, 2025-03-16:
### What happened?

After upgrading to 2.x one of my devices stopped working (As it was a device that i did not use normally I did not see this before), the device shows as unsupported but no error for this external converter are show on the log.

But there was an error related to an old device that I did not use anymore.

After some checking I found that when one external converter fails to load the others are not loaded.

Looking at the code as this is a generic JS class the problem can happen at other places 

Here de load of all JS files a single for iterating over the files, with no error check
https://github.com/Koenkk/zigbee2mqtt/blob/6d6372ccf07b9755eceb097ca43efcdfcb251f88/lib/extension/externalJS.ts#L154

Here the load of a single external converter which log the error and throw and this is cancelling the load all loop

https://github.com/Koenkk/zigbee2mqtt/blob/6d6372ccf07b9755eceb097ca43efcdfcb251f88/lib/extension/externalConverters.ts#L42

### What did you expect to happen?

Only the faulty external converter should not be load and all others can be loaded.

### How to reproduce it (minimal and precise)

Create 2 external converters that are valid, check on the log what order they are loaded. 

Change the first one to have an error.

restart z2m, it should load one and skip the one with error, but none are loaded.

The test code are using only valid converters and loading one at a time. Should have option to load all  and have invalid converter also

https://github.com/Koenkk/zigbee2mqtt/blob/6d6372ccf07b9755eceb097ca43efcdfcb251f88/test/extensions/externalConverters.test.ts#L121

### Zigbee2MQTT version

2.1.3

### Adapter firmware version

20230507

### Adapter

sonoff

### Setup

docker container, but I think this will happen on any setup

### Debug log

This error was for the external converter with error, but the problem is that others that should be loaded after this are skipped.
 

```
[2025-03-16 17:46:23] error:    z2m: Failed to call 'ExternalConverters' 'start' (/app/data/extension/TS0225_sensor_presenca.js:34
                e.occupancy(), e.illuminance_lux(),
                                 ^

TypeError: e.illuminance_lux is not a function
    at /app/data/extension/TS0225_sensor_presenca.js:34:20
    at Script.runInContext (node:vm:149:12)
    at Script.runInNewContext (node:vm:154:17)
    at runInNewContext (node:vm:310:38)
    at ExternalConverters.loadModuleFromText (/app/lib/extension/externalJS.ts:187:24)
    at ExternalConverters.loadFiles (/app/lib/extension/externalJS.ts:156:52)
    at processTicksAndRejections (node:internal/process/task_queues:105:5)
    at ExternalConverters.start (/app/lib/extension/externalJS.ts:45:9)
    at Controller.callExtensions (/app/lib/controller.ts:368:17)
    at Controller.start (/app/lib/controller.ts:186:9))

```

ivanfmartinez, 2025-03-17:
@mrmaximas the problem that I related is not about the illuminance, is about not loading other converters after one converter gives exception. 

That converter with illuminance errors is just an older converter that I dont need anymore, but as they still on the directory it breaks the load of other converter.

I'm not reporting the error with illuminance converter, I just showed an example that is the only error show, but the other converter that are correct are not loaded.

mrmaximas, 2025-03-17:
@ivanfmartinez External converters and extensions
The external_converters setting is no longer used. Instead all external converters inside data/external_converters directory are now automatically loaded. Make sure to move all your external converters to this directory.
External extensions are now loaded from data/external_extensions instead of data/extension. Make sure to rename your data/extension directory (if present) to data/external_extensions.
The constructor signature for external extensions now matches that of internal extensions (with settings and logger as extras as before). The updated signature can be inspected [here](https://github.com/Koenkk/zigbee2mqtt/blob/dev/lib/extension/externalExtensions.ts#L43-L54).

mrmaximas, 2025-03-17:
@ivanfmartinez how about this?
External converters and extensions
The external_converters setting is no longer used. Instead all external converters inside data/external_converters directory are now automatically loaded. Make sure to move all your external converters to this directory.
External extensions are now loaded from data/external_extensions instead of data/extension. Make sure to rename your data/extension directory (if present) to data/external_extensions.
The constructor signature for external extensions now matches that of internal extensions (with settings and logger as extras as before). The updated signature can be inspected [here](https://github.com/Koenkk/zigbee2mqtt/blob/dev/lib/extension/externalExtensions.ts#L43-L54).

Koenkk, 2025-03-18:
Will be fixed once https://github.com/Nerivec/zigbee2mqtt/commits/externaljs-dynamic-import is merged.

closed 2025-04-08 by Nerivec