Koenkk/zigbee2mqtt #26764: No connection to any device anymore, without an update to sync up with it.

DavyLandman, 2025-03-16:
### What happened?

Without any software or hardware changes, zigbee2mqtt cannot reach the the lights in my house. Sometimes something still happens, but 30/60s after I've pressed the button.

I did not update zigbee2mqtt, home-assistant, nor did I change any devices in my house.

I keep seeing the error: `'No network route' (205)`

### Zigbee2MQTT version

2.1.3

### Adapter firmware version

20240710 (I upgraded it while trying to solve the issue, it did not solve it)

### Adapter

slaesh's cc2652rb on a usb extension cable

### Setup

Z2M plain running on a x64 podman container 

### Debug log

Here I try to send a state change to a device that is 3m away, just open air.

```
[2025-03-16 22:51:20] debug:    z2m: Publishing 'set' 'state' to 'FLOALT Led Paneel Zolder'
[2025-03-16 22:51:20] debug:    zh:controller:endpoint: ZCL command 0x842e14fffe996674/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false})
[2025-03-16 22:51:20] debug:    zh:zstack: sendZclFrameToEndpointInternal 0x842e14fffe996674:24503/1 (0,0,1)
[2025-03-16 22:51:20] debug:    zh:zstack:znp: --> SREQ: AF - dataRequest - {"dstaddr":24503,"destendpoint":1,"srcendpoint":1,"clusterid":6,"transid":1,"options":0,"radius":30,"len":3,"data":{"type":"Buffer","data":[1,12,1]}}
[2025-03-16 22:51:20] debug:    zh:zstack:unpi:writer: --> frame [254,13,36,1,183,95,1,1,6,0,1,0,30,3,1,12,1,214]
[2025-03-16 22:51:20] debug:    zh:zstack:unpi:parser: <-- [254,1,100,1,0,100]
[2025-03-16 22:51:20] debug:    zh:zstack:unpi:parser: --- parseNext [254,1,100,1,0,100]
[2025-03-16 22:51:20] debug:    zh:zstack:unpi:parser: --> parsed 1 - 3 - 4 - 1 - [0] - 100
[2025-03-16 22:51:20] debug:    zh:zstack:znp: <-- SRSP: AF - dataRequest - {"status":0}
[2025-03-16 22:51:20] debug:    zh:zstack:unpi:parser: --- parseNext []
[2025-03-16 22:51:22] debug:    zh:zstack:unpi:parser: <-- [254,3,68,128,205,1,1,10]
[2025-03-16 22:51:22] debug:    zh:zstack:unpi:parser: --- parseNext [254,3,68,128,205,1,1,10]
[2025-03-16 22:51:22] debug:    zh:zstack:unpi:parser: --> parsed 3 - 2 - 4 - 128 - [205,1,1] - 10
[2025-03-16 22:51:22] debug:    zh:zstack:znp: <-- AREQ: AF - dataConfirm - {"status":205,"endpoint":1,"transid":1}
[2025-03-16 22:51:22] debug:    zh:zstack:unpi:parser: --- parseNext []
[2025-03-16 22:51:22] debug:    zh:zstack: Data confirm error (0x842e14fffe996674:24503,205,0)
[2025-03-16 22:51:22] debug:    zh:zstack: Wait 2000ms
[2025-03-16 22:51:24] debug:    zh:zstack: sendZclFrameToEndpointInternal 0x842e14fffe996674:24503/1 (0,1,1)
[2025-03-16 22:51:24] debug:    zh:zstack:znp: --> SREQ: AF - dataRequest - {"dstaddr":24503,"destendpoint":1,"srcendpoint":1,"clusterid":6,"transid":2,"options":0,"radius":30,"len":3,"data":{"type":"Buffer","data":[1,12,1]}}
[2025-03-16 22:51:24] debug:    zh:zstack:unpi:writer: --> frame [254,13,36,1,183,95,1,1,6,0,2,0,30,3,1,12,1,213]
[2025-03-16 22:51:24] debug:    zh:zstack:unpi:parser: <-- [254,1,100,1,0,100]
[2025-03-16 22:51:24] debug:    zh:zstack:unpi:parser: --- parseNext [254,1,100,1,0,100]
[2025-03-16 22:51:24] debug:    zh:zstack:unpi:parser: --> parsed 1 - 3 - 4 - 1 - [0] - 100
[2025-03-16 22:51:24] debug:    zh:zstack:znp: <-- SRSP: AF - dataRequest - {"status":0}
[2025-03-16 22:51:24] debug:    zh:zstack:unpi:parser: --- parseNext []
[2025-03-16 22:51:25] debug:    zh:zstack:unpi:parser: <-- [254,3,68,128,205,1,2,9]
[2025-03-16 22:51:25] debug:    zh:zstack:unpi:parser: --- parseNext [254,3,68,128,205,1,2,9]
[2025-03-16 22:51:25] debug:    zh:zstack:unpi:parser: --> parsed 3 - 2 - 4 - 128 - [205,1,2] - 9
[2025-03-16 22:51:25] debug:    zh:zstack:znp: <-- AREQ: AF - dataConfirm - {"status":205,"endpoint":1,"transid":2}
[2025-03-16 22:51:25] debug:    zh:zstack:unpi:parser: --- parseNext []
[2025-03-16 22:51:25] debug:    zh:zstack: Data confirm error (0x842e14fffe996674:24503,205,1)
[2025-03-16 22:51:25] debug:    zh:zstack: Discovering route to 24503
[2025-03-16 22:51:25] debug:    zh:zstack:znp: --> SREQ: ZDO - extRouteDisc - {"dstAddr":24503,"options":0,"radius":30}
[2025-03-16 22:51:25] debug:    zh:zstack:unpi:writer: --> frame [254,4,37,69,183,95,0,30,146]
[2025-03-16 22:51:25] debug:    zh:zstack:unpi:parser: <-- [254,1,101,69,0,33]
[2025-03-16 22:51:25] debug:    zh:zstack:unpi:parser: --- parseNext [254,1,101,69,0,33]
[2025-03-16 22:51:25] debug:    zh:zstack:unpi:parser: --> parsed 1 - 3 - 5 - 69 - [0] - 33
[2025-03-16 22:51:25] debug:    zh:zstack:znp: <-- SRSP: ZDO - extRouteDisc
[2025-03-16 22:51:25] debug:    zh:zstack:unpi:parser: --- parseNext []
[2025-03-16 22:51:28] debug:    zh:zstack: sendZclFrameToEndpointInternal 0x842e14fffe996674:24503/1 (0,2,1)
[2025-03-16 22:51:28] debug:    zh:zstack:znp: --> SREQ: AF - dataRequest - {"dstaddr":24503,"destendpoint":1,"srcendpoint":1,"clusterid":6,"transid":3,"options":0,"radius":30,"len":3,"data":{"type":"Buffer","data":[1,12,1]}}
[... 88 lines cut ...]
[2025-03-16 22:56:23] debug:    zh:zstack:unpi:writer: --> frame [254,3,37,49,183,95,0,255]
[2025-03-16 22:56:23] debug:    zh:zstack:unpi:parser: <-- [254,1,101,49,0,85]
[2025-03-16 22:56:23] debug:    zh:zstack:unpi:parser: --- parseNext [254,1,101,49,0,85]
[2025-03-16 22:56:23] debug:    zh:zstack:unpi:parser: --> parsed 1 - 3 - 5 - 49 - [0] - 85
[2025-03-16 22:56:23] debug:    zh:zstack:znp: <-- SRSP: ZDO - mgmtLqiReq
[2025-03-16 22:56:23] debug:    zh:zstack:unpi:parser: --- parseNext []
[2025-03-16 22:56:38] debug:    zh:zstack:unpi:writer: --> frame [254,3,37,49,183,95,0,255]
[2025-03-16 22:56:38] debug:    zh:zstack:unpi:parser: <-- [254,1,101,49,0,85]
[2025-03-16 22:56:38] debug:    zh:zstack:unpi:parser: --- parseNext [254,1,101,49,0,85]
[2025-03-16 22:56:38] debug:    zh:zstack:unpi:parser: --> parsed 1 - 3 - 5 - 49 - [0] - 85
[2025-03-16 22:56:38] debug:    zh:zstack:znp: <-- SRSP: ZDO - mgmtLqiReq
[2025-03-16 22:56:38] debug:    zh:zstack:unpi:parser: --- parseNext []
[2025-03-16 22:56:48] error:    z2m: Failed to execute LQI for 'FLOALT Led Paneel Zolder'
[2025-03-16 22:56:48] debug:    z2m: Error: AREQ - ZDO - mgmtLqiRsp after 10000ms
    at Object.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/utils/waitress.ts:67:23)
    at func (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:423:47)
    at Queue.execute (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/utils/queue.ts:36:20)
    at ZStackAdapter.sendZdoInternal (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:428:43)
    at ZStackAdapter.sendZdo (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:305:16)
    at request (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/controller/model/device.ts:1176:30)
    at Device.lqi (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/controller/model/device.ts:1197:42)
    at /app/lib/extension/networkMap.ts:211:75
    at requestWithRetry (/app/lib/extension/networkMap.ts:198:24)
    at NetworkMap.networkScan (/app/lib/extension/networkMap.ts:211:32
```

DavyLandman, 2025-03-16:
Note, I did not join #26738 (it happened after HA upgrade) or #23571 (different error messages, plus my error is not sporadic, it used to work fine, until it suddenly didn't.)

DavyLandman, 2025-03-17:
@Nerivec is there somewhere else I should report this? Or different debug logs I should share? I'm lacking tools to further diagnose this.

Nerivec, 2025-03-17:
It's already reported in the linked issue. Waiting for some feedback from Koenkk on zstack firmware.

DavyLandman, 2025-03-17:
I assumed this was different, as I had no software or hardware changes and it stopped working. My bad for raising the issue.

Koenkk, 2025-03-18:
Could you try with `99250304` from https://github.com/Koenkk/Z-Stack-firmware/issues/518#issuecomment-2698726919?

DavyLandman, 2025-03-19:
> Could you try with `99250304` from [Koenkk/Z-Stack-firmware#518 (comment)](https://github.com/Koenkk/Z-Stack-firmware/issues/518#issuecomment-2698726919)?

Hi @Koenkk, I have a `CC2652RB` stick, and the firmware listed is only for the `CC2652R`? 

And slaesh explicitly warns against that:

> In both cases, we will need the CC2652 **RB file.**

Is there a RB version available of that firmware?

DavyLandman, 2025-03-20:
> Added the CC2652RB fw to [Koenkk/Z-Stack-firmware#518 (comment)](https://github.com/Koenkk/Z-Stack-firmware/issues/518#issuecomment-2698726919)

Thanks, I'll try tomorrow morning.

DavyLandman, 2025-03-21:
Sadly, no, it did not fix it:

```
 error:    z2m: Publish 'set' 'state' to 'FLOALT Led Paneel Zolder' failed: 'Error: ZCL command 0x842e14fffe996674/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Data request failed with error: 'MAC channel access failure' (225))'
```

Koenkk, 2025-03-21:
This error indicates wireless interference, the z2m docs FAQ explain how to reduce interference

DavyLandman, 2025-03-21:
Nothing changed in my network, and it's already on an extension cable.
The device I test with has a clear line of sight only 2--3 meters.

I do have an esp32-c6 laying around, can I use that for sniffing? (as it has zigbee support?)

Koenkk, 2025-03-21:
I'm not sure if sniffing will bring you much here, based on the error it is clearly interference, sometimes a different USB port also helps.

DavyLandman, 2025-05-19:
I should have updated this issue. It suddenly flipped. Everything worked again, except for a single (Ikea) switch, that now only works partially. I suspect it was the cause for the issue.

closed 2025-07-27 by github-actions[bot]