Koenkk/zigbee2mqtt #26778: Zigbee network crashing without clear cause

ericvanbeurden, 2025-03-17:
### What happened?

Everything worked stable without problem. I added an RFXcom adaptor in Oracle Virtual box around the 8th or 9th of March 
Probably I also did some of the HA updates and other plugin updates and I also did a Oracle Virtual Box update to the latest version.
To the best of my knowledge after rebooting my virtual machine the zigbee network stopped working, although I'm not 100% sure since my focus was on getting the RFXcom adapter to work. 

Diggin into the zigbee problem I noticed that it runs very shortly after which it crashes. 
Zigbee2MQTT continues to run. 
Devices are listed but not responding, for example switching a device on/off gives me this: 

`z2m: Publish 'set' 'state' to 'Power Kantoor' failed: 'Error: ZCL command 0x54ef441000523187/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms)'`

### What did you expect to happen?

My zigbee network was rock solid, hope to get it back.

### How to reproduce it (minimal and precise)

When I reboot my virtual machine zigbee starts working, it also reports stats (power, temp, etc), however after 5 to 10 minutes nothing is reported anymore. After that I see that in my log the devices stop to respond. See see my log. 

### Zigbee2MQTT version

2.1.3-1

### Adapter firmware version

20210708

### Adapter

Sonoff ZIgbee 3.0 CP210x

### Setup

Hassio 15 / 25.3.3 on Virtual Box 7.1.6

### Debug log

[45df7312_zigbee2mqtt_2025-03-17T19-43-53.432Z.log](https://github.com/user-attachments/files/19296611/45df7312_zigbee2mqtt_2025-03-17T19-43-53.432Z.log)

Nerivec, 2025-03-17:
Have you tried unplugging the sonoff adapter for a while?
Looks like it's sending all kinds of corrupted frames (unknown commands). If it's not the adapter's internal state that has a problem, then I'd say you're looking at some kind of interference... at the radio, at the connection, at the passthrough...
Does it still do it if you remove that new adapter you added?

ericvanbeurden, 2025-03-18:
Thanks @Nerivec for your help. 
Tried all of your suggestions and more, but nothing helped, until I decided to load up and old virtual image in my virtual machine. 
Without any hardware changes everything works again without any errors in my log file.

Based on this: 
- No hardware problem
- No Oracle VirtualBox update problem

Potential problems: 
- Virtual image size (was 32G, but I resized to 64G). Logic would suggest this isn't the problem.
- HA version problem. I was running Hassio 15 / 25.3.3 which didn't work, now I'm back to core 2025.2.5 and supervisor 2025.03.3 which works without problem.
- Some other non-related plugin updates which I did.

I'm now going to make another backup and do the updates of the plugins and resize the virtual image size to 64G. If after that everything continues working logic suggest it is an hardware assist problem.

gras55, 2025-03-20:
I am experiencing the same issues. Is there a way to downgrade superviser?

robbinjanssen, 2025-03-22:
Anyone brave enough and tried with 2025.3.4? Im afraid my WAF will decrease again 😅

closed 2025-04-01 by ericvanbeurden