Koenkk/zigbee2mqtt #26780: [New device support]:

KlaaBoo, 2025-03-17:
### Link

https://nl.aliexpress.com/item/1005007920479553.html?spm=a2g0o.order_list.order_list_main.5.291379d2UsWEC7&gatewayAdapt=glo2nld

### Database entry

{"id":114,"type":"Router","ieeeAddr":"0xa4c138dbeae0c27f","nwkAddr":47543,"manufId":4417,"manufName":"_TZE204_m9dzckna","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"m�h/fn�h/\u0012","65506":56,"65508":0,"stackVersion":0,"dateCode":"","appVersion":74}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1742242449080}

### Zigbee2MQTT version

2.1.3-1

### Comments

I tried to follow the steps for external definition but stok at step 2..1. Extending the generated external definition
I have looked also how other people try to do it, but still couldn't find what to do. 

### External definition

```shell
zigbeeModel: ['TS0601'],
model: 'TS0601',
vendor: '_TZE204_m9dzckna',
description: 'temperature and humidity sensor',
extend: [],
meta: {},
```

### What does/doesn't work with the external definition?

I don't know what to chooze or what to fill in.

adpmel, 2025-04-02:
Good morning, I managed to make a converter for 
Modelo Zigbee
TS0601
Fabricante Zigbee
_TZE204_m9dzckna

GreetingS!
`
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const e = exposes.presets;

const definition = {
    fingerprint: [{ modelID: 'TS0601', manufacturerName: '_TZE204_m9dzckna' }],
    model: 'TS0601',
    vendor: 'Tuya',
    description: 'Tuya Temperature & Humidity Sensor',
    fromZigbee: [
        fz.ignore_tuya_set_time,
        fz.ignore_tuya_raw,
        {
            cluster: 'manuSpecificTuya',
            type: ['commandDataResponse', 'dataReport'],
            convert: (model, msg, publish, options, meta) => {
                const result = {};
                if (msg.data.dpValues) {
                    msg.data.dpValues.forEach((dpValue) => {
                        const bufferData = dpValue.data;
                         meta.logger.debug(`bufferData : ${bufferData}`)
                        // Check if bufferData has at least 4 elements
                        if (bufferData && bufferData.length >= 4) {
                            const value = (bufferData[2] << 8) | bufferData[3];

                            if (dpValue.dp === 1) {
                                result.temperature = value / 10;
                                meta.logger.debug(`Temperature raw: ${value}, converted: ${result.temperature}°C`);
                            } else if (dpValue.dp === 2) {
                                result.humidity = value / 10;
                                meta.logger.debug(`Humidity raw: ${value}, converted: ${result.humidity}%`);
                            }
                        } else {
                          meta.logger.debug(`bufferData is too short or undefined: ${JSON.stringify(bufferData)} , buffer: ${bufferData}`)
                        }
                    });
                }

                return result;
            }
        }
    ],
    toZigbee: [],
    exposes: [
        e.temperature(),
        e.humidity()
    ],
};

module.exports = definition;`

pfrase, 2025-04-02:
I have the same sensor and the same problem.

I tried to create the file with your code. Unfortunately the file is always invalid after saving. Can you upload the code in the correct format? As a text file?

pfrase, 2025-04-03:
example-user-extension1743679064499.js contains invalid code: missing ) after argument list

were is the missing ) ?

adpmel, 2025-04-03:
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const e = exposes.presets;

const definition = {
    fingerprint: [{ modelID: 'TS0601', manufacturerName: '_TZE204_m9dzckna' }],
    model: 'TS0601',
    vendor: 'Tuya',
    description: 'Tuya Temperature & Humidity Sensor',
    fromZigbee: [
        fz.ignore_tuya_set_time,
        fz.ignore_tuya_raw,
        {
            cluster: 'manuSpecificTuya',
            type: ['commandDataResponse', 'dataReport'],
            convert: (model, msg, publish, options, meta) => {
                const result = {};
                if (msg.data.dpValues) {
                    msg.data.dpValues.forEach((dpValue) => {
                        const bufferData = dpValue.data;
                         meta.logger.debug(`bufferData : ${bufferData}`)
                        // Check if bufferData has at least 4 elements
                        if (bufferData && bufferData.length >= 4) {
                            const value = (bufferData[2] << 8) | bufferData[3];

                            if (dpValue.dp === 1) {
                                result.temperature = value / 10;
                                meta.logger.debug(`Temperature raw: ${value}, converted: ${result.temperature}°C`);
                            } else if (dpValue.dp === 2) {
                                result.humidity = value / 10;
                                meta.logger.debug(`Humidity raw: ${value}, converted: ${result.humidity}%`);
                            }
                        } else {
                          meta.logger.debug(`bufferData is too short or undefined: ${JSON.stringify(bufferData)} , buffer: ${bufferData}`)
                        }
                    });
                }

                return result;
            }
        }
    ],
    toZigbee: [],
    exposes: [
        e.temperature(),
        e.humidity()
    ],
};

module.exports = definition;

pfrase, 2025-04-03:
Thank you!!! It works!

KlaaBoo, 2025-04-06:
Thank you. It is working.

closed 2025-04-06 by KlaaBoo

nckmml, 2025-04-10:
Hello,
thanks a lot for the converter!
Unfortunately this is still incomplete.
The while it shows as Celcius in zigbee2mqtt, the display itself still shows the temperature in Fahrenheit and there is no setting to switch it. According to the manual you would be able to switch the display in the tuya app.
Whats also missing is battery state. This would be important to know when to plug in the device to recharge it.

adpmel, 2025-04-14:
Hi, how are you?

I'm not a pro at making converters — in fact, this is my first one ever. I'm not sure if you need to change the temperature to Fahrenheit on the device itself or within the converter.

If it’s on the device, you can change it by pressing the pair button once.

If it’s in the converter, just let me know — I think with two lines of code, we can add the conversion.

I have the device connected via USB, so I haven’t investigated what messages are received for the battery (or maybe having it connected prevents those messages from being sent).

nckmml, 2025-04-14:
Hi, 

thx for the reply. The device has no option to change the temperature on it directly. But power cycling the whole thing by letting it run out of battery while its being connected to Home Assistant using your adapter seems to set it to Celcius, so thats fine for me now but probably not that great for people that want to use it on Fahrenheit.

For the battery, I only started using it that day I wrote the previous comment and had no idea how long the battery runs. I expected it to last quite a while before needing to be plugged in again. Turns out without a power supply the battery runs out within a day and seems more like a UPS / power backup option than being made for running on battery. So not sure how important the battery state really is.