Koenkk/zigbee2mqtt #26782: Aqara Mini Switch not connecting

Daafip, 2025-03-17:
### What happened?

I'm using this aqara switch https://www.zigbee2mqtt.io/devices/WXKG13LM.html#aqara-wxkg13lm.
It will connect, start interviewing, but won't pair:

```
error 2025-03-17 23:05:00zh:controller: Interview failed for '0x54ef441000d9c795 with error 'Error: Interview failed because can not get node descriptor ('0x54ef441000d9c795')'
error 2025-03-17 23:05:00z2m: Failed to interview '0x54ef441000d9c795', device has not successfully been paired
```

### What did you expect to happen?

It paired to another HA system using ZHA but doesn't seem to work with Z2M. The older version of the aqara mini does work for me.

### How to reproduce it (minimal and precise)

Connect aqara mini by holding pair button for 5 seconds.

### Zigbee2MQTT version

2.1.3

### Adapter firmware version

0x264d0900

### Adapter

ConBee3

### Setup

Homeassistant add-on with HAos v14.2 on a generic x86

### Debug log

Will add later

Nerivec, 2025-03-17:
Did you do the usual procedure for battery-powered devices "push a button every 1-2 seconds while pairing to keep it awake"?
You can also try to pair it to a specific device (instead of "permit join all").
Aqara is known for having pairing troubles, and flat out refuse with a lot of routers.

Daafip, 2025-03-17:
>Did you do the usual procedure for battery-powered devices "push a button every 1-2 seconds while pairing to keep it awake"?

Yup, tried a few times ...

Will try specific join, but the interview process starts but shows errors half way...

>Aqara is known for having pairing troubles, and flat out refuse with a lot of routers.

Yes I am aware but because I have a working one, and it showed compatibel thought I'd try.

Koenkk, 2025-03-18:
It may also help to pair it (very) close to the coordinator.

Daafip, 2025-03-18:
Was sitting 30cm from it, is that close enough?

pinkfloydFR, 2025-03-19:
I know my comment is not very helping, but i have 3 of this button, and never had problem connecting to z2m, perhaps yours is doa ? :(

Daafip, 2025-03-19:
> perhaps yours is doa

You never know, could be but will test on another HA instance: weird thing is that it sort of connects but doesnt finish

Daafip, 2025-03-19:
in case this helps ?

[logs_trimmed.txt](https://github.com/user-attachments/files/19350298/logs_trimmed.txt)

Daafip, 2025-03-19:
Updated firmware of my deconz, and tried a couple more times. 10cm aparently is enough and spamming the pair button continously for 30seconds or so. Thanks for your time and effort @Koenkk @Nerivec !!

closed 2025-03-19 by Daafip