Koenkk/zigbee2mqtt #26783: Обновление с версии 1.42.0-2 до 2.1.3-1

RastopchinSS, 2025-03-18:
### What happened?

**Описание проблемы:**
Я использовал zigbee2mqtt на версии 1.42.0-2 долгое время. Сегодня решил обновить дополнения до версии 2.1.3-1.
Но после обновления дополнения перестали запускаться.

**Я использую:** 
Home Assistant Core 2025.3.3
Supervisor 2025.03.3
Operating System 14.2
Пользовательский интерфейс 20250306.0

**Лог ошибок запуска:**

### What did you expect to happen?

Я ожидал после обновления нормальной работы.

### How to reproduce it (minimal and precise)

Обновить с версии 1.42.0-2 до версии 2.1.3-1

### Zigbee2MQTT version

1.42.0-2

### Adapter firmware version

20241001 (prod-dual)

### Adapter

UZG-01 XZG

### Setup

обычный x86-64

### Debug log

[2025-03-18 10:00:15] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt_4/bridge/state', payload 'offline'
[2025-03-18 10:00:15] info: 	z2m: Disconnecting from MQTT server
[2025-03-18 10:00:15] info: 	z2m: Stopping zigbee-herdsman...
[2025-03-18 10:00:27] info: 	zh:controller: Wrote coordinator backup to '/config/zigbee2mqtt_4/coordinator_backup.json'
[2025-03-18 10:00:27] info: 	zh:zstack:znp: closing
[2025-03-18 10:00:27] info: 	z2m: Stopped zigbee-herdsman
[2025-03-18 10:00:27] info: 	z2m: Stopped Zigbee2MQTT
/app/node_modules/winston/node_modules/readable-stream/lib/_stream_writable.js:264
  var er = new ERR_STREAM_WRITE_AFTER_END();
           ^
Error: write after end
    at writeAfterEnd (/app/node_modules/winston/node_modules/readable-stream/lib/_stream_writable.js:264:12)
    at DerivedLogger.Writable.write (/app/node_modules/winston/node_modules/readable-stream/lib/_stream_writable.js:300:21)
    at DerivedLogger.log (/app/node_modules/winston/lib/winston/logger.js:231:12)
    at Logger.log (/app/lib/util/logger.ts:198:25)
    at Logger.info (/app/lib/util/logger.ts:211:14)
    at Znp.onPortClose (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:96:16)
    at Object.onceWrapper (node:events:634:26)
    at Socket.emit (node:events:531:35)
    at TCP.<anonymous> (node:net:338:12)
[10:00:36] INFO: Preparing to start...
[10:00:37] INFO: Socat not enabled
[10:00:37] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2025-03-18 10:00:55] info: 	z2m: Logging to console, file (filename: log.log)
[2025-03-18 10:00:55] info: 	z2m: Starting Zigbee2MQTT version 2.1.3 (commit #unknown)
[2025-03-18 10:00:55] info: 	z2m: Starting zigbee-herdsman (3.2.7)
[2025-03-18 10:00:55] error: 	z2m: Error while starting zigbee-herdsman
[2025-03-18 10:00:55] error: 	z2m: Failed to start zigbee-herdsman
[2025-03-18 10:00:55] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-03-18 10:00:55] error: 	z2m: Exiting...
[2025-03-18 10:00:55] error: 	z2m: Error: Cannot discover TCP adapters at this time. Specify valid 'adapter' and 'port' in your configuration.
    at findTCPAdapter (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/adapter/adapterDiscovery.ts:470:15)
    at discoverAdapter (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/adapter/adapterDiscovery.ts:496:26)
    at Function.create (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/adapter/adapter.ts:68:54)
    at Controller.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/controller/controller.ts:132:38)
    at Zigbee.start (/app/lib/zigbee.ts:69:47)
    at Controller.start (/app/lib/controller.ts:142:13)
    at start (/app/index.js:161:5)
[10:01:03] INFO: Preparing to start...
[10:01:03] INFO: Socat not enabled
[10:01:04] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2025-03-18 10:01:12] info: 	z2m: Logging to console, file (filename: log.log)
[2025-03-18 10:01:12] info: 	z2m: Starting Zigbee2MQTT version 2.1.3 (commit #unknown)
[2025-03-18 10:01:12] info: 	z2m: Starting zigbee-herdsman (3.2.7)
[2025-03-18 10:01:12] error: 	z2m: Error while starting zigbee-herdsman
[2025-03-18 10:01:12] error: 	z2m: Failed to start zigbee-herdsman
[2025-03-18 10:01:12] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-03-18 10:01:12] error: 	z2m: Exiting...
[2025-03-18 10:01:12] error: 	z2m: Error: Cannot discover TCP adapters at this time. Specify valid 'adapter' and 'port' in your configuration.
    at findTCPAdapter (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/adapter/adapterDiscovery.ts:470:15)
    at discoverAdapter (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/adapter/adapterDiscovery.ts:496:26)
    at Function.create (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/adapter/adapter.ts:68:54)
    at Controller.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/controller/controller.ts:132:38)
    at Zigbee.start (/app/lib/zigbee.ts:69:47)
    at Controller.start (/app/lib/controller.ts:142:13)
    at start (/app/index.js:161:5)
[10:01:18] INFO: Preparing to start...
[10:01:18] INFO: Socat not enabled
[10:01:20] INFO: Starting Zigbee2MQTT...

RastopchinSS, 2025-03-18:
Настройки запуска дополнения z2m:

data_path: /config/zigbee2mqtt_2
socat:
  enabled: false
  master: pty,raw,echo=0,link=/tmp/ttyZ2M,mode=777
  slave: tcp-listen:8486,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5
  options: "-d -d"
  log: false
mqtt:
  base_topic: zigbee2mqtt_2
serial:
  port: tcp://192.168.177.160:6638
  baudrate: 115200

RastopchinSS, 2025-03-19:
Помогите пожалуйста разобраться в ситуации

mrmaximas, 2025-03-19:
@RastopchinSS you'll get help faster if you write in English.

RastopchinSS, 2025-03-20:
**Problem description:**
I have been using zigbee2 mqtt on version 1.42.0-2 for a long time. Today, it was decided to upgrade to version 2.1.3-1.
But after installing the updates, they stopped running.

**I use:**
Home Assistant Core 2025.3.3
Supervisor 2025.03.3
Operating System 14.2
User Interface 20250306.0

siandco, 2025-04-04:
_[2025-03-18 10:01:12] error: z2m: Error: Cannot discover TCP adapters at this time. Specify valid 'adapter' and 'port' in your configuration._

According to the error message in log you provided (see above), you need to include **adapter: zstack** under the serial: section of your config.

REMEMBER TO BACKUP YOUR CONFIG FILE PRIOR TO MAKING ANY CHANGES

RastopchinSS, 2025-04-08:
Thank you very much!

It really helped.
I have registered it in my configuration:

serial:
   adapter: zstack
   port: tcp://192.168.177.156:6638
   baudrate: 115200