Koenkk/zigbee2mqtt #26784: OTA does not work on the latest Z2M version for DIY device

andreivicol, 2025-03-18:
### What happened?

Hello. I've been developing a temperature sensor using Ti's CC2652RB MCU and testing it using Z2M. I've been working on and off for about a year on this implementation, but lately I've encountered an issue that I had resolved a long time ago. To be more specific, it seems that none of the devices I'm testing appear to support OTA updates. I find this very weird, as last year around this time I could do a 200KB ota update in about 30-40 minutes (i was on version 1.38 at that time). Not only did I manage to do an update, I was doing updates reliably at speeds faster than most off the shelf devices. I've delved deeper and found some weird things, but can't seem to find a solution.

I've since updated Z2M to version 2.1.3 and devices do not appear as having OTA anymore. I've tested this with 30 of my custom devices on three different coordinators: ZigStar UZG-01, Sonoff 3.0 USB DONGLE PLUS, and a SMLight SLZB-06M. I'm running Z2M as a Home Assistant add-on in two ways: on a virtual machine and on a Raspberry Pi 5. I've also tried running the coordinators over wifi, ethernet and USB. None of the combinations work anymore, while last summer updates worked like a charm. 

When I publish to zigbee2mqtt/bridge/request/device/ota_update/check, the response I get is:
{"data":{},"error":"Device 'ABC04S23' does not support OTA updates","status":"error"}

I thought that I may have broken ota through the changes I made to the firmware during the past year, but I took a firmware dated 22-04-2024 (a firmware from the commit stating fully functional ota) and flashed a device with it. The behavior is identical to my latest build.

My device has two endpoints and OTA is implemented on both endpoints.

Have there been any major changes concerning ota sometimes in the past year? I cannot seem to find a guide that gives a comprehensive requirements list for functional OTA updates. I've checked the changelogs and have not observed any mentions regarding breaking ota changes.

I host my firmware image as a .zigbee file on a public github repo and i've updated the ota override file with the sha512 and otaHeader, yet nothing.

I do not think it has anything to do with the OTA file, as Z2M reports that the device does not support ota, thus does not get the check the validity of the ota file or the repository.

Thank you to whoever takes their time to chime in on this one.

### What did you expect to happen?

An OTA update.

### How to reproduce it (minimal and precise)

Update to version 2.1.3 and create a custom device implementing ota as in Texas Intrument's guides.

### Zigbee2MQTT version

2.1.3

### Adapter firmware version

7.3.1.0 build 176

### Adapter

ZigStar UZG-01, Sonoff 3.0 USB DONGLE PLUS, and a SMLight SLZB-06M

### Setup

Add-on on HA running on both Raspberry Pi and virtual machine.

Nerivec, 2025-03-20:
Can you make sure your external converters match the current typing?
https://github.com/Koenkk/zigbee-herdsman-converters/blob/dedbf69866b1a7882109c8be4ecccb3120d5fdc5/src/lib/types.ts#L263-L296
More specifically, your devices supporting OTA should have the [ota](https://github.com/Koenkk/zigbee-herdsman-converters/blob/dedbf69866b1a7882109c8be4ecccb3120d5fdc5/src/lib/types.ts#L280) field set to `true` or to the appropriate metadata.
This property serves as a way to force/bypass device firmware mistakes (and to avoid logic for the large number of devices that despite the cluster presence, never had, or will -likely- have firmware updates available).

https://github.com/Koenkk/zigbee2mqtt/blob/6d6372ccf07b9755eceb097ca43efcdfcb251f88/lib/extension/otaUpdate.ts#L185-L191

andreivicol, 2025-03-20:
> Can you make sure your external converters match the current typing? https://github.com/Koenkk/zigbee-herdsman-converters/blob/dedbf69866b1a7882109c8be4ecccb3120d5fdc5/src/lib/types.ts#L263-L296 More specifically, your devices supporting OTA should have the [ota](https://github.com/Koenkk/zigbee-herdsman-converters/blob/dedbf69866b1a7882109c8be4ecccb3120d5fdc5/src/lib/types.ts#L280) field set to `true` or to the appropriate metadata. This property serves as a way to force/bypass device firmware mistakes (and to avoid logic for the large number of devices that despite the cluster presence, never had, or will -likely- have firmware updates available).
> 
> [zigbee2mqtt/lib/extension/otaUpdate.ts](https://github.com/Koenkk/zigbee2mqtt/blob/6d6372ccf07b9755eceb097ca43efcdfcb251f88/lib/extension/otaUpdate.ts#L185-L191)
> 
> Lines 185 to 191 in [6d6372c](/Koenkk/zigbee2mqtt/commit/6d6372ccf07b9755eceb097ca43efcdfcb251f88)
> 
>  if (!(device instanceof Device)) { 
>      error = `Device '${ID}' does not exist`; 
>  } else if (!device.definition || !device.definition.ota) { 
>      error = `Device '${device.name}' does not support OTA updates`; 
>  } else if (this.inProgress.has(device.ieeeAddr)) { 
>      error = `Update or check for update already in progress for '${device.name}'`; 
>  } else {

Hello and thank you for your reply.

I've updated my external converter, saved it, restarted HA and tried to update my device, but to no avail. I've also tried reconfiguring, interviewing and repairing my device, but the message i get is the same: device does not support OTA.

This is my current external converter. I used the auto-generation to which i appended OTA.

```
const m = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['AFTSHT35X2'],
    model: 'AFTSHT35X2',
    vendor: 'TeXXXInstruments',
    description: 'AFT Temperature Sensor with SHT35 and NTC',
    extend: [m.deviceEndpoints({"endpoints":{"8":8,"9":9}}), m.battery(), m.temperature({"endpointNames":["8","9"]}), m.humidity()],
    meta: {"multiEndpoint":true},
    ota: true,
    }
};

module.exports = definition;
```

This is my old external converter:
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const ota = require('zigbee-herdsman-converters/lib/ota');
const e = exposes.presets;
const ea = exposes.access;

const definitions = [
    {
    zigbeeModel: ['AFTSHT35X2'],
    model: 'AFTSHT35X2',
    vendor: 'TeXXXInstruments',
    description: 'Temperature sensor SHT35',
    ota: ota.zigbeeOTA,
    fromZigbee: [fz.temperature, fz.humidity, fz.battery],
    toZigbee: [],
    meta: {multiEndpoint: true//,
    },
    exposes: [e.temperature(1).withEndpoint(8), e.humidity().withEndpoint(8), e.battery(), e.battery_voltage(), e.temperature(2).withEndpoint(9)],
    configure: async (device, coordinatorEndpoint, logger) => {
         
        await reporting.bind(device.getEndpoint(8), coordinatorEndpoint, ['msTemperatureMeasurement', 'msRelativeHumidity', 'genPowerCfg']);
        await reporting.bind(device.getEndpoint(9), coordinatorEndpoint, ['msTemperatureMeasurement']);
        
        await reporting.batteryPercentageRemaining(device.getEndpoint(8), {min: 10, max: 60, change: 0});
        await reporting.temperature(device.getEndpoint(8), {min: 10, max: 60, change: 0});
        await reporting.humidity(device.getEndpoint(8), {min: 10, max: 60, change: 0});
        await reporting.temperature(device.getEndpoint(9), {min: 10, max: 60, change: 0});
        
        try 
        {
            await device.getEndpoint(8).read('genPowerCfg', ['batteryVoltage']);
            await device.getEndpoint(8).read('msTemperatureMeasurement', ['measuredValue']);
            await device.getEndpoint(8).read('msRelativeHumidity', ['measuredValue']);
            await device.getEndpoint(9).read('msTemperatureMeasurement', ['measuredValue']);
        } catch (e)
        {
            
        }
    }
    }
];

module.exports = definitions;
```

They both behave the same way. Also, how can i get the reporting and reading commands in the new version of external converters?

Nerivec, 2025-03-20:
> This is my current external converter.

Not sure if this is a typo on copy/paste, but this is not valid JS.
```diff
    ota: true,
-    }
```
(The old one is definitely no longer valid `ota.zigbeeOTA` no longer exists.)

PS: Whenever you change a device's firmware (changing interview details), you need to use `interview` button (the blue `i`), it's the only one that re-fetches all info from the device (bypasses the cache).

andreivicol, 2025-03-20:
Thank you for pointing that mistake out. I fixed it, saved the file, restarted HA, re-interviewed the device but the behaviour is the same. I've checked the js file with an online JS syntax validator and it seems ok. Can't figure out what might be the issue. I've used TI's guide for OTA (am using CC2652) and their libraries. Updates used to work, so i know that this project was updateable a few months back (when i was on version 1.38).

Nerivec, 2025-03-20:
If you keep getting the same error (`Device '${device.name}' does not support OTA updates`), it means the definition isn't found (converter not matching), or `ota` prop is undefined/falsy.
Can you share some `debug` logs?

andreivicol, 2025-03-20:
Sure. These are the messages seen in the Logs tab of the Z2M Addon:
```
Info 2025-03-20 13:52:59z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/check', payload '{"data":{},"error":"Device 'AFT01STH' does not support OTA updates","status":"error"}'
Error 2025-03-20 13:52:59z2m: Device 'AFT01STH' does not support OTA updates
```

When I use the 'Check All' button from the OTA tab of Z2M, nothing appears to happen. Neither in in logs tab, nor in the add-on's logs. Thank you for taking the time to respond. Let me know if there are further ways i could debug this.

Nerivec, 2025-03-20:
Set `log_level: debug` and grab the logs from the file directly (`debug` is not shown in frontend by default).

andreivicol, 2025-03-20:
Log level was already set to debug, as i always keep it like this. I've also checked the detailed logs from the add-on and the messages are the same. I've copied a lengthier log from the add-on, maybe it shows something i missed.

```
[2025-03-20 14:09:16] debug: 	z2m:mqtt: Received MQTT message on 'zigbee2mqtt/bridge/request/device/ota_update/check' with data '{"id": "0x00124b0031e5d9a8"}'
[2025-03-20 14:09:16] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/check', payload '{"data":{},"error":"Device 'AFT01STH' does not support OTA updates","status":"error"}'
[2025-03-20 14:09:16] error: 	z2m: Device 'AFT01STH' does not support OTA updates
[2025-03-20 14:09:18] debug: 	zh:ezsp:uart: <-- [5328b1a97d312a15b6588d4a2dab5593499c72e767faf89874f1c77d3188fcc081f5cdcede6f8ffd4e397e]
[2025-03-20 14:09:18] debug: 	zh:ezsp:uart: <-- DATA (5,3,0): 5328b1a9112a15b6588d4a2dab5593499c72e767faf89874f1c71188fcc081f5cdcede6f8ffd4e397e
[2025-03-20 14:09:18] debug: 	zh:ezsp:uart: --> ACK  (6)
[2025-03-20 14:09:18] debug: 	zh:ezsp:uart: --> [8610be7e]
[2025-03-20 14:09:18] debug: 	zh:ezsp:uart: <-- ACK (3): 5328b1a9112a15b6588d4a2dab5593499c72e767faf89874f1c71188fcc081f5cdcede6f8ffd4e397e
[2025-03-20 14:09:18] debug: 	zh:ezsp:ezsp: <== Frame: 6a9001450000040119000801000100003cc0cc1736ffff0c01720100bebe52260300000002
[2025-03-20 14:09:18] debug: 	zh:ezsp:ezsp: <== 0x45: {"_cls_":"incomingMessageHandler","_id_":69,"_isRequest_":false,"type":0,"apsFrame":{"profileId":260,"sequence":60,"clusterId":25,"sourceEndpoint":8,"destinationEndpoint":1,"groupId":0,"options":256},"lastHopLqi":192,"lastHopRssi":-52,"sender":13847,"bindingIndex":255,"addressIndex":255,"message":{"type":"Buffer","data":[1,114,1,0,190,190,82,38,3,0,0,0]}}
[2025-03-20 14:09:18] debug: 	zh:ezsp: processMessage: {"messageType":0,"apsFrame":{"profileId":260,"sequence":60,"clusterId":25,"sourceEndpoint":8,"destinationEndpoint":1,"groupId":0,"options":256},"lqi":192,"rssi":-52,"sender":13847,"bindingIndex":255,"addressIndex":255,"message":{"type":"Buffer","data":[1,114,1,0,190,190,82,38,3,0,0,0]}}
[2025-03-20 14:09:18] debug: 	zh:controller: Received payload: clusterID=25, address=13847, groupID=0, endpoint=8, destinationEndpoint=1, wasBroadcast=false, linkQuality=192, frame={"header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":114,"commandIdentifier":1},"payload":{"fieldControl":0,"manufacturerCode":48830,"imageType":9810,"fileVersion":3},"command":{"ID":1,"response":2,"parameters":[{"name":"fieldControl","type":32},{"name":"manufacturerCode","type":33},{"name":"imageType","type":33},{"name":"fileVersion","type":35}],"name":"queryNextImageRequest"}}
[2025-03-20 14:09:18] debug: 	z2m: Received Zigbee message from 'AFT04S02', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":3,"imageType":9810,"manufacturerCode":48830}' from endpoint 8 with groupID 0
[2025-03-20 14:09:18] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/AFT04S02', payload '{"battery":48,"humidity_8":15.21,"last_seen":"2025-03-20T14:09:18+02:00","linkquality":192,"temperature_8":43.3,"temperature_9":-80.15,"voltage":3800}'
[2025-03-20 14:09:18] debug: 	z2m: Device 'AFT04S02' requested OTA
[2025-03-20 14:09:18] debug: 	zh:controller:endpoint: CommandResponse 0x00124b0031e58b3e/8 genOta.queryNextImageResponse({"status":152}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"reservedBits":0,"writeUndiv":false})
[2025-03-20 14:09:18] debug: 	zh:ezsp: sendZclFrameToEndpointInternal 0x00124b0031e58b3e:13847/8 (0,0,1), timeout=10000
[2025-03-20 14:09:18] debug: 	zh:ezsp:ezsp: ==> sendUnicast: {"type":0,"indexOrDestination":13847,"apsFrame":{"profileId":260,"sequence":191,"clusterId":25,"sourceEndpoint":1,"destinationEndpoint":8,"groupId":0,"options":256},"messageTag":192,"message":{"type":"Buffer","data":[25,114,2,152]}}
[2025-03-20 14:09:18] debug: 	zh:ezsp:ezsp: ==> {"_cls_":"sendUnicast","_id_":52,"_isRequest_":true,"type":0,"indexOrDestination":13847,"apsFrame":{"profileId":260,"sequence":191,"clusterId":25,"sourceEndpoint":1,"destinationEndpoint":8,"groupId":0,"options":256},"messageTag":192,"message":{"type":"Buffer","data":[25,114,2,152]}}
[2025-03-20 14:09:18] debug: 	zh:ezsp:uart: --> DATA (3,6,0): 6b0001340000173604011900010800010000bfc00419720298
[2025-03-20 14:09:18] debug: 	zh:ezsp:uart: --> [362921a9602a15a56f904b3caa549a499d4e27142dca7d5ef9ff5e429e7e]
[2025-03-20 14:09:18] debug: 	zh:ezsp:uart: -?- waiting (4)
[2025-03-20 14:09:18] debug: 	zh:ezsp:uart: <-- [6429a1a9602a15f49b897e]
[2025-03-20 14:09:18] debug: 	zh:ezsp:uart: <-- DATA (6,4,0): 6429a1a9602a15f49b897e
[2025-03-20 14:09:18] debug: 	zh:ezsp:uart: --> ACK  (7)
[2025-03-20 14:09:18] debug: 	zh:ezsp:uart: --> [87009f7e]
[2025-03-20 14:09:18] debug: 	zh:ezsp:uart: <-- ACK (4): 6429a1a9602a15f49b897e
[2025-03-20 14:09:18] debug: 	zh:ezsp:ezsp: <== Frame: 6b800134000046
[2025-03-20 14:09:18] debug: 	zh:ezsp:ezsp: <== 0x34: {"_cls_":"sendUnicast","_id_":52,"_isRequest_":false,"status":0,"sequence":70}
[2025-03-20 14:09:18] debug: 	zh:ezsp:uart: -+- waiting (4) success
[2025-03-20 14:09:18] debug: 	z2m: Responded to OTA request of 'AFT04S02' with 'NO_IMAGE_AVAILABLE'
[2025-03-20 14:09:18] debug: 	zh:ezsp:uart: <-- [7429b1a96b2a15a56f904b3caa549a499d4e27ed2dce678f097e]
[... cut ...]
[2025-03-20 14:09:18] debug: 	zh:ezsp:uart: <-- [252eb1a96b2a158e29904b3caa549a499d4e27ec2cce6708bf7e]
[2025-03-20 14:09:18] debug: 	zh:ezsp:uart: <-- DATA (2,5,0): 252eb1a96b2a158e29904b3caa549a499d4e27ec2cce6708bf7e
[2025-03-20 14:09:18] debug: 	zh:ezsp:uart: --> ACK  (3)
[2025-03-20 14:09:18] debug: 	zh:ezsp:uart: --> [83401b7e]
[2025-03-20 14:09:18] debug: 	zh:ezsp:uart: <-- ACK (5): 252eb1a96b2a158e29904b3caa549a499d4e27ec2cce6708bf7e
[2025-03-20 14:09:18] debug: 	zh:ezsp:ezsp: <== Frame: 6c90013f00003c700401190001080001000047c10000
```

Nerivec, 2025-03-20:
Sorry, should have been more precise. I need the logs from startup (to see if converter is properly loadded) and from pairing (to see detection).

andreivicol, 2025-03-20:
Ok, thank you. I've taken the following two logs: one at startup and one for pairing the device. Its name is AFT01STH.

[startup.log](https://github.com/user-attachments/files/19367140/startup.log)

[pairing.log](https://github.com/user-attachments/files/19367145/pairing.log)

The logs are pretty long, i hope they contain all the info.

Nerivec, 2025-03-20:
Detection looks fine:
```
Device 'AFT01STH' is supported, identified as: TeXXXInstruments AFT Temperature Sensor with SHT35 and NTC (AFTSHT35X2)
```

However the state is a bit weird, I'm not seeing some of the logging that should be happening when getting the definition.

Can you try that:
- remove the device from Z2M (do NOT force remove)
- shutdown Z2M
- start Z2M
- pair the device again

This will ensure a cleaner pairing.

andreivicol, 2025-03-20:
Hello and thank you. I've created the log you've instructed me to.

[clean_repair.log](https://github.com/user-attachments/files/19367722/clean_repair.log)

I've noticed some failure messages in the Z2M front end after repairing, during the configuration step, but could not manage to read their contents as there were too many notifications at the same time and they all disappeared simultaneously. 

Could be this one, but i'm not 100% sure:
```
[2025-03-20 15:12:47] [34mdebug[39m: 	z2m: Received Zigbee message from 'AFT01STH', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 8 with groupID 0
[2025-03-20 15:12:47] [34mdebug[39m: 	z2m: Skipping message, still interviewing
[2025-03-20 15:12:47] [32minfo[39m: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/AFT01STH', payload '{"last_seen":"2025-03-20T15:12:47+02:00","linkquality":180}'
[2025-03-20 15:12:47] [34mdebug[39m: 	zh:controller:endpoint: Error: ZCL command 0x00124b0031e5d9a8/8 genBasic.read(["swBuildId"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false,"sendPolicy":"immediate"}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
    at Endpoint.checkStatus (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:349:28)
    at Endpoint.zclCommand (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:954:26)
    at processTicksAndRejections (node:internal/process/task_queues:105:5)
    at Endpoint.read (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:445:29)
    at Device.interviewInternal (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/controller/model/device.ts:933:42)
    at Device.interview (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/controller/model/device.ts:750:13)
    at Controller.onDeviceJoined (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/controller/controller.ts:754:17)
[2025-03-20 15:12:47] [34mdebug[39m: 	zh:controller:device: Interview - failed to read attribute 'softwareBuildID' from endpoint '8' (Error: ZCL command 0x00124b0031e5d9a8/8 genBasic.read(["swBuildId"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false,"sendPolicy":"immediate"}) failed (Status 'UNSUPPORTED_ATTRIBUTE'))
```

andreivicol, 2025-03-20:
I've checked my yaml file and, indeed, it was missing that line. I reinstalled Z2M on this instance of HA and forgot to add the override location. It works and all devices appear to have updates now. I just started a test update to make sure everything works. Thank you for taking your time to help. Greatly appreciated!

Nerivec, 2025-03-23:
Did you have any more troubles, or is it okay to close this?

andreivicol, 2025-03-23:
Everything works now! Thank you very much.

closed 2025-03-23 by Nerivec