Koenkk/zigbee2mqtt #26787: Innr SP 240 no Energy Data

Acey-yo, 2025-03-18:
### What happened?

Hello everyone,

I'm using two Innr SP 240s. Initially, the energy data transmission worked perfectly. Now I've noticed that this data is no longer being transmitted unless I manually initiate the transmission. Then everything arrives in Home Assistant. Additionally, the transmitted total consumption value is incorrect.

Does anyone have a solution or an idea what the problem might be?

Thanks

### Zigbee2MQTT version

2.1.3

### Adapter firmware version

7.4.4 [GA]

### Adapter

EmberZNet

### Setup

Addon on HomeAssistant running in Synology VM

Marrislul, 2025-04-18:
Got somewhat the same problem with my SP240s. Mine report a Value after manually updating, but the value is way too high.

InsideTheVoid, 2025-05-10:
> Now I've noticed that this data is no longer being transmitted unless I manually initiate the transmission. Then everything arrives in Home Assistant.

The fix for automatically retrieving the values in Home Assistant (HA) was to update the firmware of the plug. After that, it was mandatory to perform a factory reset by holding the power button for 5 seconds to eliminate any old values. Finally, I set the reporting settings as outlined below.

[image]

Everything works now, except that the current is measured in milliamps instead of amps, so it's off by a factor of 1000.

guydb74, 2025-05-14:
> The fix for automatically retrieving the values in Home Assistant (HA) was to update the firmware of the plug. After that, it was mandatory to perform a factory reset by holding the power button for [...]

Do you have to pair the plugs again after the reset? 
I have the same issue as the OP and hope it fixes the incorrect reporting of energyconsumption.

InsideTheVoid, 2025-05-15:
Yes. To perform the reset and pair:

1. Set z2m to 'Permit Join'.
2. Hold the power button on the plug for 5 seconds until it flashes.

Then the device leaves z2m and comes back after a few seconds.

3. Setup reporting like above.

guydb74, 2025-05-15:
OK, but I had to do a reconfigure after the rebind and then set the reporting as depicted in your screenshot.
Now everything works as exptected. Thank you very much.

closed 2025-07-23 by github-actions[bot]

me-chiel, 2025-09-03:
Thank you all for posting this!
I do have a question.
- Updated the firmware via OTA to 1.9.30
- Set permit to join & reset the INNR plug so it came back in Z2M
- opened the "Reporting" page
- filled in all the settings as in the screenshot
- Hit the Reconfigure button in the "About" page
- "Exposes" page seems to have credible results! So far, so good!

when I open the "Reporting" page again, all previous settings have changed.
is this normal behaviour? I expected to see all the settings as mentioned in the comments above. (with the 0,005 etc)
can anyone tell me what I did wrong? or right:) ?

[image]