Koenkk/zigbee2mqtt #26791: [Feature request]:

Koenkk, 2025-03-18:
### Is your feature request related to a problem? Please describe

d

### Describe the solution you'd like

d

### Describe alternatives you've considered

d

### Additional context

d

mrmaximas, 2025-03-18:
Thanks! I'm gonna need a few minutes

mrmaximas, 2025-03-18:
I put ext.js in the external_converters folder, restarted z2m, but nothing changed in the interface.

[image]

Koenkk, 2025-03-18:
I forgot to mention, to try it out, send a MQTT message to `zigbee2mqtt/YOUR_DEVICE_FRIENDLY_NAME/set` payload `{"command": ""}`

mrmaximas, 2025-03-18:
seems like works
{"brightness":164,"countdown":5,"last_seen":"2025-03-18T23:03:37+03:00","linkquality":98,"max_brightness":254,"min_brightness":30,"state":"ON","update":{"installed_version":70,"latest_version":70,"state":"idle"}}

mrmaximas, 2025-03-18:
Strange thing. I tried it on other device that I haven't tried to write a timer to through the console before. 
There is nothing about countdown.
{"brightness":141,"last_seen":"2025-03-18T23:19:23+03:00","linkquality":131,"max_brightness":254,"min_brightness":80,"state":"ON","update":{"installed_version":70,"latest_version":70,"state":"idle"}}

mrmaximas, 2025-03-18:
I changed the brightness, gave a command, in the test dimmer the brightness changed, countdown 5s is there. 
On other dimmers the countdown does not appear in the reply.
{"brightness":142,"countdown":5,"last_seen":"2025-03-18T23:23:01+03:00","linkquality":109,"max_brightness":254,"min_brightness":30,"state":"ON","update":{"installed_version":70,"latest_version":70,"state":"idle"}}

mrmaximas, 2025-03-18:
just saw the errors in the z2m log.

```
2025-03-18 23:14:10z2m: Publish 'set' 'command' to 'Kitchen Main' failed: 'Error: command is not a object, got string ("")'
2025-03-18 23:14:10z2m: Publish 'set' 'command' to 'Kitchen Main' failed: 'Error: command is not a object, got string ("")'
2025-03-18 23:14:11z2m: Publish 'set' 'command' to 'Kitchen Main' failed: 'Error: command is not a object, got string ("")'
2025-03-18 23:18:11z2m: Publish 'set' 'command' to 'Kitchen Main' failed: 'Error: command is not a object, got string ("")'
2025-03-18 23:18:11z2m: Publish 'set' 'command' to 'Kitchen Main' failed: 'Error: command is not a object, got string ("")'
2025-03-18 23:18:11z2m: Publish 'set' 'command' to 'Hall Main' failed: 'Error: command is not a object, got string ("")'
2025-03-18 23:18:11z2m: Publish 'set' 'command' to 'Hall Main' failed: 'Error: command is not a object, got string ("")'
2025-03-18 23:18:11z2m: Publish 'set' 'command' to 'Hall Entrance' failed: 'Error: command is not a object, got string ("")'
2025-03-18 23:20:26z2m: Publish 'set' 'command' to 'Hall Entrance' failed: 'Error: command is not a object, got string ("")'
2025-03-18 23:24:37z2m: Publish 'set' 'command' to 'Kitchen Table' failed: 'Error: command is not a object, got string ("")'
2025-03-18 23:24:37z2m: Publish 'set' 'command' to 'Kitchen Table' failed: 'Error: command is not a object, got string ("")'
2025-03-18 23:43:51z2m: Publish 'set' 'command' to 'Kitchen Table' failed: 'Error: command is not a object, got string ("")'
2025-03-18 23:43:51z2m: Publish 'set' 'command' to 'Kitchen Main' failed: 'Error: command is not a object, got string ("")'
2025-03-18 23:43:51z2m: Publish 'set' 'command' to 'Kitchen Main' failed: 'Error: command is not a object, got string ("")'****
```

Koenkk, 2025-03-20:
I updated https://gist.github.com/Koenkk/b402cb48a8f88915c92b1d1fa8252db7, can you try again? You have to use `tuya_command` now instead of `command`.

mrmaximas, 2025-03-20:
@Koenkk `z2m: No converter available for 'tuya_command' ("")`

mrmaximas, 2025-03-20:
z2m startup log
```
[2025-03-20 23:01:32.145] error: 	z2m: Failed to start 'ExternalConverters' (node:internal/modules/cjs/loader:646
      throw e;
      ^

Error: Cannot find module '/Users/max/zigbee2mqtt/node_modules/zigbee-herdsman-converters/dist/lib/extend.js'
    at createEsmNotFoundErr (node:internal/modules/cjs/loader:1261:15)
    at finalizeEsmResolution (node:internal/modules/cjs/loader:1250:15)
    at resolveExports (node:internal/modules/cjs/loader:639:14)
    at Function._findPath (node:internal/modules/cjs/loader:728:31)
    at Function._resolveFilename (node:internal/modules/cjs/loader:1211:27)
    at Function._load (node:internal/modules/cjs/loader:1055:27)
    at TracingChannel.traceSync (node:diagnostics_channel:322:14)
    at wrapModuleLoad (node:internal/modules/cjs/loader:220:24)
    at Module.require (node:internal/modules/cjs/loader:1311:12)
    at require (node:internal/modules/helpers:136:16))
```

mrmaximas, 2025-03-20:
Hm. I rerun the update and everything seems to work. Some components did not update the first time.

mrmaximas, 2025-03-20:
I don't know what was going on, but after restarting Mac OS, the icons are displayed.

mrmaximas, 2025-03-20:
i found `ext.js` renamed to `ext.js.invalid`

Koenkk, 2025-03-21:
Updated https://gist.github.com/Koenkk/b402cb48a8f88915c92b1d1fa8252db7, can you try again?

mrmaximas, 2025-03-21:
@Koenkk same renamed `ext.js.invalid `

Koenkk, 2025-03-22:
Updated https://gist.github.com/Koenkk/b402cb48a8f88915c92b1d1fa8252db7, please try again.

mrmaximas, 2025-03-22:
@Koenkk good news. ext.js don't touched by z2m, command accepted, this is answers after each publishing
```
2025-03-22 18:21:25z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Kitchen Main', payload '{"brightness":172,"last_seen":"2025-03-22T18:21:26+03:00","linkquality":80,"max_brightness":254,"min_brightness":90,"state":"OFF","update":{"installed_version":70,"latest_version":70,"state":"idle"}}'
Info 2025-03-22 18:21:25z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Kitchen Main', payload '{"brightness":172,"last_seen":"2025-03-22T18:21:26+03:00","linkquality":83,"max_brightness":254,"min_brightness":90,"state":"OFF","update":{"installed_version":70,"latest_version":70,"state":"idle"}}'
Info 2025-03-22 18:21:26z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Kitchen Main', payload '{"brightness":172,"last_seen":"2025-03-22T18:21:26+03:00","linkquality":80,"max_brightness":254,"min_brightness":90,"state":"OFF","update":{"installed_version":70,"latest_version":70,"state":"idle"}}'
Info 2025-03-22 18:21:26z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Kitchen Main', payload '{"brightness":172,"last_seen":"2025-03-22T18:21:26+03:00","linkquality":76,"max_brightness":254,"min_brightness":90,"state":"OFF","update":{"installed_version":70,"latest_version":70,"state":"idle"}}'
Info 2025-03-22 18:21:26z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Kitchen Main', payload '{"brightness":172,"last_seen":"2025-03-22T18:21:26+03:00","linkquality":72,"max_brightness":254,"min_brightness":90,"state":"OFF","update":{"installed_version":70,"latest_version":70,"state":"idle"}}'
```

mrmaximas, 2025-03-22:
This is debug after `{"tuya_command": "5"}`
```
2025-03-22 18:41:45.862] debug: 	z2m:mqtt: Received MQTT message on 'zigbee2mqtt/Kitchen Main/set' with data '{"tuya_command": "5"}'
[2025-03-22 18:41:45.863] debug: 	z2m: Publishing 'set' 'tuya_command' to 'Kitchen Main'
[2025-03-22 18:41:45.864] debug: 	zh:controller:endpoint: ZCL command 0xa4c138b5bb325444/1 genOnOff.tuyaCommand({"data":{"type":"Buffer","data":[255,255,0,0]}}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false})
[2025-03-22 18:41:45.864] debug: 	zh:zstack: sendZclFrameToEndpointInternal 0xa4c138b5bb325444:46233/1 (0,0,1)
[2025-03-22 18:41:45.864] debug: 	zh:zstack:znp: --> SREQ: AF - dataRequest - {"dstaddr":46233,"destendpoint":1,"srcendpoint":1,"clusterid":6,"transid":99,"options":0,"radius":30,"len":7,"data":{"type":"Buffer","data":[1,154,240,255,255,0,0]}}
[2025-03-22 18:41:45.864] debug: 	zh:zstack:unpi:writer: --> frame [254,17,36,1,153,180,1,1,6,0,99,0,30,7,1,154,240,255,255,0,0,14]
[2025-03-22 18:41:45.878] debug: 	zh:zstack:unpi:parser: <-- [254,1,100,1,0,100]
[2025-03-22 18:41:45.878] debug: 	zh:zstack:unpi:parser: --- parseNext [254,1,100,1,0,100]
[2025-03-22 18:41:45.879] debug: 	zh:zstack:unpi:parser: --> parsed 1 - 3 - 4 - 1 - [0] - 100
[2025-03-22 18:41:45.879] debug: 	zh:zstack:znp: <-- SRSP: AF - dataRequest - {"status":0}
[2025-03-22 18:41:45.879] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2025-03-22 18:41:45.881] debug: 	zh:zstack:unpi:parser: <-- [254,3,68,128,0,1,99,165]
[2025-03-22 18:41:45.882] debug: 	zh:zstack:unpi:parser: --- parseNext [254,3,68,128,0,1,99,165]
[2025-03-22 18:41:45.882] debug: 	zh:zstack:unpi:parser: --> parsed 3 - 2 - 4 - 128 - [0,1,99] - 165
[2025-03-22 18:41:45.882] debug: 	zh:zstack:znp: <-- AREQ: AF - dataConfirm - {"status":0,"endpoint":1,"transid":99}
[2025-03-22 18:41:45.882] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2025-03-22 18:41:45.911] debug: 	zh:zstack:unpi:parser: <-- [254,33,68,129,0,0,0,239,163,116,1,1,0,127,0,132,125,78,0,0,13,9,48,1,0,98,2,2,0,4,0,0,0,0,129,53,28,239]
[2025-03-22 18:41:45.912] debug: 	zh:zstack:unpi:parser: --- parseNext [254,33,68,129,0,0,0,239,163,116,1,1,0,127,0,132,125,78,0,0,13,9,48,1,0,98,2,2,0,4,0,0,0,0,129,53,28,239]
[2025-03-22 18:41:45.912] debug: 	zh:zstack:unpi:parser: --> parsed 33 - 2 - 4 - 129 - [0,0,0,239,163,116,1,1,0,127,0,132,125,78,0,0,13,9,48,1,0,98,2,2,0,4,0,0,0,0,129,53,28] - 239
[2025-03-22 18:41:45.912] debug: 	zh:zstack:znp: <-- AREQ: AF - incomingMsg - {"groupid":0,"clusterid":61184,"srcaddr":29859,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":127,"securityuse":0,"timestamp":5143940,"transseqnumber":0,"len":13,"data":{"type":"Buffer","data":[9,48,1,0,98,2,2,0,4,0,0,0,0]}}
[2025-03-22 18:41:45.912] debug: 	zh:controller: Received payload: clusterID=61184, address=29859, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=127, frame={"header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":48,"commandIdentifier":1},"payload":{"seq":25088,"dpValues":[{"dp":2,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,0]}}]},"command":{"ID":1,"parameters":[{"name":"seq","type":33},{"name":"dpValues","type":1011}],"name":"dataResponse"}}
[2025-03-22 18:41:45.913] debug: 	zh:controller:endpoint: ZCL command 0x187a3efffe358554/1 manuSpecificTuya.defaultRsp({"cmdId":1,"statusCode":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"transactionSequenceNumber":48,"writeUndiv":false})
[2025-03-22 18:41:45.913] debug: 	zh:zstack: sendZclFrameToEndpointInternal 0x187a3efffe358554:29859/1 (0,0,2)
[2025-03-22 18:41:45.913] debug: 	zh:zstack:znp: --> SREQ: AF - dataRequest - {"dstaddr":29859,"destendpoint":1,"srcendpoint":1,"clusterid":61184,"transid":100,"options":0,"radius":30,"len":5,"data":{"type":"Buffer","data":[16,48,11,1,0]}}
[2025-03-22 18:41:45.913] debug: 	zh:zstack:unpi:writer: --> frame [254,15,36,1,163,116,1,1,0,239,100,0,30,5,16,48,11,1,0,71]
[2025-03-22 18:41:45.914] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2025-03-22 18:41:45.914] debug: 	z2m: Received Zigbee message from 'Power Monitor', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,0],"type":"Buffer"},"datatype":2,"dp":2}],"seq":25088}' from endpoint 1 with groupID 0
[2025-03-22 18:41:45.944] debug: 	zh:zstack:unpi:parser: <-- [254,1,100,1,0,100,254,3,69,196,153,180,0,175,254,3,68,128,0,1,100,162,254,25,68,129,0,0,6,0,153,180,1,1,0,76,0,230,131,78,0,0,5,24,154,11,240,0,153,180,29,220]
[2025-03-22 18:41:45.944] debug: 	zh:zstack:unpi:parser: --- parseNext [254,1,100,1,0,100,254,3,69,196,153,180,0,175,254,3,68,128,0,1,100,162,254,25,68,129,0,0,6,0,153,180,1,1,0,76,0,230,131,78,0,0,5,24,154,11,240,0,153,180,29,220]
[2025-03-22 18:41:45.944] debug: 	zh:zstack:unpi:parser: --> parsed 1 - 3 - 4 - 1 - [0] - 100
[2025-03-22 18:41:45.944] debug: 	zh:zstack:znp: <-- SRSP: AF - dataRequest - {"status":0}
[2025-03-22 18:41:45.945] debug: 	zh:zstack:unpi:parser: --- parseNext [254,3,69,196,153,180,0,175,254,3,68,128,0,1,100,162,254,25,68,129,0,0,6,0,153,180,1,1,0,76,0,230,131,78,0,0,5,24,154,11,240,0,153,180,29,220]
[... cut ...]
[2025-03-22 18:41:46.037] debug: 	zh:controller: Received payload: clusterID=61184, address=29859, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=131, frame={"header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":49,"commandIdentifier":1},"payload":{"seq":25344,"dpValues":[{"dp":6,"datatype":0,"data":{"type":"Buffer","data":[9,82,0,11,12,0,2,5]}}]},"command":{"ID":1,"parameters":[{"name":"seq","type":33},{"name":"dpValues","type":1011}],"name":"dataResponse"}}

```

mrmaximas, 2025-03-25:
@Koenkk how do I write payload correctly via UI for 5 sec for example?

[image]

mrmaximas, 2025-03-25:
2025-03-25 09:59:27.844] debug: 	z2m:mqtt: Received MQTT message on 'zigbee2mqtt/Kitchen Main/set' with data '{"tuya_command": "0x05000000"}'
[2025-03-25 09:59:27.845] debug: 	z2m: Publishing 'set' 'tuya_command' to 'Kitchen Main'
[2025-03-25 09:59:27.845] debug: 	zh:controller:endpoint: ZCL command 0xa4c138b5bb325444/1 genOnOff.tuyaCommand({"data":{"type":"Buffer","data":[255,255,0,0]}}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false})
[2025-03-25 09:59:27.845] debug: 	zh:zstack: sendZclFrameToEndpointInternal 0xa4c138b5bb325444:46233/1 (0,0,2)
[2025-03-25 09:59:27.861] debug: 	zh:zstack:unpi:parser: <-- [254,1,100,1,0,100]
[2025-03-25 09:59:27.861] debug: 	zh:zstack:unpi:parser: --- parseNext [254,1,100,1,0,100]
[2025-03-25 09:59:27.861] debug: 	zh:zstack:unpi:parser: --> parsed 1 - 3 - 4 - 1 - [0] - 100
[2025-03-25 09:59:27.861] debug: 	zh:zstack:znp: <-- SRSP: AF - dataRequest - {"status":0}
[2025-03-25 09:59:27.861] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2025-03-25 09:59:27.861] debug: 	zh:zstack:znp: --> SREQ: AF - dataRequest - {"dstaddr":46233,"destendpoint":1,"srcendpoint":1,"clusterid":6,"transid":77,"options":0,"radius":30,"len":7,"data":{"type":"Buffer","data":[1,38,240,255,255,0,0]}}
[2025-03-25 09:59:27.861] debug: 	zh:zstack:unpi:writer: --> frame [254,17,36,1,153,180,1,1,6,0,77,0,30,7,1,38,240,255,255,0,0,156]
[2025-03-25 09:59:27.865] debug: 	zh:zstack:unpi:parser: <-- [254,3,68,128,0,1,76,138]
[2025-03-25 09:59:27.866] debug: 	zh:zstack:unpi:parser: --- parseNext [254,3,68,128,0,1,76,138]
[2025-03-25 09:59:27.866] debug: 	zh:zstack:unpi:parser: --> parsed 3 - 2 - 4 - 128 - [0,1,76] - 138
[2025-03-25 09:59:27.866] debug: 	zh:zstack:znp: <-- AREQ: AF - dataConfirm - {"status":0,"endpoint":1,"transid":76}
[2025-03-25 09:59:27.866] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2025-03-25 09:59:27.881] debug: 	zh:zstack:unpi:parser: <-- [254,1,100,1,0,100]
[2025-03-25 09:59:27.882] debug: 	zh:zstack:unpi:parser: --- parseNext [254,1,100,1,0,100]
[2025-03-25 09:59:27.882] debug: 	zh:zstack:unpi:parser: --> parsed 1 - 3 - 4 - 1 - [0] - 100
[2025-03-25 09:59:27.882] debug: 	zh:zstack:znp: <-- SRSP: AF - dataRequest - {"status":0}
[2025-03-25 09:59:27.882] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2025-03-25 09:59:27.888] debug: 	zh:zstack:unpi:parser: <-- [254,3,68,128,0,1,77,139]
[2025-03-25 09:59:27.888] debug: 	zh:zstack:unpi:parser: --- parseNext [254,3,68,128,0,1,77,139]
[2025-03-25 09:59:27.888] debug: 	zh:zstack:unpi:parser: --> parsed 3 - 2 - 4 - 128 - [0,1,77] - 139
[2025-03-25 09:59:27.889] debug: 	zh:zstack:znp: <-- AREQ: AF - dataConfirm - {"status":0,"endpoint":1,"transid":77}
[2025-03-25 09:59:27.889] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2025-03-25 09:59:27.944] debug: 	zh:zstack:unpi:parser: <-- [254,25,68,129,0,0,6,0,153,180,1,1,0,54,0,239,253,128,0,0,5,24,38,11,240,0,226,156,28,241,254,29,68,129,0,0,6,0,153,180,1,1,0,58,0,193,254,128,0,0,9,24,253,10,1,64,34,255,255,0,226,156,28,145]
[2025-03-25 09:59:27.944] debug: 	zh:zstack:unpi:parser: --- parseNext [254,25,68,129,0,0,6,0,153,180,1,1,0,54,0,239,253,128,0,0,5,24,38,11,240,0,226,156,28,241,254,29,68,129,0,0,6,0,153,180,1,1,0,58,0,193,254,128,0,0,9,24,253,10,1,64,34,255,255,0,226,156,28,145]
[2025-03-25 09:59:27.944] debug: 	zh:zstack:unpi:parser: --> parsed 25 - 2 - 4 - 129 - [0,0,6,0,153,180,1,1,0,54,0,239,253,128,0,0,5,24,38,11,240,0,226,156,28] - 241
[2025-03-25 09:59:27.944] debug: 	zh:zstack:znp: <-- AREQ: AF - incomingMsg - {"groupid":0,"clusterid":6,"srcaddr":46233,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":54,"securityuse":0,"timestamp":8453615,"transseqnumber":0,"len":5,"data":{"type":"Buffer","data":[24,38,11,240,0]}}
[2025-03-25 09:59:27.945] debug: 	zh:controller: Received payload: clusterID=6, address=46233, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=54, frame={"header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":38,"commandIdentifier":11},"payload":{"cmdId":240,"statusCode":0},"command":{"ID":11,"name":"defaultRsp","parameters":[{"name":"cmdId","type":32},{"name":"statusCode","type":32}]}}
[2025-03-25 09:59:27.946] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Kitchen Main', payload '{"brightness":173,"last_seen":"2025-03-25T09:59:27+03:00","linkquality":54,"max_brightness":254,"min_brightness":90,"state":"OFF","update":{"installed_version":70,"latest_version":70,"state":"idle"}}'
[2025-03-25 09:59:27.946] debug: 	zh:zstack:unpi:parser: --- parseNext [254,29,68,129,0,0,6,0,153,180,1,1,0,58,0,193,254,128,0,0,9,24,253,10,1,64,34,255,255,0,226,156,28,145]
[2025-03-25 09:59:27.949] debug: 	zh:zstack:unpi:parser: --> parsed 29 - 2 - 4 - 129 - [0,0,6,0,153,180,1,1,0,58,0,193,254,128,0,0,9,24,253,10,1,64,34,255,255,0,226,156,28] - 145
[2025-03-25 09:59:27.949] debug: 	zh:zstack:znp: <-- AREQ: AF - incomingMsg - {"groupid":0,"clusterid":6,"srcaddr":46233,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":58,"securityuse":0,"timestamp":8453825,"transseqnumber":0,"len":9,"data":{"type":"Buffer","data":[24,253,10,1,64,34,255,255,0]}}
[... cut ...]
[2025-03-25 09:59:28.432] debug: 	zh:zstack: sendZclFrameToEndpointInternal 0x187a3efffe358554:29859/1 (0,0,1)
[2025-03-25 09:59:28.432] debug: 	zh:zstack:znp: --> SREQ: AF - dataRequest - {"dstaddr":29859,"destendpoint":1,"srcendpoint":1,"clusterid":61184,"transid":78,"options":0,"radius":30,"len":5,"data":{"type":"Buffer","data":[16,43,11,1,0]}}
[2025-03-25 09:59:28.432] debug: 	zh:zstack:unpi:writer: --> frame [254,15,36,1,163,116,1,1,0,239,78,0,30,5,16,43,11,1,0,118]
[2025-03-25 09:59:28.432] debug: 	zh:zstack:unpi:parser: --- parseNext []

mrmaximas, 2025-03-25:
This is  {"tuya_command": "0x0"}'
```
[2025-03-25 10:08:24.817] debug: 	z2m:mqtt: Received MQTT message on 'zigbee2mqtt/Kitchen Main/set' with data '{"tuya_command": "0x0"}'
[2025-03-25 10:08:24.819] debug: 	z2m: Publishing 'set' 'tuya_command' to 'Kitchen Main'
[2025-03-25 10:08:24.819] debug: 	zh:controller:endpoint: ZCL command 0xa4c138b5bb325444/1 genOnOff.tuyaCommand({"data":{"type":"Buffer","data":[255,255,0,0]}}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false})
[2025-03-25 10:08:24.819] debug: 	zh:zstack: sendZclFrameToEndpointInternal 0xa4c138b5bb325444:46233/1 (0,0,1)
[2025-03-25 10:08:24.820] debug: 	zh:zstack:znp: --> SREQ: AF - dataRequest - {"dstaddr":46233,"destendpoint":1,"srcendpoint":1,"clusterid":6,"transid":37,"options":0,"radius":30,"len":7,"data":{"type":"Buffer","data":[1,80,240,255,255,0,0]}}
[2025-03-25 10:08:24.820] debug: 	zh:zstack:unpi:writer: --> frame [254,17,36,1,153,180,1,1,6,0,37,0,30,7,1,80,240,255,255,0,0,130]
[2025-03-25 10:08:24.842] debug: 	zh:zstack:unpi:parser: <-- [254,1,100,1,0,100]
[2025-03-25 10:08:24.842] debug: 	zh:zstack:unpi:parser: --- parseNext [254,1,100,1,0,100]
[2025-03-25 10:08:24.842] debug: 	zh:zstack:unpi:parser: --> parsed 1 - 3 - 4 - 1 - [0] - 100
[2025-03-25 10:08:24.843] debug: 	zh:zstack:znp: <-- SRSP: AF - dataRequest - {"status":0}
[2025-03-25 10:08:24.843] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2025-03-25 10:08:24.846] debug: 	zh:zstack:unpi:parser: <-- [254,3,68,128,0,1,37,227]
[2025-03-25 10:08:24.847] debug: 	zh:zstack:unpi:parser: --- parseNext [254,3,68,128,0,1,37,227]
[2025-03-25 10:08:24.847] debug: 	zh:zstack:unpi:parser: --> parsed 3 - 2 - 4 - 128 - [0,1,37] - 227
[2025-03-25 10:08:24.847] debug: 	zh:zstack:znp: <-- AREQ: AF - dataConfirm - {"status":0,"endpoint":1,"transid":37}
[2025-03-25 10:08:24.847] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2025-03-25 10:08:24.902] debug: 	zh:zstack:unpi:parser: <-- [254,25,68,129,0,0,6,0,153,180,1,1,0,47,0,176,27,161,0,0,5,24,80,11,240,0,226,156,28,6,254,29,68,129,0,0,6,0,153,180,1,1,0,51,0,130,28,161,0,0,9,24,14,10,1,64,34,255,255,0,226,156,28,235]
[2025-03-25 10:08:24.903] debug: 	zh:zstack:unpi:parser: --- parseNext [254,25,68,129,0,0,6,0,153,180,1,1,0,47,0,176,27,161,0,0,5,24,80,11,240,0,226,156,28,6,254,29,68,129,0,0,6,0,153,180,1,1,0,51,0,130,28,161,0,0,9,24,14,10,1,64,34,255,255,0,226,156,28,235]
[2025-03-25 10:08:24.903] debug: 	zh:zstack:unpi:parser: --> parsed 25 - 2 - 4 - 129 - [0,0,6,0,153,180,1,1,0,47,0,176,27,161,0,0,5,24,80,11,240,0,226,156,28] - 6
[2025-03-25 10:08:24.903] debug: 	zh:zstack:znp: <-- AREQ: AF - incomingMsg - {"groupid":0,"clusterid":6,"srcaddr":46233,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":47,"securityuse":0,"timestamp":10558384,"transseqnumber":0,"len":5,"data":{"type":"Buffer","data":[24,80,11,240,0]}}
[2025-03-25 10:08:24.903] debug: 	zh:controller: Received payload: clusterID=6, address=46233, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=47, frame={"header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":80,"commandIdentifier":11},"payload":{"cmdId":240,"statusCode":0},"command":{"ID":11,"name":"defaultRsp","parameters":[{"name":"cmdId","type":32},{"name":"statusCode","type":32}]}}
[2025-03-25 10:08:24.904] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Kitchen Main', payload '{"brightness":173,"last_seen":"2025-03-25T10:08:24+03:00","linkquality":47,"max_brightness":254,"min_brightness":90,"state":"OFF","update":{"installed_version":70,"latest_version":70,"state":"idle"}}'
[2025-03-25 10:08:24.905] debug: 	zh:zstack:unpi:parser: --- parseNext [254,29,68,129,0,0,6,0,153,180,1,1,0,51,0,130,28,161,0,0,9,24,14,10,1,64,34,255,255,0,226,156,28,235]
[2025-03-25 10:08:24.905] debug: 	zh:zstack:unpi:parser: --> parsed 29 - 2 - 4 - 129 - [0,0,6,0,153,180,1,1,0,51,0,130,28,161,0,0,9,24,14,10,1,64,34,255,255,0,226,156,28] - 235
[2025-03-25 10:08:24.905] debug: 	zh:zstack:znp: <-- AREQ: AF - incomingMsg - {"groupid":0,"clusterid":6,"srcaddr":46233,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":51,"securityuse":0,"timestamp":10558594,"transseqnumber":0,"len":9,"data":{"type":"Buffer","data":[24,14,10,1,64,34,255,255,0]}}
[2025-03-25 10:08:24.905] debug: 	zh:controller: Received payload: clusterID=6, address=46233, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=51, frame={"header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":14,"commandIdentifier":10},"payload":[{"attrId":16385,"dataType":34,"attrData":65535}],"command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}}
[2025-03-25 10:08:24.905] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2025-03-25 10:08:24.906] debug: 	z2m: Received Zigbee message from 'Kitchen Main', type 'attributeReport', cluster 'genOnOff', data '{"onTime":65535}' from endpoint 1 with groupID 0
[2025-03-25 10:08:24.906] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Kitchen Main', payload '{"brightness":173,"last_seen":"2025-03-25T10:08:24+03:00","linkquality":51,"max_brightness":254,"min_brightness":90,"state":"OFF","update":{"insta [... cut ...] faultResponse":true,"reservedBits":0},"transactionSequenceNumber":81,"commandIdentifier":1},"payload":[{"attrId":0,"status":0,"dataType":32,"attrData":3}],"command":{"ID":1,"name":"readRsp","parameters":[{"name":"attrId","type":33},{"name":"status","type":32},{"name":"dataType","type":32,"conditions":[{"type":"statusEquals","value":0}]},{"name":"attrData","type":1000,"conditions":[{"type":"statusEquals","value":0}]}]}}
[2025-03-25 10:08:25.378] debug: 	zh:zstack:unpi:parser: --- parseNext []
```

mrmaximas, 2025-03-25:
@Koenkk I may have misunderstood you at first, but I see that all my devices respond like this to any value, set timer 65535
`[2025-03-25 10:08:24.906] debug: 	z2m: Received Zigbee message from 'Kitchen Main', type 'attributeReport', cluster 'genOnOff', data '{"onTime":65535}' from endpoint 1 with groupID 0`
I remember that the set timer time is reported at 0x4001 and 0x4002
https://gist.github.com/Koenkk/b402cb48a8f88915c92b1d1fa8252db7?permalink_comment_id=5488626#gistcomment-5488626

Koenkk, 2025-03-25:
You cannot set it via MQTT currently, for now update the code on this line: https://gist.github.com/Koenkk/b402cb48a8f88915c92b1d1fa8252db7#file-ext-js-L17

mrmaximas, 2025-03-25:
@Koenkk it works! Bravo!
`await entity.command('genOnOff', 'tuyaCommand', {data: Buffer.from([0x05, 0x00, 0x00, 0x00])}, {});` switches the device on or off after 5 seconds depending on its current state.

mrmaximas, 2025-03-25:
@Koenkk This doesn't seem to work for _TZ3210_ngqk6jia. Tomorrow I will try to remove such a dimmer from a working network, connect it to a test hub and sniff traffic.

mrmaximas, 2025-03-25:
The _TZ3210_weaqkhab countdown timer value is reported to:
```
0x4001 > 16385 > "onTime"
0x4002 > 16386 > "offWaitTime"
```
depending on the current state of the device - On or Off

mrmaximas, 2025-03-26:
Hi @Koenkk 
This is pcap with _TZ3210_ngqk6jia ( https://www.zigbee2mqtt.io/devices/QS-Zigbee-D02-TRIAC-LN_1.html ) and my tuya hub.
https://drive.google.com/file/d/1Q6CG6dbg-4OCDAdq1MeVnSg-HDmKyTzg/view?usp=share_link
Pcap includes interview with device. No other active devices in the network.
The countdown command is the same `0xf0`, but the further dialogue between the coordinator and the device is slightly different. Please check it.
Packets of interest:
530 - i set countdown for 300s then some dialogue. I assume that because of this the current version of the converter does not work on the _TZ3210_ngqk6jia
703 - report about remaining count 180s
857 - report about remaining count 180s
953 - report about remaining count 60s
1048 - end of countdown then 1053 off
1154 -  i set countdown for 1800s then some dialogue

mrmaximas, 2025-03-27:
packet 530 from hub to _TZ3210_ngqk6jia ??
[image]

mrmaximas, 2025-03-27:
report from _TZ3210_ngqk6jia about remaining time
[image]

mrmaximas, 2025-03-27:
I don't fully understand the proper way to look at command sequences.

Koenkk, 2025-03-28:
I added countdown support to https://gist.github.com/Koenkk/b402cb48a8f88915c92b1d1fa8252db7

mrmaximas, 2025-03-28:
@Koenkk it's works on _TZ3210_weaqkhab 🙏🏻 Will try to find the reason why it doesn't work on _TZ3210_ngqk6jia or will we take a break?

mrmaximas, 2025-03-28:
She State tab does not contain the current state of Countdown while it is running.
{
    "brightness": 141,
    "last_seen": "2025-03-28T23:16:17+03:00",
    "linkquality": 134,
    "max_brightness": 254,
    "min_brightness": 80,
    "state": "OFF",
    "update": {
        "installed_version": 70,
        "latest_version": 70,
        "state": "idle"
    }
}

Koenkk, 2025-03-30:
I'm not sure if the device reports anything when counting down (otherwise the number would be static, not really useful). Could you provide the debug log when settings countdown to 60 seconds and wait 2 minutes?

mrmaximas, 2025-03-30:
@Koenkk debug from z2m or from classic tuya hub?

Koenkk, 2025-03-30:
From z2m

mrmaximas, 2025-03-30:
how do I separate the records we need?

Koenkk, 2025-03-30:
You cannot, the full log is fine, it will only be 2 mins of log anyway

mrmaximas, 2025-03-30:
ready. here I first set the countdown to 60 seconds from ON to OFF, then 120 from OFF to ON.
Friendly name Kitchen Table
IEEE Address 0xa4c1385c28385d70
Network address 0x2C58 / 11352
[log.log.zip](https://github.com/user-attachments/files/19527336/log.log.zip)

mrmaximas, 2025-03-30:
Here is attempt of 60s countdown without success on:
Friendly name Bedroom Main
Zigbee Model TS110E
Zigbee Manufacturer _TZ3210_ngqk6jia
IEEE Address 0xa4c138243e41c8dd
Network address 0x21CA / 8650

[log.log.zip](https://github.com/user-attachments/files/19527385/log.log.zip)

mrmaximas, 2025-03-30:
@Koenkk everything seems to work on _TZ3210_ngqk6jia as well. My wife just saw the bedroom light on and asked me why I had it on ;)

mrmaximas, 2025-03-31:
Morning update about: _TZ3210_ngqk6jia 
The command is not always executed, I can't figure out why yet.

mrmaximas, 2025-03-31:
@Koenkk after hours of investigation, I can point out that:
_TZ3210_weaqkhab countdown works with any value like 1, 3, 5, 11, 30, 111 seconds etc, i.e. a multiple of 1 seconds.
_TZ3210_ngqk6jia countdown works only with value like 30, 60, 90, 120, 150, 180, 210 seconds etc, i.e. a multiple of 30 seconds.
Thanks!!!

Koenkk, 2025-03-31:
Added!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

closed 2025-03-31 by Koenkk