Koenkk/zigbee2mqtt #26794: Moes BHT-002/BHT-006 (_TZE204_aoclfnxz) Does Not Support 0.5°C Steps in UI

Lopsti-sys, 2025-03-18:
### What happened?

My Moes BHT-002/BHT-006 thermostat (recognized as `_TZE204_aoclfnxz`) should support 0.5°C steps according to its specifications. However, in the Zigbee2MQTT web UI, I can only adjust the temperature in full 1°C steps. When Home Assistant tries to send a value like 20.5°C, the thermostat resets the setpoint to 0.

### What did you expect to happen?

- The Zigbee2MQTT UI should allow 0.5°C steps.
- The thermostat should accept values like 20.5°C without resetting to 0.

### How to reproduce it (minimal and precise)

1. Set a temperature like 20.5°C for `climate.thermostat_wohnzimmer` in Home Assistant.
2. The thermostat resets the setpoint to 0.
3. In the Zigbee2MQTT UI, only 1°C steps are available.

### Zigbee2MQTT version

2.1.3

### Adapter firmware version

2025.3.3

### Adapter

Conbee II 

### Setup

Add-ON Lenovo ThinkCenter

Lopsti-sys, 2025-03-18:
Currently, I’m rounding values to full degrees (`round(0)`) in Home Assistant, which works but sacrifices the 0.5°C precision.

brvn0, 2025-05-06:
It doesnt support setting them manually via mqtt too, I tried doing that with an mqtt client and only integer values are commited succesful to the devices data and float numbers will be converted into crazy values like 

```
{
  "child_lock": "UNLOCK",
  "current_heating_setpoint": 23,
  "deadzone_temperature": 0,
  "heat": "OFF",
  "linkquality": 184,
  "local_temperature": 21,
  "local_temperature_calibration": -15833496,
  "max_temperature": 28,
  "max_temperature_limit": 28,
  "min_temperature_limit": 5,
  "preset": "hold",
  "preset_mode": "hold",
  "program": {
    "saturday_p1_hour": 6,
    "saturday_p1_minute": 0,
    "saturday_p1_temperature": 19,
    "saturday_p2_hour": 8,
    "saturday_p2_minute": 0,
    "saturday_p2_temperature": 19,
    "saturday_p3_hour": 11,
    "saturday_p3_minute": 30,
    "saturday_p3_temperature": 19,
    "saturday_p4_hour": 18,
    "saturday_p4_minute": 0,
    "saturday_p4_temperature": 16,
    "sunday_p1_hour": 6,
    "sunday_p1_minute": 0,
    "sunday_p1_temperature": 19,
    "sunday_p2_hour": 8,
    "sunday_p2_minute": 0,
    "sunday_p2_temperature": 19,
    "sunday_p3_hour": 11,
    "sunday_p3_minute": 30,
    "sunday_p3_temperature": 19,
    "sunday_p4_hour": 18,
    "sunday_p4_minute": 0,
    "sunday_p4_temperature": 16,
    "weekdays_p1_hour": 6,
    "weekdays_p1_minute": 0,
    "weekdays_p1_temperature": 19,
    "weekdays_p2_hour": 7,
    "weekdays_p2_minute": 30,
    "weekdays_p2_temperature": 19,
    "weekdays_p3_hour": 13,
    "weekdays_p3_minute": 0,
    "weekdays_p3_temperature": 19,
    "weekdays_p4_hour": 18,
    "weekdays_p4_minute": 0,
    "weekdays_p4_temperature": 16
  },
  "running_state": "idle",
  "sensor": "IN",
  "system_mode": "off"
}
```

with the following set command to zigbee2mqtt/NAME/set
`{"local_temperature_calibration": -1.6}`

it doesnt seem like 0.1 steps are supported like suggested in the readme on the wiki website

schrej-zz, 2025-06-27:
I have the same issues. Entering non integer values results in crazy values. Additionally, it is tricky, to enter negative values. You cannot enter them directly into the field, als a minus ("-") is not supported as input. You hve to use the slider to generate a negative value with the right integer value and random number behind the dot and then manually delete the non-integer numbers including the dot.

schrej-zz, 2025-08-30:
Hi,
I have the same issues with _TZE204_ltwbm23f

closed 2025-11-06 by github-actions[bot]