Koenkk/zigbee2mqtt #26796: Migrating from Sonoff Dongle to SLZB-06

ADawoud83, 2025-03-18:
### What happened?

I have a running zigbee2mqtt instance by Sonoff dongle E and am trying to migrate to SLZB-06 without repiaring 

Step I followed:
1- Update the SLZB06 to the latest 
2- Flashed the IEEEE of the old one to the new one 
3- I copied the configuration to the Z2M with the new TCP IP 
Unfortunately, it keeps failing 

I also tried to edit the yaml file with the new serial, but it didn't help.
I followed the instructions on SLZB06 to shut off the routers for more than 15 seconds, but it didn't work; I got the same error.

[2025-03-19 00:11:42] error: 	z2m: Error while starting zigbee-herdsman
[2025-03-19 00:11:42] error: 	z2m: Failed to start zigbee-herdsman
[2025-03-19 00:11:42] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-03-19 00:11:42] error: 	z2m: Exiting...
[2025-03-19 00:11:42] error: 	z2m: Error: network commissioning timed out - most likely network with the same panId or extendedPanId already exists nearby (Error: AREQ - ZDO - stateChangeInd after 60000ms
    at Object.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/utils/waitress.ts:67:23)
    at ZnpAdapterManager.beginCommissioning (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:372:31)
    at processTicksAndRejections (node:internal/process/task_queues:105:5)
    at ZnpAdapterManager.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:91:21)
    at ZStackAdapter.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:158:16)
    at Controller.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/controller/controller.ts:136:29)
    at Zigbee.start (/app/lib/zigbee.ts:69:27)
    at Controller.start (/app/lib/controller.ts:142:13)
    at start (/app/index.js:161:5))
    at ZnpAdapterManager.beginCommissioning (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:374:23)
    at ZnpAdapterManager.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:91:21)
    at ZStackAdapter.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:158:16)
    at Controller.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/controller/controller.ts:136:29)
    at Zigbee.start (/app/lib/zigbee.ts:69:27)
    at Controller.start (/app/lib/controller.ts:142:13)
    at start (/app/index.js:161:5)

### What did you expect to happen?

Successful Migration 

### Zigbee2MQTT version

2.1.3-1

### Adapter firmware version

v2.8.2.dev3

### Adapter

SLZB-06

### Setup

HAOS, NUC x86-64

jdiazbb, 2025-03-22:
I have the same problem here, how you solve it?

ADawoud83, 2025-03-22:
> I have the same problem here, what's your solution?

Unfortunately, I couldn't, I have defined the new dongle as a separate instance, and then I started to pair one by one 😢.

jdiazbb, 2025-03-22:
> > I have the same problem here, what's your solution?
> 
> Unfortunately, I couldn't, I have defined the new dongle as a separate instance, and then I started to pair one by one 😢.

I have done the same... Added 33 devices...