Koenkk/zigbee2mqtt #26798: ZBOSS device unable to interview

Mowglli, 2025-03-19:
### What happened?

I have trouble interviewing new devices. It just hangs at fail on timeout.

### What did you expect to happen?

I expected that it would succesfully interview the device. Like it does with example Sonoff ezDongle running ember.

### How to reproduce it (minimal and precise)

Configuration: 
`homeassistant:
  enabled: false
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://localhost
serial:
  port: COM11
  adapter: ember
  baudrate: 115200
advanced:
  channel: 15
  log_level: info
frontend:
  enabled: true
  port: 8080
version: 4
availability:
  enabled: true
  active:
    timeout: 10
  passive:
    timeout: 1500
permit_join: false
`

`
mosquitto_pub.exe -h localhost -t "zigbee2mqtt/bridge/request/permit_join" -m "{\"value\": true, \"time\": 254}"`

### Zigbee2MQTT version

master branch

### Adapter firmware version

2.2.5

### Adapter

NRF52840 Dongle

### Setup

Windows

### Debug log

[log.log](https://github.com/user-attachments/files/19334185/log.log)

kirovilya, 2025-03-19:
https://www.zigbee2mqtt.io/guide/configuration/logging.html#debugging
Please, turn on the debug logs and record the pairing process.

By the logs, the device does not report its endpoints:
```
[2025-03-18 16:12:52] info: 	zh:controller: Interview for '0xd0cf5efffe03927b' started
[2025-03-18 16:12:52] info: 	z2m: Starting interview of '0xd0cf5efffe03927b'
[2025-03-18 16:12:52] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xd0cf5efffe03927b","ieee_address":"0xd0cf5efffe03927b","status":"started"},"type":"device_interview"}'
[2025-03-18 16:13:13] error: 	zh:zboss:driv: ==> Error: Error: {"commandId":769,"tsn":27} after 10000ms
[2025-03-18 16:13:33] error: 	zh:controller: Interview failed for '0xd0cf5efffe03927b with error 'Error: Interview failed because can not get active endpoints ('0xd0cf5efffe03927b')'
[2025-03-18 16:13:33] error: 	z2m: Failed to interview '0xd0cf5efffe03927b', device has not successfully been paired
```

Perhaps the connection between the device and the coordinator is poor

Mowglli, 2025-03-19:
@kirovilya  , Sorry, i have updated the log with debug enabled. 
The device and dongle is approx. 40 cm from each other. 
It can also be found here:  [log.log](https://github.com/user-attachments/files/19336671/log.log)

kirovilya, 2025-03-19:
For some reason, the device descriptor could not be requested - timeout, the device did not respond.
```
[2025-03-19 09:45:32] debug: 	zh:zboss:uart: --> package type 6, flags 11{"isACK":true,"retransmit":false,"sequence":0,"ACKseq":1,"isFirst":false,"isLast":false}
[2025-03-19 09:45:32] debug: 	zh:zboss:uart: -->  ACK: dead05000611c0
[2025-03-19 09:45:32] debug: 	zh:zboss:uart: --> DATA (2,1,0): dead05000611c0
[2025-03-19 09:45:32] debug: 	zh:zboss:uart: --> [dead05000611c0]
[2025-03-19 09:45:32] debug: 	zh:zboss:uart: <-- FRAME: 00010402180585
[2025-03-19 09:45:32] debug: 	zh:zboss:driv: <== Frame: {"version":0,"type":1,"commandId":516,"tsn":24,"payload":{"category":5,"zdoClusterId":32770,"zdo":[133,null]}}
[2025-03-19 09:45:32] debug: 	zh:zboss: processMessage: {"version":0,"type":1,"commandId":516,"tsn":24,"payload":{"category":5,"zdoClusterId":32770,"zdo":[133,null]}}
[2025-03-19 09:45:32] debug: 	zh:controller: Received ZDO response: clusterId=NODE_DESCRIPTOR_RESPONSE, status=TIMEOUT, payload=undefined
```

check that there is no interference nearby, for example: USB3 devices, microwaves, Wi-Fi routers. maybe it is worth putting the coordinator on a USB extension cord.

Mowglli, 2025-03-25:
Hello i have been testing today, it didnt help with the suggestions. 
i have been using NCP host firmware from Nordic Semiconductor from 2.2.1 to 2.2.5, some crashes, some just never finishes interview

mrrdmr, 2025-04-10:
@Mowglli Could you try 3.0.0-p1 fw https://github.com/mrrdmr/nrf52840_zboss ?

Mowglli, 2025-04-10:
@mrrdmr How do you program the device with just the hex ? via NRF Connect Desktop ? - Do i need to include a special softdevice version ? :)

mrrdmr, 2025-04-10:
Done. 

[nrf52840dongle_nrf52840_zboss_3.0.0-p1.zip](https://github.com/user-attachments/files/19684952/nrf52840dongle_nrf52840_zboss_3.0.0-p1.zip)

Hex, Nordic MBR needed to boot.

kirovilya, 2025-04-10:
@Mowglli what device does not connect? where can I buy it to check?

Mowglli, 2025-04-11:
@kirovilya Im using a [Danalock V3 BTZB](https://danalock.com/products/danalock-v3-smart-lock?_gl=1*1doq9fs*_ga*MTE1ODY3NTUwMC4xNzQ0MzUzMjgw*_up*MQ..*_ga_DKYNTPZRT6*MTc0NDM1MzI3OS4xLjAuMTc0NDM1MzI3OS4wLjAuMA..). Im using a [NRF52840 Dongle ](https://dk.rs-online.com/web/p/wi-fi-adaptere/1769054?cm_mmc=DK-PLA-DS3A-_-google-_-CSS_DK_DK_Pmax_Test-_--_-1769054&matchtype=&&gclsrc=aw.ds&gad_source=1&gclid=Cj0KCQjw2N2_BhCAARIsAK4pEkXrcJBWl2wKXHDbLEpnLAC21sFmET5_F3ECpx_H1vqfYOOybZuJ5f8aAsU4EALw_wcB)

kirovilya, 2025-04-11:
> [@kirovilya](https://github.com/kirovilya) Im using a [Danalock V3 BTZB]

I see. It will be difficult for me to buy one like that.

Mowglli, 2025-04-11:
@kirovilya, I see that. 
I can try buying another zigbee-device and test that. And see if the issue persist.

closed 2025-06-19 by github-actions[bot]