Koenkk/zigbee2mqtt #26807: [New device support]: TOQCB2-80 2P _TZE284_mrffaamu Zigbee Smart Switch Measurement Circuit Breaker

LordRoge, 2025-03-20:
### Link

https://tongousmart.com/products/wi-fi-smart-mini-circuit-breaker-toqcb2-metering-1-63a-adjustable

### Database entry

I can't find it.

### Zigbee2MQTT version

2.1.3

### Comments

I can only find the device _TZE204_mrffaamu, but not the device _TZE284_mrffaamu. They are both Tongou's QCB2, maybe the chip has been updated.

### External definition

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE284_mrffaamu',
        },
    ],
        model: "TOQCB2-80",
        vendor: "Tongou",
        description: "Smart circuit breaker",
        fromZigbee: [tuya.fz.datapoints],
        toZigbee: [tuya.tz.datapoints],
        // Important: respondToMcuVersionResponse should be false otherwise there is an avalanche of commandMcuVersionResponse messages every second.
        // queryIntervalSeconds: is doing a pooling to update the device's parameters, now defined to update data every 3 minutes.
        onEvent: tuya.onEvent({
            respondToMcuVersionResponse: false,
            queryIntervalSeconds: 3 * 60,
        }),
        configure: tuya.configureMagicPacket,
        exposes: [
            tuya.exposes.switch(),
            e.energy(),
            e.power(),
            e.voltage(),
            e.current(),
            e.temperature(),
            tuya.exposes.voltageWithPhase("a"),
            tuya.exposes.voltageWithPhase("b"),
            tuya.exposes.voltageWithPhase("c"),
            tuya.exposes.powerWithPhase("a"),
            tuya.exposes.powerWithPhase("b"),
            tuya.exposes.powerWithPhase("c"),
            tuya.exposes.currentWithPhase("a"),
            tuya.exposes.currentWithPhase("b"),
            tuya.exposes.currentWithPhase("c"),
            e
                .enum("last_event", ea.STATE, [
                    "normal",
                    "trip_over_current",
[... 135 lines cut ...]
                        manual_off: tuya.enum(14),
                        value_15: tuya.enum(15),
                        value_16: tuya.enum(16),
                        factory_reset: tuya.enum(17),
                    }),
                ],
                [112, "automatic_closing", tuya.valueConverter.onOff],
                [113, "factory_reset", tuya.valueConverter.onOff],
                [114, "current_threshold", tuya.valueConverter.raw],
                [115, "over_voltage_threshold", tuya.valueConverter.raw],
                [116, "under_voltage_threshold", tuya.valueConverter.raw],
                [118, "temperature_threshold", tuya.valueConverter.divideBy10],
                [119, "over_power_threshold", tuya.valueConverter.raw],
                //[125, 'forward_electricity', tuya.valueConverter.divideBy100],
                [131, "temperature", tuya.valueConverter.divideBy10],
            ],
        },
    };

module.exports = definition;
```

### What does/doesn't work with the external definition?

This is a 2P device, but the 4P setting is also usable.

Koenkk, 2025-03-20:
Added!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

closed 2025-03-20 by Koenkk

luismipr, 2025-04-01:
Great, it works now, but it recognizes it as three-phase even though it is single-phase.

[image]

[image]

[image]

Is there any way to keep it in single phase, the total consumption will take into account the 3 phases and having only one, it will not give wrong measurements?

Thank you.

LordRoge, 2025-04-11:
fingerprint: tuya.fingerprint("TS0601", ["_TZE204_q22avxbv", "_TZE204_mrffaamu", "_TZE204_tzreobvu", "_TZE284_mrffaamu"]),
        model: "TOQCB2-80",
        vendor: "Tongou",
        description: "Smart circuit breaker",
        fromZigbee: [tuya.fz.datapoints],
        toZigbee: [tuya.tz.datapoints],
        // Important: respondToMcuVersionResponse should be false otherwise there is an avalanche of commandMcuVersionResponse messages every second.
        // queryIntervalSeconds: is doing a pooling to update the device's parameters, now defined to update data every 3 minutes.
        onEvent: tuya.onEvent({
            respondToMcuVersionResponse: false,
            queryIntervalSeconds: 3 * 60,
        }),
Mine is also a 1P device, but I noticed in the code that all QCB2 devices are defined under a definition. I'm not entirely sure whether it's possible to write them separately as multi-P configurations.