Koenkk/zigbee2mqtt #26809: State of cover QS-Zigbee-C03 remains open after closing

Andifront7, 2025-03-20:
### What happened?

When closing the cover the state remains in OPEN. The logs of Zigbee2MQTT show that the last event proclaims position = 0 and state = CLOSE.
The position attribute seems to be processed, the state not (see screenshots).

[image]
[image]
[image]

### What did you expect to happen?

The state of the cover should change from OPEN to CLOSE after closing (position = 0).

### How to reproduce it (minimal and precise)

closing cover

### Zigbee2MQTT version

2.1.3

### Adapter firmware version

7.4.4 [GA]

### Adapter

EmberZNet

### Setup

Add-on on Home Assistant OS

Andifront7, 2025-05-20:
This bug still exists.

Koenkk, 2025-05-20:
Could you provide the debug log of this?

See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

KG3RK3N, 2025-06-12:
I've the same issue and share the logs the next days.

AndiLuft, 2025-09-18:
Hi @Koenkk 
here are the logs of two shutters, where one changed it's state to close (Badezimmer) and the other not (Schlafzimmer).
In the logs I see no difference between the two shutters. Maybe the timing (last two changes happen at the same time) is the problem. But it seems to be not a specific issue of QS-Zigbee-C03, since in Schlafzimmer I use TS130F_GIRIER.

[image]

[Badezimmer.log](https://github.com/user-attachments/files/22406926/Badezimmer.log)
[Schlafzimmer.log](https://github.com/user-attachments/files/22406927/Schlafzimmer.log)