Koenkk/zigbee2mqtt #26813: Zigbee2mqtt docker ota function not using proxy setting?

jasmir, 2025-03-21:
### What happened?

Starting the container:
```
docker run -d --name zigbee2mqtt --privileged --restart=unless-stopped -e TZ=Europe/Berlin \
-e HTTP_PROXY='http://myproxy.local:8000' -e HTTPS_PROXY='http://myproxy.local:8000' \
-e http_proxy='http://myproxy.local:8000' -e https_proxy='http://myproxy.local:8000' \
-p 8080:8080 -v '/data/zigbee2mqtt:/app/data' --network=host koenkk/zigbee2mqtt:latest
```
Going to the Webinterface -> OTA -> on a Device there "check for updates"
Got:
`error 2025-03-21 08:58:43z2m: Failed to check if update available for 'Rauchmelder_Wohnzimmer' (fetch failed)`

TCP-Dump on the Host shows, that there are outgoing packets to 185.199.108.133(github) directly at this exactly time, instead of using the proxy

### What did you expect to happen?

I expect that the OTA function will use the proxy from that variables.
-or-
That there will be config items in the configuration.yaml for that purpose. But I didn't find any of these in the docs.

### How to reproduce it (minimal and precise)

See docker-run, above

### Zigbee2MQTT version

2.1.3 commit: ba337bd329aeb4ca17735c0cf09b31293c8cff06

### Adapter firmware version

7.4.5 [GA]

### Adapter

SLZB-06M

### Setup

As docker container koenkk/zigbee2mqtt:latest

jasmir, 2025-04-02:
Retested with 2.2.0 -> same problem

jasmir, 2025-05-01:
Retested with 2.3.0 commit: aec59cb774aa09a793ae41cd60e6a5cef15b293c, same problem.
Is there a way to work around this? Or do I use the variables wrong?

jasmir, 2025-11-02:
Tested with 2.6.3, commit: 6e001b48ec278c6c141579c9eacc045f190fc289 -> same problem

mmlion, 2026-01-17:
+1

The same problem. None of these works

      - HTTP_PROXY=http://http-bridge:8118
      - HTTPS_PROXY=http://http-bridge:8118
      - http_proxy=http://http-bridge:8118
      - https_proxy=http://http-bridge:8118
      - NPM_CONFIG_PROXY=http://http-bridge:8118
      - NPM_CONFIG_HTTPS_PROXY=http://http-bridge:8118
      - NO_PROXY=localhost,127.0.0.1,192.168.1.0/24,172.16.0.0/12
      - no_proxy=localhost,127.0.0.1,192.168.1.0/24,172.16.0.0/12
      - NPM_CONFIG_NO_PROXY=localhost,127.0.0.1,192.168.1.0/24,172.16.0.0/12