Koenkk/zigbee2mqtt #26817: z2m: Failed to call 'Bridge' 'start' - conbee ii adapter on HAOS

sjsmarts, 2025-03-21:
### What happened?

I have a conbee ii that is older and I am using to rebuild my HomeAssistant zigbee integration.  When I try to use Z2M I get the following error message:

error: 	z2m: Failed to call 'Bridge' 'start' (TypeError: Cannot read properties of undefined (reading 'getDevice')

This is followed by error details and then:

z2m: Failed to call 'HomeAssistant' 'start' (TypeError: Cannot read properties of undefined (reading 'deviceIeeeAddress')

Immediately after that I get these info logs:

info: 	z2m: Zigbee2MQTT started!
[2025-03-20 22:46:36] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"online"}'

When I go to the WebUI I see no devices, and the problem is generall identical to the issue posted here: https://github.com/Koenkk/zigbee2mqtt/issues/25717

### What did you expect to happen?

I expected Z2m to start

### How to reproduce it (minimal and precise)

When I enable debug logs it seems to fail at an earlier point in the cycle. In the logs below, I have ignored the Conbee ii ZHA discovery notifcation, uninstalled z2m, deleted the local files, reinstalled z2m and accepted default configurations.  I then stop z2m, enabled debug logging, and start z2m again:
~~~~~~~~~start without debug logs enabled~~~~~~~~~~~~~~~~~~~~~
[22:46:32] INFO: Preparing to start...
[22:46:32] INFO: Socat not enabled
[22:46:33] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2025-03-20 22:46:33] info: 	z2m: Logging to console, file (filename: log.log)
[2025-03-20 22:46:33] info: 	z2m: Starting Zigbee2MQTT version 2.1.3 (commit #unknown)
[2025-03-20 22:46:33] info: 	z2m: Starting zigbee-herdsman (3.2.7)
[2025-03-20 22:46:34] info: 	z2m: zigbee-herdsman started (resumed)
[2025-03-20 22:46:34] info: 	z2m: Coordinator firmware version: '{"meta":{"maintrel":0,"majorrel":38,"minorrel":88,"product":0,"revision":"0x26580700","transportrev":0},"type":"ConBee2/RaspBee2"}'
[2025-03-20 22:46:34] info: 	z2m: Currently 0 devices are joined.
[2025-03-20 22:46:34] info: 	z2m: Connecting to MQTT server at mqtt://core-mosquitto:1883
[2025-03-20 22:46:34] info: 	z2m: Connected to MQTT server
[2025-03-20 22:46:34] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"online"}'
[2025-03-20 22:46:34] error: 	z2m: Failed to call 'Bridge' 'start' (TypeError: Cannot read properties of undefined (reading 'getDevice')
    at Bridge.publishInfo (/app/lib/extension/bridge.ts:692:69)
    at Bridge.start (/app/lib/extension/bridge.ts:179:9)
    at Controller.callExtensions (/app/lib/controller.ts:368:17)
    at Controller.start (/app/lib/controller.ts:186:9)
    at start (/app/index.js:161:5))
[2025-03-20 22:46:34] info: 	z2m: Started frontend on port 8099
[2025-03-20 22:46:34] error: 	z2m: Failed to call 'HomeAssistant' 'start' (TypeError: Cannot read properties of undefined (reading 'deviceIeeeAddress')
    at HomeAssistant.getBridgeEntity (/app/lib/extension/homeassistant.ts:1943:78)
    at HomeAssistant.start (/app/lib/extension/homeassistant.ts:440:28)
    at Controller.callExtensions (/app/lib/controller.ts:368:17)
    at Controller.start (/app/lib/controller.ts:186:9)
    at start (/app/index.js:161:5))
[2025-03-20 22:46:34] info: 	z2m: Zigbee2MQTT started!
[2025-03-20 22:46:36] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"online"}'
~~~~~~~~~stop without debug logs enabled~~~~~~~~~~~~~~~~~~~~~
[2025-03-20 22:46:53] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"offline"}'
[2025-03-20 22:46:53] info: 	z2m: Disconnecting from MQTT server
[2025-03-20 22:46:53] info: 	z2m: Stopping zigbee-herdsman...
[2025-03-20 22:46:54] info: 	z2m: Stopped zigbee-herdsman
[2025-03-20 22:46:54] info: 	z2m: Stopped Zigbee2MQTT
~~~~~~~~~debug logs enabled~~~~~~~~~~~~~~~~~~~~~
~~~~~~~~~start with debug logs enabled~~~~~~~~~~~~~~~~~~~~~
[22:47:01] INFO: Preparing to start...
[22:47:01] INFO: Socat not enabled
[22:47:01] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
/app/node_modules/.pnpm/js-yaml@4.1.0/node_modules/js-yaml/lib/loader.js:183
  return new YAMLException(message, mark);
         ^
YAMLException: duplicated mapping key (43:1)
 40 | frontend:
[... 116 lines cut ...]
    at read (/app/lib/util/settings.ts:279:20)
[22:47:14] INFO: Preparing to start...
[22:47:15] INFO: Socat not enabled
[22:47:15] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
/app/node_modules/.pnpm/js-yaml@4.1.0/node_modules/js-yaml/lib/loader.js:183
  return new YAMLException(message, mark);
         ^
YAMLException: duplicated mapping key (43:1)
 40 | frontend:
 41 |   enabled: true
 42 |   port: 8099
 43 | advanced:
------^
 44 |     log_level: debug
    at generateError (/app/node_modules/.pnpm/js-yaml@4.1.0/node_modules/js-yaml/lib/loader.js:183:10)
    at throwError (/app/node_modules/.pnpm/js-yaml@4.1.0/node_modules/js-yaml/lib/loader.js:187:9)
    at storeMappingPair (/app/node_modules/.pnpm/js-yaml@4.1.0/node_modules/js-yaml/lib/loader.js:358:7)
    at readBlockMapping (/app/node_modules/.pnpm/js-yaml@4.1.0/node_modules/js-yaml/lib/loader.js:1173:9)
    at composeNode (/app/node_modules/.pnpm/js-yaml@4.1.0/node_modules/js-yaml/lib/loader.js:1441:12)
    at readDocument (/app/node_modules/.pnpm/js-yaml@4.1.0/node_modules/js-yaml/lib/loader.js:1625:3)
    at loadDocuments (/app/node_modules/.pnpm/js-yaml@4.1.0/node_modules/js-yaml/lib/loader.js:1688:5)
    at Object.load (/app/node_modules/.pnpm/js-yaml@4.1.0/node_modules/js-yaml/lib/loader.js:1714:19)
    at Object.read (/app/lib/util/yaml.ts:23:29)
    at read (/app/lib/util/settings.ts:279:20)

sjsmarts, 2025-03-21:
oh I see what I got the yaml error :) one sec will update with the correct logs...

sjsmarts, 2025-03-21:
also just to verify, the adapter and current devices are currently working using the ZHA integration...

Koenkk, 2025-03-21:
Could you try to:
- Stop z2m
- Enable [debug logging](https://www.zigbee2mqtt.io/guide/configuration/logging.html)
- Remove the `data/database.db`
- Start z2m
- Provide the debug logging here

anthropo, 2025-03-27:
Hello, same here, CONBEE II with domoticz and Z2M.

The strange thing is that if i start z2m manually it never stops, and if i start 
systemctl start zigbee2mqtt.service it stops very often

the debug mode says  the same error as yhis ticket, but also : 
[2025-03-27 14:10:12] debug:     zh:deconz: Permit joining on coordinator for 0 sec.
[2025-03-27 14:10:12] debug:     zh:controller:database: Writing database to '/opt/zigbee2mqtt/data/database.db'
[2025-03-27 14:10:12] debug:     zh:deconz:driver: DATA_REQUEST - destAddr: 0xfffc EP:0 SeqNr. 49 request id: 7
[2025-03-27 14:10:12] info:     z2m: Stopped zigbee-herdsman
[2025-03-27 14:10:12] info:     z2m: Stopped Zigbee2MQTT

It stopped z2m, enabled debug logging and removed the data/database.db and finally restarted  z2m

Here are the logs 

[zigbee2mqtt.log](https://github.com/user-attachments/files/19487215/zigbee2mqtt.log)

Thank you

markg85, 2025-03-29:
Same issue here:
```
[2025-03-29 23:03:40] info: 	zh:adapter:discovery: Matched adapter: {"path":"/dev/ttyUSB1","manufacturer":"ITead","serialNumber":"52087ffe414fef11a76b46b3174bec31","pnpId":"usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_52087ffe414fef11a76b46b3174bec31-if00-port0","vendorId":"10c4","productId":"ea60"} => zstack: path=/dev/ttyUSB1, score=4
[2025-03-29 23:03:40] info: 	zh:zstack:znp: Opening SerialPort with {"path":"/dev/ttyUSB1","baudRate":115200,"rtscts":false,"autoOpen":false}
[2025-03-29 23:03:40] info: 	zh:zstack:znp: Serialport opened
[2025-03-29 23:03:43] info: 	z2m: zigbee-herdsman started (resumed)
[2025-03-29 23:03:43] info: 	z2m: Coordinator firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20210708,"transportrev":2},"type":"zStack3x0"}'
[2025-03-29 23:03:43] info: 	z2m: Currently 0 devices are joined.
[2025-03-29 23:03:43] info: 	z2m: Connecting to MQTT server at mqtt://10.0.3.96:1883
[2025-03-29 23:03:43] info: 	z2m: Connected to MQTT server
[2025-03-29 23:03:43] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"online"}'
[2025-03-29 23:03:43] error: 	z2m: Failed to call 'Bridge' 'start' (TypeError: Cannot read properties of undefined (reading 'getDevice')
    at Bridge.publishInfo (/app/lib/extension/bridge.ts:692:69)
    at Bridge.start (/app/lib/extension/bridge.ts:179:9)
    at Controller.callExtensions (/app/lib/controller.ts:368:17)
    at Controller.start (/app/lib/controller.ts:186:9)
    at start (/app/index.js:161:5))
[2025-03-29 23:03:43] info: 	z2m: Started frontend on port 0.0.0.0:8080
[2025-03-29 23:03:43] info: 	z2m: Zigbee2MQTT started!
[2025-03-29 23:03:45] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"online"}'
```

The zigbee dongle itself works fine under ZHA though i was hoping for zigbee2mqtt to be of more use to me.
What might be of concern is that i'm using this in docker and i disable homeassistant integration (as that is another error in the code.. sigh).

Is there any info you'd like me to share to help debugging this?

Koenkk, 2025-03-31:
Should be fixed now, to apply the fix stop z2m, remove the `Coordinator` line from the `data/database.db` and start z2m.

sjsmarts, 2025-03-31:
Apologies for the delay, confirmed this is fixed for me. Stopping z2m, deleting the DB, and then restarting did the trick!