Koenkk/zigbee2mqtt #26818: Dimmers not updating ANYTHING (status, dimming level, etc) when physical button is pushed

greekgremlin, 2025-03-21:
### What happened?

I have 15 Sunricher  "SR-ZG9101CS" and 3 MiBoxer M-LIght "PZ2" dimmers.
All were working fine with the Philips Hue (zigbee) controller.
I decided to move all dimmers to "Home Assistant Yellow" in which zigbee2mqtt is running.
The dimmers were discovered with different model names.
The Sunricher were added as "Sunricher SRP-ZG9105-CC" (same manufacturer but different model) and the MiBoxer as "Tuya TS0502B" (different manufacturer and model).
All dimmers work fine as long as you instruct them to do anything from WITHIN Home Assistant. If however you press their corresponding button (to turn on/off or to dim up/down) NO feedback is returned to zigbee2mqtt (no change in power state or dimming level).
I also have some NOUS relays which work perfectly well (pressing their corresponding buttons returns their current state fine)

### Zigbee2MQTT version

2.1.3 

### Adapter firmware version

ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20230830183545-if00

### Adapter

ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20230830183545-if00

### Setup

Add on in a "Home Assistant Yellow"

chris-1243, 2025-03-21:
You have to check the reporting section of your devices and set values according to your needs

greekgremlin, 2025-03-21:
I dont understand your answer. Can you please rephrase?

chris-1243, 2025-03-21:
For each device, you have the possibility to set the reporting

Open your device tab and then reporting.
There, you may set value according to your need if the device accepts.

Below an example...

[image]

greekgremlin, 2025-03-21:
Thanks for your feedback and try to help me.
I am struggling however to find the relation between your suggestion and the problem I am facing.
How does this corelate with the lack of feedback from the dimmers to zigbee2mqtt when pressing the PHYSICAL button to turn them on or continuously press them in order to change their brightness?
Like I previously posted, the NOUS B1Z switches work fine (when pressing the physical button the change of state is reported) and NOTHING was needed from my side in order to make them work (they were added by me in Home Assistant and I didnt have to user "reporting" which I didnt know that it exists up to now that you mentioned it).

chris-1243, 2025-03-21:
First of all and take into consideration, many devices added are done by the community. This means, some features might have been ommited or the configuration after pairing has been done according to the need of a user and not yours.

I have understood some NOUS devices are working as you expect. Other not, unfortunatelly.
This is why you have to set some parameters by yourself. Not everything is done automatically.

How would you a device to report anything if you do not instruct to do so? So please, try to consider what I wrote previously.
If you wish a device to report its brightness level after pressing a long time on your physical switch, you have to tell it. You have no entry in your log as it seems the device has not been instructed to report its latest state.

Go to the reporting tab and manually configure what you need by selecting the endpoint, cluster, attribute,...

greekgremlin, 2025-03-21:
Ahh, now it makes more sense.
I wasn't aware that development wise one must actually instruct the device to report back something.

When going to "reporting" I get "1" and "242" in the "endpoints" dropdown for me to select from.

Selecting "1"-> "select cluster": ota, genBasic, Identify, Groups, Scenes, OnOff, LevelCtrl, 65535, haDiagnostic, Touchlink
Selecting  "242"-> "select cluster": greenpower or none
 
Making a choice in "Cluster" changes the "Attribute" dropdown accordingly.

I have no clue what to select/do as I dont understand how any of those relates to a "physical button". Maybe it is the  "OnOff" cluster? But then what?
Is there some literature/documentation where I can read and understand what I should do?
I am clueless.....

greekgremlin, 2025-03-21:
And so I just realized that the root cause of the problem is the wrong recognition of my dimmers!
Like I mentioned in my 1st post my dimmers:

1) Sunricher "SR-ZG9101CS" dimmer has been recognized as Sunricher model  "SRP-ZG9105-CC" 
2) MiBoxer M-LIght "PZ2" dimmer has been recongized as  "Tuya TS0502B" 

I did a search for Sunricher "SRP-ZG9105-CC" and guess what I noticed: this is a 220v AC dimmer WITHOUT an ON/OFF external toggle button. My "SR-ZG9101CS" model is DC and supports a toggle ON/OFF external switch. Therefore zigbee2mqtt is technically "blind" as it knows that this dimmer does NOT have any external input.
I didnt search for MiBoxer but I believe its gonna be similar.

And so, going back to my previous question: how can you add a missing functionality? 
I suppose either through "reporting" OR by the kindness of a developer adding a new device to the zigbee2mqtt list of supported devices.

Please anyone who can help me with either solutions I will be grateful.

chris-1243, 2025-03-21:
Would you share your database.db or extract all the information regarding your Sunricher device and share it.

greekgremlin, 2025-03-22:
From database.db:

This is the  Sunricher "SR-ZG9101CS" dimmer which is wrongly recognized as "SRP-ZG9105-CC":

{"id":46,"type":"Router","ieeeAddr":"0x943469fffef31985","nwkAddr":27618,"manufId":4644,"manufName":"Sunricher","powerSource":"Mains (single phase)","modelId":"HK-ZD-DIM-A","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[0,3,4,5,6,8,65535,2821,4096],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"appVersion":0,"stackVersion":0}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":102,"inClusterList":[33],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":0,"hwVersion":1,"dateCode":"NULL","swBuildId":"2.9.2_r72","zclVersion":3,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1742632021707}

And this is the MiBoxer M-LIght "PZ2" dimmer which is wrongly recognized as "Tuya TS0502B"

{"id":34,"type":"Router","ieeeAddr":"0xb43522fffe55bfc6","nwkAddr":51148,"manufId":4098,"manufName":"_TZB210_0bkzabht","powerSource":"Mains (single phase)","modelId":"TS0502B","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":268,"inClusterList":[0,3,4,5,6,4096,8,768,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\u0000\u0000\u0000\u0000\u0005","65506":31,"65508":0,"65534":0,"modelId":"TS0502B","manufacturerName":"_TZB210_0bkzabht","stackVersion":0,"dateCode":"","zclVersion":3,"appVersion":71,"powerSource":1}},"lightingColorCtrl":{"attributes":{"57344":270,"57600":3448924274688,"61453":1,"colorCapabilities":16,"tuyaRgbMode":0}},"genOnOff":{"attributes":{"onOff":0}},"genLevelCtrl":{"attributes":{"61440":803}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":71,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1742631943246}

Anything else I can provide let me know.

chris-1243, 2025-03-22:
I had a quick look for your Sunricher dimmer. We will have to try some external converter but it remains mainly a reporting problem

https://github.com/Koenkk/zigbee-herdsman-converters/blob/3dad96af89068aeb2cc5195564aaf9c37dc432fb/src/devices/sunricher.ts#L1106

For your second device, it recognized as a Tuya device which is correct as it is a Tuya model

greekgremlin, 2025-03-22:
Is there anything I can do from my side to help?

chris-1243, 2025-03-22:
You may try to configure the reporting for your Sunricher device

(On/Off)
endpoint: 1
cluster: OnOff
attribute: OnOff
min rep interval: 0
max rep interval: 3600
min rep change: 0

Click apply

(Brightness)
endpoint: 1
cluster: LevelCTRL
attribute: current_Level
min rep interval: 1
max rep interval: 0
min rep change: 10

greekgremlin, 2025-03-22:
I have good news!
It works!
I set both the on/Off and the Brightness as you suggested then asked someone to go and try pressing the button (the location is 10km away from me).
We did some testing over the phone (me at my place, him where the dimmers are) and from the looks of it now both the on/off and the brightness works.
We had twice a glitch (the feedback was not correct) but I am not sure whether it was his fault or not (he did me a favor going there and so I didnt want to waste more of his time).
I plan to go over there tomorrow so I do my own testing AND to also rewire the MiBoxer dimmers in order to try to same settings to them (currently the button cuts their power supply, I will have to change this behavior).

I cant thank you enough for this!!! 

What happens next? 
How will they be integrated in the project once I give you the green light that both dimmer models work 100% correct?

chris-1243, 2025-03-22:
As SR-ZG9101CS shares the same modelID as SRP-ZG9105-CC, it will be added as a white label.

For your Miboxer dimmer, I have not taken the time yet to check if it is already added or not.

As I told you previously, reporting is a matter how people wish to have their device to report any value or not

chris-1243, 2025-03-22:
The white label is mainly used to recognized easily a device without creating tons of new lines of code. The model will appear in the documentation as well with the same definition as the "main" model.
On the frontend, it depends how the converter is written. I am just surprised Sunricher share the same modelID for those two dimmers. Have you a picture of your SR-ZG9101CS ?

> In other words, once a new zigbee2mqtt gets released with my new model included (as white label), how will I be able to re-adapt the dimmers and discard my custom reporting settings?

When you update Zigbee2mqtt, your custom settings remain unless you remove the device from your network. You may also use your own converter with the risk when an Z2M update is released to have it broken...

greekgremlin, 2025-03-22:
- This is the SR-ZG9101CS as shown in the Sunricher website: [https://www.sunricher.com/push-compatible-constant-voltage-single-channel-zigbee-led-controller-sr-zg9101cs.html](url)

If and when my model is added in the supported devices us "white label' how will I be able to change all the dimmers to this white label model? (so I can delete the manual "reporting" settings for each unit and be able to add more dimmers with the correct "white label" model)

- I have 40 dimmers all in total. 18 have already been installed in Home Assistant. Yes, I just checked the 18 one by one and all have the same "HK-ZD-DIM-A" Zigbee Model and "SRP-ZG9105-CC" Model under Zigbee2MQTT.
What is is that you found in the converter?

chris-1243, 2025-03-22:
This comment surprised me...
https://github.com/Koenkk/zigbee-herdsman-converters/blob/2bb3bebf6ab72c359b9e486611cd8614162dabf0/src/devices/sunricher.ts#L1120

```javascript
const m = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
        zigbeeModel: ["HK-ZD-DIM-A"],
        model: "SR-ZG9101CS",
        vendor: "Sunricher",
        description: "Constant Current Zigbee LED dimmable driver",
        extend: [m.light({configureReporting: true})],
    meta: {},
};

module.exports = definition;
``` 

Would add those lines.

Create a folder in data called `external_converters`. Then, add a new file with `.js`as an extension and copy paste the code above. Restart Z2M and let me know the output. It should configure the reporting normally.

greekgremlin, 2025-03-22:
Amazing discovery! So it seems my dimmers have manufacturing defect and its not the 1st time that Sunricher has done this,,,

I tried to follow your instructions but I dont have a "data" folder under the "zigbee2mqtt"  (to cross check I read the webpage  [https://www.zigbee2mqtt.io/advanced/more/external_converters.html](url) )
Should I create a "data" folder and then in it a "external_converters"?

When you say "It should configure the reporting normally.", do you mean that I should delete from a dimmer the manual reporting I have added and check whether it now works fine WITHOUT the deleted reporting?

chris-1243, 2025-03-23:
> Amazing discovery! So it seems my dimmers have a manufacturing defect and its not the 1st time that Sunricher has done this,,,

It is just a supposition. There us dedinitely something strange between those 2 devices and there respective modelID.

In your "zigbee2mqtt" folder where you have all your data like database.db or comfiguration.yaml, create a sub-folder `external_converters`. Inside this sub-folder, you create a `*.js` file with my code.

Pair a new Sunricher dimmer and check the output in Z2M. Let us know.

greekgremlin, 2025-03-23:
Ok, subfolder "external_converters" created and new "sunricher_SR-ZG9101CS.js" with your suggested contents created as well.
Tomorrow morning I will go to the location, test and let you know!

chris-1243, 2025-03-23:
Check your logs after restart of Z2M to ensure there is no error thrown due to the converter

greekgremlin, 2025-03-23:
I restarted Home Assistant yellow and checked the zigbee2MQTT  log. The converter fails to load:

[2025-03-23 10:04:30] info: 	z2m: Connected to MQTT server
[2025-03-23 10:04:30] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"online"}'
[2025-03-23 10:04:30] error: 	z2m: Failed to load external converter 'sunricher_SR-ZG9101CS.js'
[2025-03-23 10:04:30] error: 	z2m: Check the code for syntax error and make sure it is up to date with the current Zigbee2MQTT version.
[2025-03-23 10:04:30] error: 	z2m: External converters are not meant for long term usage, but for local testing after which a pull request should be created to add out-of-the-box support for the device
[2025-03-23 10:04:30] error: 	z2m: Failed to call 'ExternalConverters' 'start' (TypeError: Cannot set properties of undefined (setting 'externalConverterName')
    at ExternalConverters.loadJS (/app/lib/extension/externalConverters.ts:48:49)
    at ExternalConverters.loadFiles (/app/lib/extension/externalJS.ts:156:24)
    at ExternalConverters.start (/app/lib/extension/externalJS.ts:45:20)
    at processTicksAndRejections (node:internal/process/task_queues:105:5)
    at Controller.callExtensions (/app/lib/controller.ts:368:17)
    at Controller.start (/app/lib/controller.ts:186:9)
    at start (/app/index.js:161:5))

This is what I have pasted in the "sunricher_SR-ZG9101CS.js'" file:

const m = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
        zigbeeModel: ["HK-ZD-DIM-A"],
        model: "SR-ZG9101CS",
        vendor: "Sunricher",
        description: "Constant Current Zigbee LED dimmable driver",
        extend: [m.light({configureReporting: true})],
    meta: {},
};

module.exports = definition;

greekgremlin, 2025-03-23:
yes, now it shows in the log (I renamed the filename):

[2025-03-23 15:24:49] info: 	z2m: Loaded external converter 'sunricher_SR-ZG9101CS.js'.

greekgremlin, 2025-03-24:
I have good news!

I added a new SR-ZG9101CS Sunricher and it worked when pressing the physical button for the on/off feedback and change of brightness. A minor issue (maybe its not an issue at all) is that the updating of brightness in the log happens when you actually release the button. While you keep on pressing it -and the brightness of the light changes- there is no change in the feedback (nothing written to log). Once you release then a log entry appears showing the final value of the brightness.

Also, the MiBoxer M-LIght "PZ2" also work! (I have added manual entries in the "reporting" settings).
Can you please provide an external converter for those as well?

I remind you:
MiBoxer M-LIght "PZ2" dimmers have been recognized as "Tuya TS0502B"

chris-1243, 2025-03-24:
> I added a new SR-ZG9101CS Sunricher and it worked when pressing the physical button for the on/off feedback and change of brightness. A minor issue (maybe its not an issue at all) is that the [...]

May you provide a screenshot of the reporting tab of one of this device? I just wish to see what values have been set.

> Also, the MiBoxer M-LIght "PZ2" also work! (I have added manual entries in the "reporting" settings).
Can you please provide an external converter for those as well?

> I remind you:
MiBoxer M-LIght "PZ2" dimmers have been recognized as "Tuya TS0502B"

It is in fact a Tuya device. Are those devices working as expected even if wrongly detected for the time (not the reporting of course)?

Let's finish the Sunricher problem and then, I will try to help for your Mi-boxer/Tuya device

chris-1243, 2025-03-24:
> What is it exactly you need?

I would like to see what values have been set by the external converter in the `reporting` tab. I suspect a default value set which might lead to your brightness not reporting until you release the physical button

greekgremlin, 2025-03-25:
I will visit the location tomorrow to do physical test by pressing the button.
However, because I have remote access here's what I just did:

1) opened 2 instances of home assistant
2) on instance #1 I opened a dimmer in the dashboard and was looking at its brightness slider
3) on instance #2 I opened the same dimmer and increased/decreased brightness
4) on instance #1 I observed the brightness change ONLY when I stopped playing with the brightness on instance #26350 

So, the behavior is the same: regardless of whether I press the physical button to change the brightness or I programmatically (via Home Assistant) change the brightness, the % brightness reported is changed ONLY after stopping playing with brightness.

Example: this is the ONLY message reported after I stop changing the brightness:

_2025-03-25 12:07:46z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Dim.-LivingQuarters- ceiling', payload '{"brightness":175,"effect":null,"linkquality":255,"power_on_behavior":"previous","state":"ON"}'_

And in "MQTT Explorer" while playing with brightness nothing is reported/changed. As soon as I stop adjusting the brightness I only see 1 "final" message with the latest brightness level:

_logging = {"level":"info","message":"z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Dim.-LivingQuarters- ceiling', payload '{\"brightness\":167,\"effect\":null,\"linkquality\":255,\"power_on_behavior\":\"previous\",\"state\":\"ON\"}'"}
Dim.-LivingQuarters- ceiling = {"brightness":167,"effect":null,"linkquality":255,"power_on_behavior":"previous","state":"ON"}
set = {"state":"ON","brightness":167}_

chris-1243, 2025-03-25:
I might have found a way to add you Sunricher device with ots own entry/model/picture in Z2M.👍

I will need your help and check the `software build` for all the SR-ZG9101CS you already pair. In your `database.db` it would look like `"swBuildId":"2.9.2_r72`.
If you find different software version for the same model, please provide them all

greekgremlin, 2025-03-25:
I checked all dimmers for the reported "Firmware version" under "Devices", "About"
I confirm that ALL have the same "2.9.2_r72"

chris-1243, 2025-03-26:
May you just confirm this is working and reporting works please

[SR-ZG9101CS.txt](https://github.com/user-attachments/files/19474657/SR-ZG9101CS.txt)

greekgremlin, 2025-03-26:
I see that in the attached file:

_const m = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    fingerprint: [{modelID: "HK-ZD-DIM-A", softwareBuildID: "2.9.2_r72"}],  **<-------- this has been added**
    //zigbeeModel: ["HK-ZD-DIM-A"],
    model: "SR-ZG9101CS",
    vendor: "Sunricher",
    description: "Constant Current Zigbee LED dimmable driver",
    extend: [m.light({configureReporting: true})],
    meta: {},    
};

module.exports = definition;_

I suppose you want me to:

1) change the previous external converter file with this new one
2) add a new dimmer and see whether it works

correct?

chris-1243, 2025-03-26:
Exactly👍

Before submitting a pull request, I would like to be sure it is fully working.

greekgremlin, 2025-03-26:
Ok, will do once I go there in the morning!

chris-1243, 2025-03-27:
Stop the add-on and restart it again. Provide the logs here.

I just had a try on a test network and the converter is loaded

```
zigbee2mqtt  | [2025-03-27 08:54:01] info:      z2m: Coordinator firmware version: '{"meta":{"build":397,"ezsp":14,"major":8,"minor":0,"patch":2,"revision":"8.0.2 [GA]","special":0,"type":170},"type":"EmberZNet"}'
zigbee2mqtt  | [2025-03-27 08:54:01] info:      z2m: Currently 0 devices are joined.
zigbee2mqtt  | [2025-03-27 08:54:01] info:      z2m: Connecting to MQTT server at mqtt://mosquitto:1883
zigbee2mqtt  | [2025-03-27 08:54:01] info:      z2m: Connected to MQTT server
zigbee2mqtt  | [2025-03-27 08:54:01] info:      z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"online"}'
zigbee2mqtt  | [2025-03-27 08:54:01] info:      z2m: Loaded external converter 'SR-ZG9101CS.js'.
```

greekgremlin, 2025-03-27:
My Home Assistant loves you.
No matter what I did, nada, no log entries.
As soon as I followed your instructions to do a plugin restart (instead of restarting the whole yellow HA):

1) Created new directory and logfile (was not happening before):

_> [2025-03-27 11:09:43] debug: 	z2m: Removing old log directory '/config/zigbee2mqtt/log/2025-03-21.08-19-49'
> [2025-03-27 11:09:43] info: 	z2m: Logging to console, file (filename: log.log)
> [2025-03-27 11:09:43] debug: 	z2m: Loaded state from file /config/zigbee2mqtt/state.json
> [2025-03-27 11:09:43] info: 	z2m: Starting Zigbee2MQTT version 2.1.3 (commit #unknown)
> [2025-03-27 11:09:43] debug: 	z2m: sd-notify loaded
> [2025-03-27 11:09:43] info: 	z2m: Starting zigbee-herdsman (3.2.7)_

2) Wrote finally the entries:

_> [2025-03-27 11:09:50] info: 	z2m: Connected to MQTT server
> [2025-03-27 11:09:50] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"online"}'
> [2025-03-27 11:09:50] info: 	z2m: Loaded external converter 'sunricher_SR-ZG9101CS.js'.

I think that only when you do a plugin restart it does the above. Performing a full restart never ends up with the above.
Well, back to square one: once I visit today I will let you know about the result!

greekgremlin, 2025-03-27:
So, I added a new "SR-ZG9101CS" dimmer.

"Devices", "About" reports this:

Zigbee Model: HK-ZD-DIM-A   (expected)
Model:    SR-ZG9101CS       (<--- correctly now recognized!)

In fact, what I noticed is that ALL previously SR-ZG9101CS paired dimmers are now reporting in the "About" the correct model

One thing however that is different is the "reporting".
These are the reporting settings of the newly added "SR-ZG9101CS":

_(On/Off)
endpoint: 1
cluster: OnOff
attribute: OnOff
min rep interval: 0
max rep interval: 65000
min rep change: 1

(Brightness)
endpoint: 1
cluster: LevelCTRL
attribute: current_Level
min rep interval: 10
max rep interval: 65000
min rep change: 1_

------------------------------
Whereas the following are the settings of the rest "SR-ZG9101CS" dimmers which has been applied by me manually as per your instructions. There is a HUGE difference regarding responsiveness between the newly added and the old ones. 
When I change the brightness -by pressing the physical button - on any old one, the message IMMEDIATELY shows (example: 2025-03-27 13:29:54z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xe0798dfffe762033', payload '{"brightness":75,"effect":null,"linkquality":216,"power_on_behavior":null,"state":"ON"}')
However, the newly added takes several seconds to return the same message when I change the brightness.
To confirm, I edited the "LevelCTRL" and changed the 10,65000,1  to 1,0,10.
Once I applied the new dimmer acted exactly like the old ones: reporting is now immediate as soon as a release the physical button (and brightness has changed).

_(On/Off)
endpoint: 1
cluster: OnOff
attribute: OnOff
min rep interval: 0
max rep interval: 3600
min rep change: 0

(Brightness)
endpoint: 1
cluster: LevelCTRL
attribute: current_Level
min rep interval: 1
max rep interval: 0
min rep change: 10_

Overall it has been a 99% success. Now the dimmers are correctly recognized. If you can also add those specific reporting settings (the ones you initially suggested) then we will have 100% success.

chris-1243, 2025-03-27:
This is due to the default values set in herdsman-converter.

You do not need to change all of them, only `min rep change interval` should do the trick. As per default, it is set to 10 seconds. If set, let's say to 5, it will be 5 seconds...

What I suggest is to create the PR and add already your device. It would be available in the next release in april. For the reporting, I need to dig a bit and try on my side how I may change it to be closer to your needs

greekgremlin, 2025-03-27:
Indeed, I forced removed the new dimmer, re-paired it and ONLY changed the "min rep interval" to "1".
Now the new dimmer behaves exactly like the old ones: immediate reporting as soon as I release the button.

_

> "What I suggest is to create the PR and add already your device. It would be available in the next release in april. For the reporting, I need to dig a bit and try on my side how I may change it to be closer to your needs"

_

What do you mean by "PR"? 
"Next release in April". I suppose my dimmer will be included in the next release, correct?
"dig a bit... to your needs". You mean that you will try to find a way to have a default of "min rep interval = 1" for my model by default?

Also, - if I am not pushing you in any way-, can we also do something with my MiBoxer dimmers as well?

chris-1243, 2025-03-27:
> What do you mean by "PR"?
"Next release in April". I suppose my dimmer will be included in the next release, correct?

Yes to have it fully supported. External converter is not made to be used in the long term

> "dig a bit... to your needs". You mean that you will try to find a way to have a default of "min rep interval = 1" for my model by default?

Like you, I need to learn and search for solution. It is not straight forward. (I wish I could...)

> Also, - if I am not pushing you in any way-, can we also do something with my MiBoxer dimmers as well?

As I told you, I wanted to finish with your first device, then I will see if I am able to help you with your second device.

chris-1243, 2025-03-28:
If everything goes well, your two devices should be supported in the april release.

greekgremlin, 2025-03-28:
Man, I cant thank you enough!!!

chris-1243, 2025-03-28:
> Irrelevant to all above:

error 2025-03-28 08:39:08zhc: Failed to apply calibration to 'temperature': 'temperature_calibration' is not a number, got string ()
The log shows endless errors of this type. I have a lot of temperature & humidity sensors installed. It seems that the plugin is trying to do some kind of calibration to one ore more sensors but fails. Is there a way to set the plugin not to try any calibration whatsoever?

Would you open a new issue. As it is not related to this one and it might be something other users may encounter as well. Don't forget to be as precise as possible (device, some errors from the log...).

closed 2025-04-07 by greekgremlin