Koenkk/zigbee2mqtt #26829: PTM215Z MQTT action button disappearance

schumi69, 2025-03-23:
### What happened?

Hello,

Since the Zigbee2Mqtt 2.0 update, the PTM215Z button action module no longer appears in Devices & Services ==> MQTT.

Attached is the action possible via Z2MQTT in version 1.42.0-2

However, the action is clearly visible in the Z2MQTT module (device + action recognition).

I remain available if you need further information.

[image]

### Zigbee2MQTT version

2.1.3.-1

### Adapter firmware version

1.42.0-2

### Adapter

Phoscon ConBee II

### Setup

Pi 4 X64

chris-1243, 2025-03-23:
`action` is a deprecated feature in 2.x or later.

You should set the lines below in your configuration.yaml

```
homeassistant:
  legacy_action_sensor: true
```

You may also wish to use mqtt or the experimental device events

I also provide a fully working blueprint for HA

https://github.com/chris-1243/HA---Blueprint/blob/main/PTM_215Z_ZE.yaml

schumi69, 2025-03-23:
Thank you!! ;) 

It's work. 

I just had to add the legacy line to the configuration.yaml file in the Zigbee2Mqtt folder!

Thanks for the quick response.

chris-1243, 2025-03-23:
Avec plaisir....

closed 2025-03-23 by Nerivec