Koenkk/zigbee2mqtt #26830: [New device support]: Enocean PTM210

schumi69, 2025-03-23:
### Link

https://www.enocean.com/en/product/ptm-210-ptm-215/

### Database entry

pairing impossible

### Zigbee2MQTT version

2.1.3-1 and 1.42.0-2

### Comments

Hello,

I integrated an Enocean PTM 215 Z module, and now I have a PTM 210. The latter is not recognized by Z2M. Is it possible to add the integration?

Thank you, and I remain available.

### External definition

```shell
Module not recognized by Z2M
```

### What does/doesn't work with the external definition?

Module not recognized by Z2M

chris-1243, 2025-03-23:
Are you sure the PTM210 is a zigbee product? As far as I may find on the web, PTM210 is an EnOcean 868MHz band and not an 2.4 GHz band as zigbee is using...

https://www.enocean-alliance.org/fr/product/ptm-210-wireless-switch-module-next-generation/

schumi69, 2025-03-23:
..... aïe....

Thank you. 

I won the right to add this protocol to home assistant….

closed 2025-03-23 by Nerivec