Koenkk/zigbee2mqtt #26838: zstack SLZB findings & issues

habitats-tech, 2025-03-24:
### What happened?

This thread is where I document all findings as well as issues pertaining to _every model_ of **`zstack` SLZB PoE Zigbee Hub**.

In addition to running several production environments, I have a test environment where in the same area I test several zigbee hubs from various vendors with a greater focus on SLZB and UZG hubs.

If you are using **`zstack` based SLZB**, you might want to consider following this thread. Feel free to contribute with any issues you might experience so we can all assist @Koenkk improve zigbee2mqtt.

Any contribution with devices for testing are welcome. Anything you can send me will permanently stay on my test network as I test new versions of everything pertaining to zigbee, thread, zigbee2mqtt, and Home Assistant.

Together we stand, divided we fail ....

Technology Architect/Designer
AudioVisual - Domotics - Electronics - Energy - Informatics - Security / ICT & IoT

Do it once - Do it right
https://portal.habitats.tech
https://ko-fi.com/habitatstech

Cloudflare -> x86-64 -> Ventoy -> Proxmox VE -> Pi-hole -> Traefik -> n8n -> HAOS

### Zigbee2MQTT version

x.x.x

### Adapter firmware version

xxxxxxxx

### Adapter

xxx

### Setup

Linux

Jit1978, 2025-03-24:
For testing purposes I fired up an old Virtualbox VM, which was set up and imaged on a fresh install not even a login user name for HomeAssistant defined. Created a basic set up and installed the add ons. Mosquito Broker then zigbee2mqtt and added the details to the config... Even then it still fails ever few minutes.

SMLIGHT SLZB-06p7 Flashed V2.7.1
Coordinator version - Revision: 20240315
Z2M Version - 2.1.3-1

Crashes very few minutes with the following:

[18:05:10] INFO: Preparing to start...
[18:05:10] INFO: Socat not enabled
[18:05:11] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2025-03-24 18:05:13] info: 	z2m: Logging to console, file (filename: log.log)
[2025-03-24 18:05:13] info: 	z2m: Starting Zigbee2MQTT version 2.1.3 (commit #unknown)
[2025-03-24 18:05:13] info: 	z2m: Starting zigbee-herdsman (3.2.7)
[2025-03-24 18:05:13] info: 	zh:zstack:znp: Opening TCP socket with 192.168.8.235:6638
[2025-03-24 18:05:13] info: 	zh:zstack:znp: Socket connected
[2025-03-24 18:05:13] info: 	zh:zstack:znp: Socket ready
[2025-03-24 18:05:13] info: 	zh:zstack:znp: Writing CC2530/CC2531 skip bootloader payload
[2025-03-24 18:05:14] info: 	zh:zstack:znp: Skip bootloader for CC2652/CC1352
[2025-03-24 18:05:17] info: 	z2m: zigbee-herdsman started (resumed)
[2025-03-24 18:05:17] info: 	z2m: Coordinator firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20240316,"transportrev":2},"type":"zStack3x0"}'
[2025-03-24 18:05:17] info: 	z2m: Currently 0 devices are joined.
[2025-03-24 18:05:17] info: 	z2m: Connecting to MQTT server at mqtt://core-mosquitto:1883
[2025-03-24 18:05:17] info: 	z2m: Connected to MQTT server
[2025-03-24 18:05:17] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"online"}'
[2025-03-24 18:05:17] info: 	z2m: Started frontend on port 8099
[2025-03-24 18:05:17] info: 	z2m: Zigbee2MQTT started!
[2025-03-24 18:05:22] info: 	z2m:mqtt: MQTT publish: topic 'homeassistant/binary_sensor/1221051039810110150109113116116_0x00124b00294bc033/connection_state/config', payload '{"device":{"hw_version":"zStack3x0 20240316","identifiers":["zigbee2mqtt_bridge_0x00124b00294bc033"],"manufacturer":"Zigbee2MQTT","model":"Bridge","name":"Zigbee2MQTT Bridge","sw_version":"2.1.3"},"device_class":"connectivity","entity_category":"diagnostic","name":"Connection state","object_id":"zigbee2mqtt_bridge_connection_state","origin":{"name":"Zigbee2MQTT","sw":"2.1.3","url":"https://www.zigbee2mqtt.io"},"payload_off":"offline","payload_on":"online","state_topic":"zigbee2mqtt/bridge/state","unique_id":"bridge_0x00124b00294bc033_connection_state_zigbee2mqtt","value_template":"{{ value_json.state }}"}'
[2025-03-24 18:09:53] **error: 	zh:zstack:znp: Socket error Error: read ECONNRESET**
[2025-03-24 18:09:53] info: 	zh:zstack:znp: Port closed
[2025-03-24 18:09:53] info: 	zh:zstack:znp: closing
[2025-03-24 18:09:53] **error: 	z2m: Adapter disconnected, stopping**
[2025-03-24 18:09:53] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"offline"}'
[2025-03-24 18:09:53] info: 	z2m: Disconnecting from MQTT server
[2025-03-24 18:09:53] info: 	z2m: Stopping zigbee-herdsman...
[2025-03-24 18:09:53] info: 	z2m: Stopped zigbee-herdsman
[2025-03-24 18:09:53] info: 	z2m: Stopped Zigbee2MQTT

Bare in mind this set up doesn't have any integrations installed, no Zigbee devices added all with brand new config data. This is leading me to believe there maybe a problem with the POE side of things. 

Thanks

J

habitats-tech, 2025-03-24:
> I installed today a new SMLIGHT SLZB-06p7 Adapter. Coordinator-Version: 20250321 Coordinator-Typ: zStack3x0 Z2M Version: 2.13 After 5 hours there was a warning in Z2M: 'Zigbee2MQTT server has exited unexpectedly' After 10s Z2m restarts!

It will be helpful if you can provide screenshots as in the following 2 screenshots (also what type of Z2M installation method you have used - e.g. Linux, Docker, HA, something else):

[image]

[image]

habitats-tech, 2025-03-24:
> For testing purposes I fired up an old Virtualbox VM, which was set up and imaged on a fresh install not even a login user name for HomeAssistant defined. Created a basic set up and installed the [...]

If you have installed the SLZB integration in HA, ensure you are not accessing it using mDNS. To ensure you are not using mDNS, delete the adapter in the SLZB integration, and when you add it again use its IPv4 address.

Optionally disable mDNS at the SLZB dashboard level. Also use static IPv4 for your SLZB and proper DNS settings. Example below.

[image]

Jit1978, 2025-03-26:
It's been stable now for the last 24 hours I've only changed the mDNS option so far. The channel is still on default 11. If it stops again I'll change it to 25

Will keep you updated..

TomKassel, 2025-03-28:
> I installed today a new SMLIGHT SLZB-06p7 Adapter. Coordinator-Version: 20250321 Coordinator-Typ: zStack3x0 Z2M Version: 2.13 After 5 hours there was a warning in Z2M: 'Zigbee2MQTT server has exited unexpectedly' After 10s Z2m restarts!
> 
> Z2M installed on Windows 11 PC, together with Homeseer Software.
> 
> [image]
> 
> [image]

In the meantime, Z2M starts about every 2 days. The SMLIGHT SLZB-06p7 works perfectly.

TomKassel, 2025-03-28:
> In the meantime, Z2M starts about every 2 days. The SMLIGHT SLZB-06p7 works perfectly.

Now it took just 4 hours to restart!

TomKassel, 2025-03-29:
Since I have been using the Zstack adapter, Z2M starts every 4-24 hours.
Again the adapter works, only the connection is lost for a short time due to the restart.

[image]

habitats-tech, 2025-04-02:
I would strongly suggest you update the FW to the version highlighted in red below. If problem persists I suggest you update Z2M to v2.2.0. If all this fails I would like to see a screenshot of your setup from Z2M About page.

[image]

habitats-tech, 2025-04-02:
> Thanks, but I've been on firmware 20250321, V2.8.2.dev3 since my first post Next i will update Z2M to version 2.2. What I also noticed is that OTA updates that used to take 20-60 minutes now take up to 12 hours (!).

Are you possibly running something in your network which can affect things, like for e.g. Pi-hole, Adguard, any forward or reverse proxy, or some other network probing/monitoring service. Could it be your router the issue - e.g. some recent update. OTA should be around 30mins for battery devices.

habitats-tech, 2025-04-02:
I see your Z2M is running inside HA. Let me try to emulate your environment.

TomKassel, 2025-04-02:
I don't use HA but Homeseer.
Nothing has changed in the system except for the coordinator. The Conbee2 needed about 30 minutes for an update on the same system.

TomKassel, 2025-04-02:
A friend of mine does not use HA and no Homeseer, but also Z2M with the SLZB-06P7 and has the same OTA problem.

habitats-tech, 2025-04-02:
Same version of FW as yourself?

habitats-tech, 2025-04-02:
Can you share your and your friend's screenshor of Z2M -> About (please include the list of devices).

TomKassel, 2025-04-02:
He has the same SLZB-06P7 with the same FW.

Here is my About first:

Zigbee2MQTT-Version
[2.1.3](https://github.com/Koenkk/zigbee2mqtt/releases/tag/2.1.3) commit: [unknown](https://github.com/Koenkk/zigbee2mqtt/commit/unknown)
Coordinator-Typ
zStack3x0
Coordinator-Version
20250321
Coordinator IEEE Adresse
0x00212effff09743c
Frontend-Version
0.9.4
Zigbee Herdsman Konverter Version
23.2.0
Zigbee Herdsman Version
3.2.7
Statistiken
Gesamt 186
Nach Typ
Endgerät: 97
Router: 89
Nach Stromquelle
Batterie: 94
Netz (einphasig): 90
unknown: 2
Nach Hersteller
Schneider Electric: 61
LUMI: 44
D5X84YU: 38
Signify Netherlands B.V.: 9
AEOTEC: 7
HEIMAN: 4
_TYZB01_qiplt4jq: 4
Paulmann Licht GmbH: 4
Sunricher: 3
_TZE204_fncxk3ob: 3
NodOn: 3
SONOFF: 2
Samjin: 1
undefined: 1
LEDVANCE: 1
AL001: 1
Nach Modell
1GANG/SWITCH/1: 33
eT093WRG: 26
lumi.sensor_magnet.aq2: 18
lumi.weather: 13
eT093WRO: 12
PUCK/SWITCH/1: 10
1GANG/DIMMER/1: 9
LOM007: 9
lumi.magnet.agl02: 8
ZGA002: 7
50131: 4
TS0205: 4
HS3HT-EFA-3.0: 3
SOCKET/OUTLET/1: 3
ZGRC-KEY-013: 3
TS0601: 3
SIN-4-1-20: 3
PUCK/DIMMER/1: 2
1GANG/SHUTTER/1: 2
lumi.flood.agl02: 2
lumi.sensor_ht.agl02: 2
SNZB-05P: 2
lumi.vibration.aq1: 1
button: 1
W599001: 1
lumi.plug.maeu01: 1
PLUG EU EM T: 1
SmokeSensor-EF-3.0: 1
WG001-Z01: 1
W564100: 1

habitats-tech, 2025-04-02:
You seem to have some undefined and unknown entries. You might want to try to remove them from the network and see if they might be the root cause of the issues.

Can you enable debug log in Z2M and when the system resets provide the lines pertaining to the error.

TomKassel, 2025-04-02:
At 17:37: Z2M restarted:
Zigbee2MQTT server has exited unexpectedly' 
After 10s Z2m restarts!

[log.log](https://github.com/user-attachments/files/19572277/log.log)

habitats-tech, 2025-04-02:
It seems a timeout failure. If you can remove the device called: EG Langer Flur Bewegungsmelder.

Please provide feedback in case issue solved. Do not reconnect even if u get another error. We need to try eliminate all errors in your system.

TomKassel, 2025-04-02:
Is it enough to delete the device in Z2M?

habitats-tech, 2025-04-02:
Yes, because it will also delete it from the ZB network. You can force delete with no issues.

TomKassel, 2025-04-02:
I'll do that right away. I'll get back to you, thank you!

classicsdriver, 2025-04-19:
Hi,
thanks for the helpful thread. I am running Z2Q on Homeassistant since 6 years. Never had an issue so far, but in the last time, I am having quite some troubles. Hope to get some insight or tipps on what to check, to get it running smoothly again.

Adapter: Zigstar Stick v4
Firmware: CC1352P2_CC2652P_launchpad_coordinator_20240710
Coordinator-Version: 20240710

The two "undefined" devices are 2 EnOcean switches. I counted all other devices.

[image]
[image]
[image]

One day my network just started getting instabilities and some devices being unresponsive. I tried restarting, flashing the coordinator, turning off all devices, .... then 2 days later, over night, everything came back on. It was running nicely for 4-5 weeks now. But 4-5 days ago, it all started again, that alot of devices went offline.  I had the hope, they would come back online, but of course they didn't. Yesterday I started pulling all fuses and reconnection each room one by one. Most of the devices came back online. But this morning, most of the plugged in devices are offline again. Most of the batterie powered devices are still online (same behaviour like last time). I get alot of "no network route" messages. Yesterday, to make sure, I also deactivated my 2,4 Ghz Wifi (Channel 11) to make sure, there are no interferance with my zigbee (channel 11). It did not change anything. And it would be kinda wired, as it kept running like this for 6 years.

I attached a current log after setting all up again tonight. 

Any help is highly appreciated. Thanks alot in advance

[log.log](https://github.com/user-attachments/files/19821557/log.log)

SeLLeRoNe, 2025-06-19:
Hi guys,

I have been looking for a solution to the same issue for the past month or so, tried downgraed/upgrade/sidedgrade... can't seem to have it working.

The idea I had so far is the that the Zigbee network just collapse and as a consequence Z2M goes onto a restart loop (which drain docker rate limit in TrueNAS , but that's another story) for I guess the TrueNAS team :)

This is what I have right now:

[image]

In Z2M I have paired 118 Devices, one of which is a SLZB-06M set as a Router on another floor.
Beside the random battery-powered device lost overtime, I have never expedited something this impactful on the Zigbee network.

Please let me know what I can provide that could be of any help to try to find the culprite and possibly a solution, as my spouse approval factor of the smart home is falling very very quickly due to this :)

Thank you very much