Koenkk/zigbee2mqtt #26839: ember SLZB findings & issues

habitats-tech, 2025-03-24:
### What happened?

This thread is where I document all findings as well as issues pertaining to _every model_ of **`ember` SLZB PoE Zigbee Hub**.

In addition to running several production environments, I have a test environment where in the same area I test several zigbee hubs from various vendors with a greater focus on SLZB and UZG hubs.

If you are using **`ember` based SLZB**, you might want to consider following this thread. Feel free to contribute with any issues you might experience so we can all assist @Koenkk improve zigbee2mqtt.

Any contribution with devices for testing are welcome. Anything you can send me will permanently stay on my test network as I test new versions of everything pertaining to zigbee, thread, zigbee2mqtt, and Home Assistant.

Together we stand, divided we fail ....

Technology Architect/Designer
AudioVisual - Domotics - Electronics - Energy - Informatics - Security / ICT & IoT

Do it once - Do it right
https://portal.habitats.tech
https://ko-fi.com/habitatstech

Cloudflare -> x86-64 -> Ventoy -> Proxmox VE -> Pi-hole -> Traefik -> n8n -> HAOS

### Zigbee2MQTT version

x.x.x

### Adapter firmware version

xxxxxxxx

### Adapter

xxx

### Setup

Linux

habitats-tech, 2025-03-24:
I have documented an issue where zigbee on ember SLZB resets at random intervals, even when no devices attached on the network. All future updates pertaining to my findings on this and future issues will be documented in this thread.

https://github.com/Koenkk/Z-Stack-firmware/issues/518#issuecomment-2728207982
https://github.com/Koenkk/Z-Stack-firmware/issues/518#issuecomment-2735269891
https://github.com/Koenkk/Z-Stack-firmware/issues/518#issuecomment-2745126173
https://github.com/Koenkk/Z-Stack-firmware/issues/518#issuecomment-2745134263

Tazmanian79, 2025-04-01:
I have an SLZB-06M with Zigbee2mqtt version 2.1.3

Sometimes (still figuring out when) the Zigbee container is restarted.
Is really annoying because when zigbee is coming back all entities gets updated and sometimes it triggers my alarmo (Alarmo in Home Assistant)

This is my log just when the error happens:

`
[2025-04-01 20:57:09] error: 	z2m: Publish 'set' 'state' to 'Verlichting Living Play 1' failed: 'Error: ZCL command 0x00178801091b7c77/11 genLevelCtrl.moveToLevelWithOnOff({"level":0,"transtime":20}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed ({"target":6177,"apsFrame":{"profileId":260,"clusterId":8,"sourceEndpoint":1,"destinationEndpoint":11,"options":4416,"groupId":0,"sequence":17},"zclSequence":93,"commandIdentifier":11} timed out after 10000ms)'
[2025-04-01 20:57:09] error: 	z2m: Publish 'set' 'brightness' to 'Verlichting Barkast Lamp 2' failed: 'Error: ZCL command 0x001788010bb2b143/11 genLevelCtrl.moveToLevelWithOnOff({"level":254,"transtime":20}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed ({"target":36562,"apsFrame":{"profileId":260,"clusterId":8,"sourceEndpoint":1,"destinationEndpoint":11,"options":4416,"groupId":0,"sequence":18},"zclSequence":128,"commandIdentifier":11} timed out after 10000ms)'
[2025-04-01 20:57:09] error: 	z2m: Publish 'set' 'brightness' to 'Verlichting Barkast Lamp 3' failed: 'Error: ZCL command 0x001788010bb2a951/11 genLevelCtrl.moveToLevelWithOnOff({"level":254,"transtime":20}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed ({"target":32275,"apsFrame":{"profileId":260,"clusterId":8,"sourceEndpoint":1,"destinationEndpoint":11,"options":4416,"groupId":0,"sequence":19},"zclSequence":129,"commandIdentifier":11} timed out after 10000ms)'
[2025-04-01 20:57:09] error: 	zh:ember:ezsp: ERROR Transaction failure; status=ASH_ERROR_TIMEOUTS. Last Frame: [FRAME: ID=52:"SEND_UNICAST" Seq=168 Len=10].
[2025-04-01 20:57:09] error: 	zh:ember: Adapter fatal error: ASH_ERROR_TIMEOUTS
[2025-04-01 20:57:09] info: 	zh:ember:uart:ash: ASH COUNTERS since last clear:
[2025-04-01 20:57:09] info: 	zh:ember:uart:ash:   Total frames: RX=607, TX=775
[2025-04-01 20:57:09] info: 	zh:ember:uart:ash:   Cancelled   : RX=0, TX=0
[2025-04-01 20:57:09] info: 	zh:ember:uart:ash:   DATA frames : RX=572, TX=170
[2025-04-01 20:57:09] info: 	zh:ember:uart:ash:   DATA bytes  : RX=13628, TX=4034
[2025-04-01 20:57:09] info: 	zh:ember:uart:ash:   Retry frames: RX=31, TX=1
[2025-04-01 20:57:09] info: 	zh:ember:uart:ash:   ACK frames  : RX=3, TX=603
[2025-04-01 20:57:09] info: 	zh:ember:uart:ash:   NAK frames  : RX=0, TX=0
[2025-04-01 20:57:09] info: 	zh:ember:uart:ash:   nRdy frames : RX=0, TX=0
[2025-04-01 20:57:09] info: 	zh:ember:uart:ash:   CRC errors      : RX=0
[2025-04-01 20:57:09] info: 	zh:ember:uart:ash:   Comm errors     : RX=0
[2025-04-01 20:57:09] info: 	zh:ember:uart:ash:   Length < minimum: RX=0
[2025-04-01 20:57:09] info: 	zh:ember:uart:ash:   Length > maximum: RX=0
[2025-04-01 20:57:09] info: 	zh:ember:uart:ash:   Bad controls    : RX=0
[2025-04-01 20:57:09] info: 	zh:ember:uart:ash:   Bad lengths     : RX=0
[2025-04-01 20:57:09] info: 	zh:ember:uart:ash:   Bad ACK numbers : RX=0
[2025-04-01 20:57:09] info: 	zh:ember:uart:ash:   Out of buffers  : RX=0
[2025-04-01 20:57:09] info: 	zh:ember:uart:ash:   Retry dupes     : RX=31
[2025-04-01 20:57:09] info: 	zh:ember:uart:ash:   Out of sequence : RX=0
[2025-04-01 20:57:09] info: 	zh:ember:uart:ash:   ACK timeouts    : RX=1
[2025-04-01 20:57:09] info: 	zh:ember:uart:ash: ======== ASH stopped ========
[2025-04-01 20:57:09] info: 	zh:ember:ezsp: ======== EZSP stopped ========
[2025-04-01 20:57:09] info: 	zh:ember: ======== Ember Adapter Stopped ========
[2025-04-01 20:57:09] error: 	zh:ember:ezsp: ERROR Transaction failure; status=NOT_CONNECTED. Last Frame: [FRAME: ID=52:"SEND_UNICAST" Seq=168 Len=10].
[2025-04-01 20:57:09] error: 	z2m: Adapter disconnected, stopping
[2025-04-01 20:57:09] error: 	zh:ember:ezsp: ERROR Transaction failure; status=NOT_CONNECTED. Last Frame: [FRAME: ID=52:"SEND_UNICAST" Seq=168 Len=10].
[2025-04-01 20:57:09] error: 	zh:ember:ezsp: ERROR Transaction failure; status=NOT_CONNECTED. Last Frame: [FRAME: ID=52:"SEND_UNICAST" Seq=168 Len=10].
[2025-04-01 20:57:09] error: 	zh:ember:ezsp: ERROR Transaction failure; status=NOT_CONNECTED. Last Frame: [FRAME: ID=52:"SEND_UNICAST" Seq=168 Len=10].
[2025-04-01 20:57:09] error: 	z2m: Publish 'set' 'brightness' to 'Verlichting Barkast Lamp 4' failed: 'Error: ZCL command 0x001788010bc12a19/11 genLevelCtrl.moveToLevelWithOnOff({"level":254,"transtime":20}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (~x~> [ZCL to=0x001788010bc12a19:55250 apsFrame={"profileId":260,"clusterId":8,"sourceEndpoint":1,"destinationEndpoint":11,"options":4416,"groupId":0,"sequence":0}] Failed to send request with status=FAIL.)'
[2025-04-01 20:57:09] error: 	zh:ember:ezsp: ERROR Transaction failure; status=NOT_CONNECTED. Last Frame: [FRAME: ID=52:"SEND_UNICAST" Seq=168 Len=10].
[... cut ...]
[2025-04-01 20:57:09] error: 	zh:ember:ezsp: ERROR Transaction failure; status=NOT_CONNECTED. Last Frame: [FRAME: ID=52:"SEND_UNICAST" Seq=168 Len=10].
[2025-04-01 20:57:09] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"offline"}'
[2025-04-01 20:57:09] info: 	z2m: Disconnecting from MQTT server
[2025-04-01 20:57:09] info: 	z2m: Stopping zigbee-herdsman...
[2025-04-01 20:57:10] info: 	z2m: Stopped zigbee-herdsman
[2025-04-01 20:57:10] info: 	z2m: Stopped Zigbee2MQTT
`

Do you need any other info?

habitats-tech, 2025-04-02:
This is a known problem. Even with no devices on the ZB network same issue, adapter resets.

I am testing recent Z2M 2.2.0 update and in a couple should be confident if issue fixed.

Tazmanian79, 2025-04-02:
I have 2 different location, same setup both have same issue.

On one location I upgraded today to 2.2.0 but the container just restarted.

The logs:

[2025-04-02 18:01:10] error: 	zh:ember:uart:ash: Received ERROR from adapter, with code=ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT.
[2025-04-02 18:01:10] error: 	zh:ember:uart:ash: ASH disconnected | Adapter status: ASH_NCP_FATAL_ERROR
[2025-04-02 18:01:10] error: 	zh:ember:uart:ash: Error while parsing received frame, status=ASH_NCP_FATAL_ERROR.
[2025-04-02 18:01:10] error: 	zh:ember: Adapter fatal error: HOST_FATAL_ERROR
[2025-04-02 18:01:10] info: 	zh:ember:uart:ash: ASH COUNTERS since last clear:
[2025-04-02 18:01:10] info: 	zh:ember:uart:ash:   Total frames: RX=77, TX=83
[2025-04-02 18:01:10] info: 	zh:ember:uart:ash:   Cancelled   : RX=0, TX=0
[2025-04-02 18:01:10] info: 	zh:ember:uart:ash:   DATA frames : RX=73, TX=7
[2025-04-02 18:01:10] info: 	zh:ember:uart:ash:   DATA bytes  : RX=3310, TX=192
[2025-04-02 18:01:10] info: 	zh:ember:uart:ash:   Retry frames: RX=3, TX=0
[2025-04-02 18:01:10] info: 	zh:ember:uart:ash:   ACK frames  : RX=0, TX=76
[2025-04-02 18:01:10] info: 	zh:ember:uart:ash:   NAK frames  : RX=0, TX=0
[2025-04-02 18:01:10] info: 	zh:ember:uart:ash:   nRdy frames : RX=0, TX=0
[2025-04-02 18:01:10] info: 	zh:ember:uart:ash:   CRC errors      : RX=0
[2025-04-02 18:01:10] info: 	zh:ember:uart:ash:   Comm errors     : RX=0
[2025-04-02 18:01:10] info: 	zh:ember:uart:ash:   Length < minimum: RX=0
[2025-04-02 18:01:10] info: 	zh:ember:uart:ash:   Length > maximum: RX=0
[2025-04-02 18:01:10] info: 	zh:ember:uart:ash:   Bad controls    : RX=0
[2025-04-02 18:01:10] info: 	zh:ember:uart:ash:   Bad lengths     : RX=0
[2025-04-02 18:01:10] info: 	zh:ember:uart:ash:   Bad ACK numbers : RX=0
[2025-04-02 18:01:10] info: 	zh:ember:uart:ash:   Out of buffers  : RX=0
[2025-04-02 18:01:10] info: 	zh:ember:uart:ash:   Retry dupes     : RX=3
[2025-04-02 18:01:10] info: 	zh:ember:uart:ash:   Out of sequence : RX=0
[2025-04-02 18:01:10] info: 	zh:ember:uart:ash:   ACK timeouts    : RX=0
[2025-04-02 18:01:10] info: 	zh:ember:uart:ash: ======== ASH stopped ========
[2025-04-02 18:01:10] info: 	zh:ember:ezsp: ======== EZSP stopped ========
[2025-04-02 18:01:10] info: 	zh:ember: ======== Ember Adapter Stopped ========
[2025-04-02 18:01:10] error: 	z2m: Adapter disconnected, stopping
[2025-04-02 18:01:10] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"offline"}'
[2025-04-02 18:01:10] info: 	z2m: Disconnecting from MQTT server
[2025-04-02 18:01:10] info: 	z2m: Stopping zigbee-herdsman...
[2025-04-02 18:01:12] info: 	z2m: Stopped zigbee-herdsman
[2025-04-02 18:01:12] info: 	z2m: Stopped Zigbee2MQTT

habitats-tech, 2025-04-02:
You have to either wait for new firmware to hopefully solve this issue or go back to either 20241127 or 20240510. If memory serves me well the most solid version was 20240510.

In the past I was uncertain if it was ZB FW or Z2M the issue. Now I am confident the issue is the ZB FW.

Tazmanian79, 2025-04-02:
Ok I have installed 20240510 and will let you know if that solves it.

Tazmanian79, 2025-04-03:
Don't know if it was coincidence or not.  But on my netwerk which is still on the latest firmware my internet went down (not local network, but outside internet) and at same time I had the same error.

Maybe the adapter crashes because of problem with outside connections?

habitats-tech, 2025-04-03:
Coincidence from a Z2M perspective. The Z2M ecosystem is local and only requires Internet for OTA or updates to Z2M.

However how SLZB handles Internet disconnect is another issue. In theory no correlation, but how is Internet disconnect recovery handled by the FW when for example SLZB was checking for updates while disconnection took place. Sometimes accepting this as an edge case, we put no effort to write a recovery procedure, and therefore with no proper error recovery could cause the SLZB to disconnect. But this is me thinking loud from a developer perspective rather than a fact.

Tazmanian79, 2025-04-03:
I will check it later and disconnect the internet myself to see if this this triggers the error or not.

Tazmanian79, 2025-04-06:
unfortunately with version 20240510 also have restarts :-(

[2025-04-06 09:05:39] error: 	zh:ember:uart:ash: Received ERROR from adapter, with code=ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT.
[2025-04-06 09:05:39] error: 	zh:ember:uart:ash: ASH disconnected | Adapter status: ASH_NCP_FATAL_ERROR
[2025-04-06 09:05:39] error: 	zh:ember:uart:ash: Error while parsing received frame, status=ASH_NCP_FATAL_ERROR.
[2025-04-06 09:05:39] error: 	zh:ember: Adapter fatal error: HOST_FATAL_ERROR
[2025-04-06 09:05:39] info: 	zh:ember:uart:ash: ASH COUNTERS since last clear:
[2025-04-06 09:05:39] info: 	zh:ember:uart:ash:   Total frames: RX=3088, TX=3729
[2025-04-06 09:05:39] info: 	zh:ember:uart:ash:   Cancelled   : RX=0, TX=0
[2025-04-06 09:05:39] info: 	zh:ember:uart:ash:   DATA frames : RX=3068, TX=643
[2025-04-06 09:05:39] info: 	zh:ember:uart:ash:   DATA bytes  : RX=75574, TX=16806
[2025-04-06 09:05:39] info: 	zh:ember:uart:ash:   Retry frames: RX=18, TX=0
[2025-04-06 09:05:39] info: 	zh:ember:uart:ash:   ACK frames  : RX=1, TX=3086
[2025-04-06 09:05:39] info: 	zh:ember:uart:ash:   NAK frames  : RX=0, TX=0
[2025-04-06 09:05:39] info: 	zh:ember:uart:ash:   nRdy frames : RX=0, TX=0
[2025-04-06 09:05:39] info: 	zh:ember:uart:ash:   CRC errors      : RX=0
[2025-04-06 09:05:39] info: 	zh:ember:uart:ash:   Comm errors     : RX=0
[2025-04-06 09:05:39] info: 	zh:ember:uart:ash:   Length < minimum: RX=0
[2025-04-06 09:05:39] info: 	zh:ember:uart:ash:   Length > maximum: RX=0
[2025-04-06 09:05:39] info: 	zh:ember:uart:ash:   Bad controls    : RX=0
[2025-04-06 09:05:39] info: 	zh:ember:uart:ash:   Bad lengths     : RX=0
[2025-04-06 09:05:39] info: 	zh:ember:uart:ash:   Bad ACK numbers : RX=0
[2025-04-06 09:05:39] info: 	zh:ember:uart:ash:   Out of buffers  : RX=0
[2025-04-06 09:05:39] info: 	zh:ember:uart:ash:   Retry dupes     : RX=18
[2025-04-06 09:05:39] info: 	zh:ember:uart:ash:   Out of sequence : RX=0
[2025-04-06 09:05:39] info: 	zh:ember:uart:ash:   ACK timeouts    : RX=0
[2025-04-06 09:05:39] info: 	zh:ember:uart:ash: ======== ASH stopped ========
[2025-04-06 09:05:39] info: 	zh:ember:ezsp: ======== EZSP stopped ========
[2025-04-06 09:05:39] info: 	zh:ember: ======== Ember Adapter Stopped ========
[2025-04-06 09:05:39] error: 	z2m: Adapter disconnected, stopping
[2025-04-06 09:05:39] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"offline"}'
[2025-04-06 09:05:39] info: 	z2m: Disconnecting from MQTT server
[2025-04-06 09:05:40] info: 	z2m: Stopping zigbee-herdsman...
[2025-04-06 09:05:42] info: 	z2m: Stopped zigbee-herdsman
[2025-04-06 09:05:42] info: 	z2m: Stopped Zigbee2MQTT

ulikasper, 2025-04-10:
Hey everyone. 
I had the same issue now for about 4 weeks. Tried a lot, rollback firmwares, changed connection-types.
Nothing fixed the issue of ECONNRESET or TIMEOUT errors like mentioned here. 

In my case it was just the network-connection causing all this. Connecting the SLZB-06M directly to the main-switch in my house, solved it for me. 

See here: 
https://github.com/Koenkk/zigbee2mqtt/issues/25920#issuecomment-2792087247

Tazmanian79, 2025-04-10:
In one location I have the SLZB connected to the main switch.  Same as where my docker container is running.
In another location I have the SLZB connected in another switch as the main does not deliver PoEE.

But both instances have the same issue.

I'm now on fw 20240510 and seems to be much more stable. Althought till now had one time the same problem.

ulikasper, 2025-04-10:
Propably it depends on how the hardware is able to manage the traffic!? 
My main switch is an Edge-corE ES4524C Managed switch without poe. And my HomeAssistant is running as HA-OS VM in a Proxmox environment.

Could it be, that the docker container is the bottleneck causing network inconsistencies?

Tazmanian79, 2025-04-10:
Don't know, don't have any issues with my other docker containers.

I'm using a Synology DS-918+ and my network are all Unifi devices.

Tazmanian79, 2025-04-16:
It has definitely something to do with the connection.

Few mins ago my internet was down (outside not local) and the container restarted again.

ulikasper, 2025-04-16:
In my case, I had the same port in SLZB and in the Z2M addon, but it did not work stable, connected to additional POE switch. 
But I see you have set up "zstack" as adapter type. Did you have the same problems like all "ember" users had?

Mywk, 2025-04-17:
My issue was always ECONNRESET with the SLZB-06M after updating zigbee2mqtt, it's running stable so far.

ulikasper, 2025-04-17:
okay, but usually you need to set adapter to "ember" if you use the SLZB-06M. this is what I always read about configure it in HA. Because the "M" uses a completly different chip than the SLZB-06

Mywk, 2025-06-11:
I've noticed that after an update ZHA was starting automatically, therefore conflicting with zigbee2mqtt. Those still having issues should check that as well.

habitats-tech, 2025-06-12:
So far the best combination of Z2M and 06M is:

06M on FW v2.8.8
Z2M on 2.2.1 commit: a14feef8 (note Z2M running in LXC on Proxmox)

benjamin-dcs, 2025-06-12:
> but anything above 2.2.1 is fine (tested with 2.4.0 as well)

What is considered 'fine'. For me the only option would be 'setup and forget', not 'reboot on internal failure and it's back' 😃

habitats-tech, 2025-06-12:
Fine means no issues. 2.4 is recent and therefore do not have longer term data, but so far 2.2 and above versions in combination with 06M FW v2.8.8 run flawless.

IoT is a fast moving landscape and therefore nobody can guarantee anything long term. You system is stable with the configuration you have. No guarantee this will be the case if you introduce new devices in years to come.

What is more important is the support you get when there are issues. So in summary 06M v2.8.8 with Z2M 2.x runs stable.

Tazmanian79, 2025-06-12:
It is stable.  Beside of the restart ... but that's not really an issue as everything comes back itself within a min.  no interaction needed.

9shearer, 2025-08-07:
One more from me: https://github.com/Koenkk/zigbee2mqtt/issues/28197

The ember radio on my SLZB-MR3 (chip EFR32MG24) restarts every hour on the clock. Error details in the linked thread (ASH_ERROR_TIMEOUTS). The zstack one (chip CC2674P10) is fine all along. Only one end device connected to each (same model to both - Ikea Parasoll), no connection to Home Assistant.

9shearer, 2025-08-08:
Not necessarily. I am not an expert, but I would guess multi-threading would improve responsiveness of the adapter and thus reduce timeout risk, especially in a network with many devices (as several requests/responses could be processed in parallel rather than sequentially).
My test networks only have one device each, so that shouldn't be an issue - worrying though that the packet processing collapses the whole thing.

I'll try some older firmwares for the SLZB-06M that keeps resetting to see if it cures it. It bounced 3 times in the last hour.

9shearer, 2025-08-08:
The SLZB-06M seems to be stabilized for the past 5 hours or so (longest I had without disconnects before was ~6h).
Setup combination:
- SLZB OS 2.9.6
- Zigbee Coordinator firmware 20240510
- "Enable Zigbee Socket packet processing" is disabled
- "Allow multi-threaded socket connection" is enabled
- mDNS is enabled

The SLZB-MR3 (connected to a standalone z2m instance, and with a single end device) is still resetting every 60 minutes on the EFR chip (the CP is fine and steady).
I have enabled "availability", but doesn't seem to have helped (and "last seen" is still not shown).

9shearer, 2025-08-08:
20240510 still throws disconnects, going for 20240330.

Mrsash2020, 2025-08-09:
> The SLZB-06M seems to be stabilized for the past 5 hours or so (longest I had without disconnects before was ~6h). Setup combination:
> 
>     * SLZB OS 2.9.6
> 
>     * Zigbee Coordinator firmware 20240510
> 
>     * "Enable Zigbee Socket packet processing" is disabled
> 
>     * "Allow multi-threaded socket connection" is enabled
> 
>     * mDNS is enabled
> 
> 
> The SLZB-MR3 (connected to a standalone z2m instance, and with a single end device) is still resetting every 60 minutes on the EFR32MG24 chip (the CC2674P10) is fine and steady). I have enabled "availability", but doesn't seem to have helped (and "last seen" is still not shown). This one is also on SLZB 2.9.6, firmware versions are 20250212 (on the EFR) and 20240716 (on the CC) - both being latest stable.

Ok so mine is very stable with  SLZB-06 with network on  channel: 25

    SLZB OS 2.9.4
    Zigbee Coordinator firmware 20240710
    "Enable Zigbee Socket packet processing" is disabled
    "Allow multi-threaded socket connection" is disabled
    mDNS is disabled

[image]

[image]

I always thought the EFR32MG24 chip is for matter while the CC2674P10 is for Zigbee for some reason when I originally read about them. Could be mistaken. I blame my age.....

For me if I swap to the SLZB-MR3 picks up the network and is stable for half a day then drops a few or 40 odd devices and have to play reboot gymnastics to get it working again.

Mrsash2020, 2025-08-10:
I have just flashed the 20240716 again moving from the custom one they provided me and will wait to see what happens.

I dont have any matter devices either. This was my attempt to bring matter in my home plus consolidate as you have. I have a query. 

I setup matter and thread integrations. Do I also need Open Thread Border Router? If yes what is the URL that I need to add?

Mrsash2020, 2025-08-10:
Ok I think I have resolved it somehow.

pilot1981, 2025-08-10:
Using SLZB-06M, today I encountered problem to pair IKEAFYRTUR blinds:
- unsupported device
- not possible to interview it

9shearer, 2025-08-11:
Try moving them closer (if possible) when doing the pairing, or pairing explicitly through a nearby router (don't select "All" when enabling pairing in Z2M, but pick a close-by router from the list and enable it). I had the same issue with a couple of end devices.

pilot1981, 2025-08-11:
> Try moving them closer (if possible) when doing the pairing, or pairing explicitly through a nearby router (don't select "All" when enabling pairing in Z2M, but pick a close-by router from the list and enable it). I had the same issue with a couple of end devices.

Hi,
thanks for reply. I tried to select nearest zigbee router but this don't help...The only solution found is to move blind close to coordinator SMLIGHT SLZ06: it paired fine.

thanks a lot for suggestions!

closed 2025-10-19 by github-actions[bot]