Koenkk/zigbee2mqtt #26841: Watchdog doesn't work

alex-v-fraser, 2025-03-24:
### What happened?

I noticed that no automation work with my Z2M devices. So I logged in my HA and discovered that all Z2M devices were offline. I just restarted Z2M addon and they all appeared online. I noticed the following string in  Z2M log, though watchdog is enabled:
`Starting Zigbee2MQTT without watchdog.`

### What did you expect to happen?

restart Z2M when it can't ping all devices

### Zigbee2MQTT version

2.1.3-1

### Adapter firmware version

20230507

### Adapter

zStack3x0 (Sonoff Zigbee USB stick)

### Setup

Add-on on Home Assistant OS core-2025.3.4 (ova)

### Debug log

```

[2025-03-24 06:14:38] [33mwarning[39m: 	z2m: Failed to ping 'Bedroom LED' (attempt 1/1, ZCL command 0xxxxxxxxxxxxxxxxxx/11 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms))
[2025-03-24 06:15:08] [33mwarning[39m: 	z2m: Failed to ping 'tambour_switch' (attempt 1/1, ZCL command 0xxxxxxxxxxxxxx/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms))
[2025-03-24 06:18:01] [33mwarning[39m: 	z2m: Failed to ping 'gas_sensor' (attempt 1/1, ZCL command 0xxxxxxxxxxxxxxxxxxxx/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms))
[2025-03-24 06:18:18] [33mwarning[39m: 	z2m: Failed to ping 'koridor_relay2' (attempt 1/1, ZCL command 0xxxxxxxxxxxxxx/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms))
[2025-03-24 06:18:44] [33mwarning[39m: 	z2m: Failed to ping 'children's switch' (attempt 1/1, ZCL command 0xxxxxxxxxxxxxxxxx/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms))
[2025-03-24 06:19:07] [33mwarning[39m: 	z2m: Failed to ping 'kitchen_curtain' (attempt 1/1, ZCL command 0xxxxxxxxxxxxxxxxx/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms))
[2025-03-24 06:19:24] [33mwarning[39m: 	z2m: Failed to ping 'Kitchen LED' (attempt 1/1, ZCL command 0xxxxxxxxxxxxxxx/11 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms))
[2025-03-24 06:19:52] [33mwarning[39m: 	z2m: Failed to ping 'kitchen switch' (attempt 1/1, ZCL command 0xxxxxxxxxxxxxxxxxx/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms))
[2025-03-24 06:20:29] [33mwarning[39m: 	z2m: Failed to ping 'childroom_curtain' (attempt 1/1, ZCL command 0xxxxxxxxxxxxxxxx/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms))
[2025-03-24 06:20:46] [33mwarning[39m: 	z2m: Failed to ping 'childroom_tull' (attempt 1/1, ZCL command 0xxxxxxxxxxxxxxxx/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms))
[2025-03-24 06:21:03] [33mwarning[39m: 	z2m: Failed to ping 'bedroom_lamp' (attempt 1/1, ZCL command 0xxxxxxxxxxxxxxxx/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms))
[2025-03-24 06:21:20] [33mwarning[39m: 	z2m: Failed to ping 'koridor relay' (attempt 1/1, ZCL command 0xxxxxxxxxxxxxxxx/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms))
[2025-03-24 06:21:37] [33mwarning[39m: 	z2m: Failed to ping 'bedroom_curtain' (attempt 1/1, ZCL command 0xxxxxxxxxxxxxxx/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms))
[2025-03-24 06:22:29] [33mwarning[39m: 	z2m: Failed to ping 'zal_led' (attempt 1/1, ZCL command 0xxxxxxxxxxxxxx/11 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms))
[2025-03-24 06:23:06] [33mwarning[39m: 	z2m: Failed to ping 'zal switch' (attempt 1/1, ZCL command 0xxxxxxxxxxxxxxxxxx/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms))
[2025-03-24 06:23:12] [31merror[39m: 	zh:controller: Failed to disable join on stop: Error: SRSP - AF - dataRequestExt after 6000ms
[06:23:17] INFO: [32mPreparing to start...[0m
[06:23:18] INFO: [32mSocat not enabled[0m
[06:23:18] INFO: [32mStarting Zigbee2MQTT...[0m
Starting Zigbee2MQTT without watchdog.
[2025-03-24 07:23:25] [33mwarning[39m: 	zh:controller:database: Found '/config/zigbee2mqtt/database.db.tmp' when writing database, indicating past write failure; renamed it to '/config/zigbee2mqtt/database.db.tmp.2025-03-24T04-23-25.203Z'

```

Koenkk, 2025-03-24:
It looks like your adapter crashed, 20230507 firmware is not considered, stable, try the fw from https://github.com/Koenkk/Z-Stack-firmware/discussions/545

alex-v-fraser, 2025-03-25:
> try the fw from [Koenkk/Z-Stack-firmware#545](https://github.com/Koenkk/Z-Stack-firmware/discussions/545)

@Koenkk is it beta? I can't see 20250321 version in releases.

Koenkk, 2025-03-26:
> > try the fw from [Koenkk/Z-Stack-firmware#545](https://github.com/Koenkk/Z-Stack-firmware/discussions/545)
> 
> [@Koenkk](https://github.com/Koenkk) is it beta? I can't see 20250321 version in releases.

it is indeed