Koenkk/zigbee2mqtt #26843: Stale device/timouts - GIDEALED ZC05M is recognized as Tuya TS0503B

m-wo, 2025-03-24:
### What happened?

I have a GIDEALED Zigbee 3.0 LED Controller (ZC05M) that works for a while and then cannot be controlled anymore - Any command via Z2M ends up in a timeout, interviewing the device does not work, too. Only thing helps is a full reset (pretty annoying since it is not easily accessible in my case).

I found a similar thread describing the same problem, unresolved: https://github.com/Koenkk/zigbee2mqtt/issues/23953
My device is also detected as Tuya  TS0505B_1

Timeout msg:
z2m: Publish 'set' 'state' to 'Bad-Spiegel' failed: 'Error: ZCL command 0xa4c1385a21ddd7d8/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Timeout - 64394 - 1 - 253 - 6 - 11 after 10000ms)'

### How to reproduce it (minimal and precise)

Not directly reproducable - it works a few days, then the first timeouts appear, sometimes the device still can be interviewed and is working again after that until the next timeouts

z2m: Publish 'set' 'state' to 'Bad-Spiegel' failed: 'Error: ZCL command 0xa4c1385a21ddd7d8/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Timeout - 64394 - 1 - 253 - 6 - 11 after 10000ms)'

### Zigbee2MQTT version

2.1.3-1

### Adapter firmware version

6.10.3.0

### Adapter

Sonoff ZBDongle-E 3.0

### Setup

Add-on on HA running on Intel NUC via proxmox

chris-1243, 2025-03-24:
> 6.10.3.0

You are using a quite old firmware on your Dongle-E. Latest release is 8.0.2. Also, `ezsp` driver is deprecated and has been replaced by `ember`. I would first start to update and then check if that helps

m-wo, 2025-03-24:
Thanks for pointing it out - I just updated the firmware now to 7.4.4 and switched to ember - I am able to control the GIDEALED again, it sill is showing up as Tuya TS0503B device. Let´s hope this permanentely fixes it

closed 2025-03-24 by m-wo