Koenkk/zigbee2mqtt #26846: Cannot set specific values for external sensor for sonoff trvzb

mohoo, 2025-03-24:
### What happened?

https://github.com/Koenkk/zigbee-herdsman-converters/pull/8873
There are certainly float values that are not transmitted or grounded correctly (see logs). 

An attempt is made to set the value 19.4°C as an external sensor value. The value ‘1939.9999999999998’ is displayed in the logs. Later the device then reports the value 19.3°C

```shell
[2025-03-24 09:52:13] debug:    z2m:mqtt: Received MQTT message on 'zigbee2mqtt/Arbeitszimmer Heizung/set' with data '{"external_temperature_input":19.4}'
[2025-03-24 09:52:13] debug:    z2m: Publishing 'set' 'external_temperature_input' to 'Arbeitszimmer Heizung'
[2025-03-24 09:52:13] debug:    zh:controller:endpoint: ZCL command 0x38398ffffe068565/1 customSonoffTrvzb.write({"externalTemperatureInput":1939.9999999999998}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false})
[2025-03-24 09:52:13] debug:    zh:controller:endpoint: Request Queue (0x38398ffffe068565/1): send request
[2025-03-24 09:52:13] debug:    zh:zstack: sendZclFrameToEndpointInternal 0x38398ffffe068565:4288/1 (0,0,1)
[2025-03-24 09:52:13] debug:    zh:zstack:znp: --> SREQ: AF - dataRequest - {"dstaddr":4288,"destendpoint":1,"srcendpoint":1,"clusterid":64529,"transid":203,"options":0,"radius":30,"len":8,"data":{"type":"Buffer","data":[16,205,2,13,96,41,147,7]}}
[2025-03-24 09:52:13] debug:    zh:zstack:unpi:writer: --> frame [254,18,36,1,192,16,1,1,17,252,203,0,30,8,16,205,2,13,96,41,147,7,216]
[2025-03-24 09:52:13] debug:    zh:zstack:unpi:parser: <-- [254,1,100,1,0,100]
[2025-03-24 09:52:13] debug:    zh:zstack:unpi:parser: --- parseNext [254,1,100,1,0,100]
[2025-03-24 09:52:13] debug:    zh:zstack:unpi:parser: --> parsed 1 - 3 - 4 - 1 - [0] - 100
[2025-03-24 09:52:13] debug:    zh:zstack:znp: <-- SRSP: AF - dataRequest - {"status":0}
[2025-03-24 09:52:13] debug:    zh:zstack:unpi:parser: --- parseNext []
[2025-03-24 09:52:13] debug:    zh:zstack:unpi:parser: <-- [254,3,68,128,0,1,203,13]
[2025-03-24 09:52:13] debug:    zh:zstack:unpi:parser: --- parseNext [254,3,68,128,0,1,203,13]
[2025-03-24 09:52:13] debug:    zh:zstack:unpi:parser: --> parsed 3 - 2 - 4 - 128 - [0,1,203] - 13
[2025-03-24 09:52:13] debug:    zh:zstack:znp: <-- AREQ: AF - dataConfirm - {"status":0,"endpoint":1,"transid":203}
[2025-03-24 09:52:13] debug:    zh:zstack:unpi:parser: --- parseNext []
[2025-03-24 09:52:13] debug:    zh:zstack:unpi:parser: <-- [254,24,68,129,0,0,17,252,192,16,1,1,0,153,0,121,71,188,0,0,4,24,205,4,0,177,198,28,69]
[2025-03-24 09:52:13] debug:    zh:zstack:unpi:parser: --- parseNext [254,24,68,129,0,0,17,252,192,16,1,1,0,153,0,121,71,188,0,0,4,24,205,4,0,177,198,28,69]
[2025-03-24 09:52:13] debug:    zh:zstack:unpi:parser: --> parsed 24 - 2 - 4 - 129 - [0,0,17,252,192,16,1,1,0,153,0,121,71,188,0,0,4,24,205,4,0,177,198,28] - 69
[2025-03-24 09:52:13] debug:    zh:zstack:znp: <-- AREQ: AF - incomingMsg - {"groupid":0,"clusterid":64529,"srcaddr":4288,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":153,"securityuse":0,"timestamp":12339065,"transseqnumber":0,"len":4,"data":{"type":"Buffer","data":[24,205,4,0]}}
[2025-03-24 09:52:13] debug:    zh:controller: Received payload: clusterID=64529, address=4288, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=153, frame={"header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":205,"commandIdentifier":4},"payload":[{"status":0}],"command":{"ID":4,"name":"writeRsp","parameters":[{"name":"status","type":32},{"name":"attrId","type":33,"conditions":[{"type":"statusNotEquals","value":0}]}]}}
[2025-03-24 09:52:13] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Arbeitszimmer Heizung', payload '{"battery":98,"child_lock":"UNLOCK","closing_steps":347,"external_temperature_input":19.3,"frost_protection_temperature":4,"idle_steps":291,"last_seen":"2025-03-24T09:52:13.329Z","linkquality":153,"local_temperature":19.3,"local_temperature_calibration":0,"occupied_heating_setpoint":20,"open_window":"OFF","running_state":"heat","system_mode":"heat","temperature_accuracy":-0.3,"temperature_sensor_select":"external","update":{"installed_version":4864,"latest_version":4864,"state":"idle"},"update_available":true,"valve_closing_degree":95,"valve_closing_limit_voltage":2862,"valve_motor_running_voltage":1277,"valve_opening_degree":12,"valve_opening_limit_voltage":2780}'
``` 

### What did you expect to happen?

I expect that the value 19.4 will also be accepted. This problem does not occur when I transmit the value 19.3 (see logs).

```shell
[2025-03-24 09:54:43] debug:    z2m:mqtt: Received MQTT message on 'zigbee2mqtt/Arbeitszimmer Heizung/set' with data '{"external_temperature_input":19.3}'
[2025-03-24 09:54:43] debug:    z2m: Publishing 'set' 'external_temperature_input' to 'Arbeitszimmer Heizung'
[2025-03-24 09:54:43] debug:    zh:controller:endpoint: ZCL command 0x38398ffffe068565/1 customSonoffTrvzb.write({"externalTemperatureInput":1930}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false})
[2025-03-24 09:54:43] debug:    zh:controller:endpoint: Request Queue (0x38398ffffe068565/1): send request
[2025-03-24 09:54:43] debug:    zh:zstack: sendZclFrameToEndpointInternal 0x38398ffffe068565:4288/1 (0,0,1)
[2025-03-24 09:54:43] debug:    zh:zstack:znp: --> SREQ: AF - dataRequest - {"dstaddr":4288,"destendpoint":1,"srcendpoint":1,"clusterid":64529,"transid":214,"options":0,"radius":30,"len":8,"data":{"type":"Buffer","data":[16,215,2,13,96,41,138,7]}}
[2025-03-24 09:54:43] debug:    zh:zstack:unpi:writer: --> frame [254,18,36,1,192,16,1,1,17,252,214,0,30,8,16,215,2,13,96,41,138,7,198]
[2025-03-24 09:54:43] debug:    zh:zstack:unpi:parser: <-- [254,1,100,1,0,100]
[2025-03-24 09:54:43] debug:    zh:zstack:unpi:parser: --- parseNext [254,1,100,1,0,100]
[2025-03-24 09:54:43] debug:    zh:zstack:unpi:parser: --> parsed 1 - 3 - 4 - 1 - [0] - 100
[2025-03-24 09:54:43] debug:    zh:zstack:znp: <-- SRSP: AF - dataRequest - {"status":0}
[2025-03-24 09:54:43] debug:    zh:zstack:unpi:parser: --- parseNext []
[2025-03-24 09:54:43] debug:    zh:zstack:unpi:parser: <-- [254,3,68,128,0,1,214,16]
[2025-03-24 09:54:43] debug:    zh:zstack:unpi:parser: --- parseNext [254,3,68,128,0,1,214,16]
[2025-03-24 09:54:43] debug:    zh:zstack:unpi:parser: --> parsed 3 - 2 - 4 - 128 - [0,1,214] - 16
[2025-03-24 09:54:43] debug:    zh:zstack:znp: <-- AREQ: AF - dataConfirm - {"status":0,"endpoint":1,"transid":214}
[2025-03-24 09:54:43] debug:    zh:zstack:unpi:parser: --- parseNext []
[2025-03-24 09:54:43] debug:    zh:zstack:unpi:parser: <-- [254,24,68,129,0,0,17,252,192,16,1,1,0,153,0,206,132,91,0,0,4,24,215,4,0,177,198,28,204]
[2025-03-24 09:54:43] debug:    zh:zstack:unpi:parser: --- parseNext [254,24,68,129,0,0,17,252,192,16,1,1,0,153,0,206,132,91,0,0,4,24,215,4,0,177,198,28,204]
[2025-03-24 09:54:43] debug:    zh:zstack:unpi:parser: --> parsed 24 - 2 - 4 - 129 - [0,0,17,252,192,16,1,1,0,153,0,206,132,91,0,0,4,24,215,4,0,177,198,28] - 204
[2025-03-24 09:54:43] debug:    zh:zstack:znp: <-- AREQ: AF - incomingMsg - {"groupid":0,"clusterid":64529,"srcaddr":4288,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":153,"securityuse":0,"timestamp":5997774,"transseqnumber":0,"len":4,"data":{"type":"Buffer","data":[24,215,4,0]}}
[2025-03-24 09:54:43] debug:    zh:controller: Received payload: clusterID=64529, address=4288, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=153, frame={"header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":215,"commandIdentifier":4},"payload":[{"status":0}],"command":{"ID":4,"name":"writeRsp","parameters":[{"name":"status","type":32},{"name":"attrId","type":33,"conditions":[{"type":"statusNotEquals","value":0}]}]}}
[2025-03-24 09:54:43] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Arbeitszimmer Heizung', payload '{"battery":98,"child_lock":"UNLOCK","closing_steps":347,"external_temperature_input":19.3,"frost_protection_temperature":4,"idle_steps":291,"last_seen":"2025-03-24T09:54:43.528Z","linkquality":153,"local_temperature":19.3,"local_temperature_calibration":0,"occupied_heating_setpoint":20,"open_window":"OFF","running_st [... cut ...] _degree":95,"valve_closing_limit_voltage":2862,"valve_motor_running_voltage":1277,"valve_opening_degree":12,"valve_opening_limit_voltage":2780}'
``` 

### How to reproduce it (minimal and precise)

Try setting the value 19.4 or 20.4 for the external sensor. You will receive the value 19.3 or 20.3 back from the device.

### Zigbee2MQTT version

 2.1.3 commit: ba337bd

### Adapter firmware version

20240710

### Adapter

Sonoff zb dongle

### Setup

plain and Home Assistant Core as Docke Container

marazmarci, 2025-03-30:
I fixed this in https://github.com/Koenkk/zigbee-herdsman-converters/pull/9072 and it should be already available on the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html).

It will be included in the next Z2M release (which will be called 2.1.4 or 2.2.0, I'm not sure).

@mohoo, please test it and report back if the issue is really fixed for you!

mohoo, 2025-03-31:
I tested it with `zigbee-herdsman-converters` version `23.14.0` and everything works now. Thank you very much 👍

closed 2025-03-31 by mohoo