Koenkk/zigbee2mqtt #26848: z2m: Failed to check if update available for 'device_name' (fetch failed)

Vitek-1977, 2025-03-24:
### What happened?

After "check for update" Z2M reports error "z2m: Failed to check if update available for 'device_name' (fetch failed)"

### What did you expect to happen?

Check if there is update available

### Zigbee2MQTT version

2.1.3

### Adapter firmware version

	20240710

### Adapter

SLZB-06

### Setup

Proxmox VM, Docker

### Debug log

[container-logs-2025-03-24T12-24-51.zip](https://github.com/user-attachments/files/19429622/container-logs-2025-03-24T12-24-51.zip)

Nerivec, 2025-03-25:
I see you had a `ECONNRESET`, that combined with the `fetch` failing would suggest some instability in your local network (adapter is crashing, can't fetch resources from the web).

Vitek-1977, 2025-03-25:
> I see you had a few `ECONNRESET`, that combined with the `fetch` failing would suggest some instability in your local network (adapter is crashing, can't fetch resources from the web).

These would be forced restarts of the dongle when trying latest zigbee firmware and then reverting back to the original - it didnt work at all and the version disappeared from the menu already :-)

closed 2025-03-27 by Vitek-1977