Koenkk/zigbee2mqtt #26852: [New device support]: Shelly 1PM Mini Gen4

marce14-git, 2025-03-24:
### Link

https://www.shelly.com/products/shelly-1pm-mini-gen4

### Database entry

{"id":230,"type":"Router","ieeeAddr":"0x7c2c67fffe7b8fd4","nwkAddr":56139,"manufId":5264,"manufName":"Shelly","powerSource":"Mains (single phase)","modelId":"Mini1PM","epList":[1,239,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,3,4,5,6,2820,1794],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"modelId":"Mini1PM","manufacturerName":"Shelly","powerSource":0,"zclVersion":8}},"genOnOff":{"attributes":{"onOff":0}},"haElectricalMeasurement":{"attributes":{"acPowerDivisor":100,"acPowerMultiplier":1,"acCurrentDivisor":100,"acCurrentMultiplier":1,"acVoltageDivisor":100,"acVoltageMultiplier":1,"acFrequency":5002,"activePower":0,"rmsCurrent":0,"rmsVoltage":22514,"acFrequencyMultiplier":1,"acFrequencyDivisor":100}},"seMetering":{"attributes":{"divisor":1000000,"multiplier":1,"currentSummReceived":0,"status":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0031d80810","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00124b0031d80810","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00124b0031d80810","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":65000,"repChange":1},{"cluster":2820,"attrId":1291,"minRepIntval":10,"maxRepIntval":65000,"repChange":500},{"cluster":2820,"attrId":1288,"minRepIntval":10,"maxRepIntval":65000,"repChange":5},{"cluster":2820,"attrId":1285,"minRepIntval":10,"maxRepIntval":65000,"repChange":500}],"meta":{}},"239":{"profId":49153,"epId":239,"devId":8193,"inClusterList":[64513,64514],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"zclVersion":8,"interviewCompleted":true,"meta":{},"lastSeen":1742826019462}

### Zigbee2MQTT version

2.1.3 

### Comments

I followed the Guide at the Add new device page and created the automatic external converter

### External definition

```shell
const m = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['Mini1PM'],
    model: 'Mini1PM',
    vendor: 'Shelly',
    description: 'Automatically generated definition',
    extend: [m.deviceEndpoints({"endpoints":{"1":1,"239":239}}), m.onOff({"powerOnBehavior":false}), m.electricityMeter()],
    meta: {"multiEndpoint":true},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

[image] 
These are the states, I briefly saw something like switch_mode: toggle but i cant seem to reproduce it

[image] 
This are the Exposes, there could be much more, the device is able to switch to detached mode, has an input etc.etc.

The device reports very frequently, like every second but Last seen is always saying 20seconds

marce14-git, 2025-03-24:
after restarting i get this error:

Failed to configure 'Büro Lichtschalter', attempt 2 (Error: ZCL command 0x7c2c67fffe7b6dc4/1 seMetering.configReport([{"minimumReportInterval":10,"maximumReportInterval":65000,"reportableChange":100000,"attribute":"currentSummDelivered"}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'UNREPORTABLE_ATTRIBUTE') at Endpoint.checkStatus (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:349:28) at Endpoint.zclCommand (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:954:26) at processTicksAndRejections (node:internal/process/task_queues:105:5) at Endpoint.configureReporting (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:686:9) at setupAttributes (/app/node_modules/.pnpm/zigbee-herdsman-converters@23.2.0/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:159:17) at result.configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@23.2.0/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:2016:29) at Object.configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@23.2.0/node_modules/zigbee-herdsman-converters/src/index.ts:357:21) at Configure.configure (/app/lib/extension/configure.ts:124:13) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:204:17))

rionshin, 2025-03-24:
This screenshot is provided by one of  Shelly team members, who are  already running their PM devices in Z2M , it may help . I hope they expose Zigbee clusters in their documentation (soon).

@Koenkk  it will be great if you can add Shelly as vendor , as they plan to expand their portfolio with Zigbee devices now , and all Gen4 will be Zigbee and Gen3 as well with new firmware release .  
Gen 4 devices are here https://www.shelly.com/collections/all-products

[image]

marce14-git, 2025-03-25:
> > This screenshot is provided by a friend, who are already running their PM devices in Z2M , it may help . I hope they expose Zigbee clusters in their documentation (soon).
> > [@Koenkk](https://github.com/Koenkk) it will be great if you can add Shelly as vendor , as they plan to expand their portfolio with Zigbee devices now , and all Gen4 will be Zigbee and Gen3 as well with new firmware release . Gen 4 devices are here https://www.shelly.com/collections/all-products
> 
> Where is the information that GEN3 will get Zigbee?

Gen 3 wont get Zigbee, Gen 3 will gett Matter over Wifi

MeisterQ, 2025-03-25:
> Gen 3 wont get Zigbee, Gen 3 will gett Matter over Wifi

Gen4 has zigbee.

FrodoVDR, 2025-03-26:
Knowledge base for Shelly 1pm mini gen4:
[](https://kb.shelly.cloud/knowledge-base/shelly-1pm-mini-gen4)

rionshin, 2025-03-26:
> Knowledge base for Shelly 1pm mini gen4: https://kb.shelly.cloud/knowledge-base/shelly-1pm-mini-gen4

Still not information for the Zigbee clusters.

rionshin, 2025-03-26:
> Where is the information that GEN3 will get Zigbee?

My mistake it will be matter over wifi, no zigbee.

MeisterQ, 2025-03-26:
[image]

Maybe this helps? I dont know. Its the information i receive from IObroker zigbee adapter. Its a Gen4 1PM Mini

MiAutomations, 2025-03-27:
> > Knowledge base for Shelly 1pm mini gen4: https://kb.shelly.cloud/knowledge-base/shelly-1pm-mini-gen4
> 
> Still not information for the Zigbee clusters.

Hello I have the mini 1 , Mini PM and the models 1 and 1PM 

in order to try help this implementation, please let me know where I can try to find this information? 

Tks

Mosher23, 2025-03-28:
I hope shelly will add input state reporting over ZigBee. For example to use it in detached mode

marce14-git, 2025-03-28:
```
const m = require('zigbee-herdsman-converters/lib/modernExtend');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const e = exposes.presets;

const definition = {
    zigbeeModel: ['Mini1PM'],
    model: 'Mini1PM',
    vendor: 'Shelly',
    description: 'Automatically generated definition',
    extend: [
        m.deviceEndpoints({"endpoints": {"1": 1, "239": 239}}),
        // Removed m.onOff() here to prevent duplication
    ],
    exposes: [
        // Manually define the onOff state to ensure it's first
        {
            "access": 7,
            "category": "state",
            "description": "On/off state of the switch",
            "label": "State",
            "name": "state",
            "property": "state",
            "type": "binary",
            "value_off": "OFF",
            "value_on": "ON",
            "value_toggle": "TOGGLE"
        },

        // Add the rest of the electricityMeter exposes with '_1' suffix
        ...m.electricityMeter().exposes.map(expose => {
            // Custom mapping for '_1' suffix
            if (expose.property === 'voltage') expose.property = 'voltage_1';
            if (expose.property === 'current') expose.property = 'current_1';
            if (expose.property === 'power') expose.property = 'power_1';
            if (expose.property === 'energy') expose.property = 'energy_1';
            if (expose.property === 'ac_frequency') expose.property = 'ac_frequency_1';
            return expose;
        })
    ],
    meta: { multiEndpoint: true },
};

module.exports = definition;
```

I tried using ChatGPT to create a working external converter, it kinda works but its probably not the most elegant solution

Mosher23, 2025-03-29:
> ```
> 
> const m = require('zigbee-herdsman-converters/lib/modernExtend');
> const exposes = require('zigbee-herdsman-converters/lib/exposes');
> const e = exposes.presets;
> 
> const definition = {
>     zigbeeModel: ['Mini1PM'],
>     model: 'Mini1PM',
>     vendor: 'Shelly',
>     description: 'Automatically generated definition',
>     extend: [
>         m.deviceEndpoints({"endpoints": {"1": 1, "239": 239}}),
>         m.onOff({"powerOnBehavior": false}),
>     ],
>     exposes: [
>   
> 
>         // Add the rest of the electricityMeter exposes with '_1' suffix
>         ...m.electricityMeter().exposes.map(expose => {
>             // Custom mapping for '_1' suffix
>             if (expose.property === 'voltage') expose.property = 'voltage_1';
>             if (expose.property === 'current') expose.property = 'current_1';
>             if (expose.property === 'power') expose.property = 'power_1';
>             if (expose.property === 'energy') expose.property = 'energy_1';
>             if (expose.property === 'ac_frequency') expose.property = 'ac_frequency_1';
>             return expose;
>         })
>     ],
>     meta: { multiEndpoint: true },
> };
> 
> module.exports = definition;
> ```
> 
> I tried using ChatGPT to create a working external converter, it kinda works but its probably not the most elegant solution

Could you please share what entities you see in Exposed tab?

marce14-git, 2025-03-29:
[image]

marce14-git, 2025-03-29:
> hope shelly will add input state reporting over ZigBee. For example to use it in detached mode

Shelly support said not possible, hope they add it in the future 

For now I’m using the Shelly with Wifi which exposes input state and control the lights over zigbee using Homeassistant

On the Shelly is a script to run in detached mode and  fallback to toggle mode if Homeassistant is offline

> [image]

Mosher23, 2025-03-29:
> > On the Shelly is a script to run in detached mode and  fallback to toggle mode if Homeassistant is offline
> >

Could you please share this script?
Thanks

marce14-git, 2025-03-29:
> > > On the Shelly is a script to run in detached mode and  fallback to toggle mode if Homeassistant is offline
> 
> Could you please share this script? Thanks

https://pastebin.com/faCH8UDN

Change IP, HA Token and desired state of the HA sensor

MeisterQ, 2025-03-29:
Btw, i think this request has to be placed in the herdsman converter repository, right? I dont think @Koenkk  will respond here.

Koenkk, 2025-03-30:
Could you check if the issue is fixed with the following [external converter](https://gist.github.com/Koenkk/9139da2d05982fa633a9e56463db8a75): 

- save this as file next to `configuration.yaml` in the `external_converters` directory (create if it doesn't exist yet) as `ext_converter.js` (so `external_converters/ext_converter.js`)
- start z2m, 
- in the frontend, check if the description of this device now ends with `CUSTOM` (this indicates the external converter has been loaded correctly)
- check if issue is fixed

marce14-git, 2025-03-30:
> Could you check if the issue is fixed with the following [external converter](https://gist.github.com/Koenkk/9139da2d05982fa633a9e56463db8a75):
>
> * save this as file next to `configuration.yaml` in [...]

The converter is applied and device is shown as Shelly 1 Mini instead of Shelly 1PM Mini, but doesnt solve the issue

States when connected to a 7W bulb: 
`{
    "ac_frequency": 50.04,
    "ac_frequency_1": 5006,
    "current": null,
    "current_1": 0.05,
    "energy": null,
    "energy_1": 0.003084,
    "last_seen": "2025-03-30T21:26:55+02:00",
    "linkquality": 138,
    "power": null,
    "power_1": 6.97,
    "produced_energy": 0,
    "produced_energy_1": 0,
    "state": "ON",
    "state_1": "ON",
    "voltage": null,
    "voltage_1": 223.49
}`

[image]

[image]

Koenkk, 2025-03-30:
Could you provide the debug log when removing and attaching the 7w load?

See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

marce14-git, 2025-03-30:
> Could you provide the debug log when removing and attaching the 7w load?
> 
> See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

I am sorry, after repairing the device everything works 

`{
    "last_seen": "2025-03-30T21:56:21+02:00",
    "linkquality": 174,
    "current": 0.05,
    "energy": 0.01,
    "power": 7.05,
    "state": "ON",
    "voltage": 226.57,
    "ac_frequency": 5008,
    "produced_energy": 0
}`

JoSeBu1, 2025-03-31:
@marce14-git Can you confirm that all features works by zigbee? I will receive soon 15 Mini 1 to start with my home domotization. Thanks

marce14-git, 2025-03-31:
> [@marce14-git](https://github.com/marce14-git) Can you confirm that all features works by zigbee? I will receive soon 15 Mini 1 to start with my home automation and I want to use it with HA and Zigbee2MQTT. Thanks

With the new converter all features work

But the 1PM Mini (don’t know about the 1Mini)?reports every 5seconds, and does not seem to abide to increasing the reporting Interval 

I have a fairly stable network of 80 devices and a cc2652p and even if I only have 1 or 2 of them, the network gets noticeably slower
Compared to all other devices , the Shelly’s spam the network pretty much

marce14-git, 2025-03-31:
> [@marce14-git](https://github.com/marce14-git) does it show input state with the converter from Koenkk?

Input state is currently not supported (statement from Shelly support)

You can create a ticket on the Shelly website with future suggestions, hopefully they add this feature

Mosher23, 2025-03-31:
I have submitted a ticket.

marce14-git, 2025-03-31:
> Could you check if the issue is fixed with the following [external converter](https://gist.github.com/Koenkk/9139da2d05982fa633a9e56463db8a75):
>
> * save this as file next to `configuration.yaml` in [...]

Converter still works, but everytime the POWER changes I get this error, however the value is displayed correctly:

`zhc: Failed to apply calibration to 'power': 'power_calibration' is not a number, got string () `

Mosher23, 2025-03-31:
> After unbinding the seMetering cluster in the Bind tab, my Shelly stopped sending reports every 5 seconds.

How do you unbind only one cluster?

MeisterQ, 2025-03-31:
I made a converter for this shelly and it works "pretty" good.

```
// 1) Benötigte Module laden:
const fz = require('zigbee-herdsman-converters/dist/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/dist/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/dist/lib/exposes');
const reporting = require('zigbee-herdsman-converters/dist/lib/reporting');
const e = exposes.presets;
const ea = exposes.access;

module.exports = {
    fingerprint: [
        {
            modelID: 'Mini1PM',
            manufacturerName: 'Shelly',
        },
    ],
    model: 'Mini1PM',
    vendor: 'Shelly',
    description: 'Shelly Mini1PM - Zigbee In-Wall Switch with Power Measurement',
    fromZigbee: [
        fz.on_off,                  // Schaltzustände (genOnOff)
        fz.electrical_measurement,  // Elektrische Messwerte (haElectricalMeasurement)
        fz.metering,                // Energieverbrauch (seMetering)
    ],
    toZigbee: [
        tz.on_off, // Ein-/Ausschalten
    ],
    exposes: [
        e.switch(),
        e.power(),
        e.current(),
        e.voltage(),
        e.energy(),
    ],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'haElectricalMeasurement', 'seMetering']);

        // Reporting für den haElectricalMeasurement-Cluster konfigurieren
        try {
            await reporting.configureReporting(endpoint, {
                cluster: 'haElectricalMeasurement',
                attribute: 'activePower',
                minimumReportInterval: 10,
                maximumReportInterval: 600,
                reportableChange: 1,
            });
            await reporting.configureReporting(endpoint, {
                cluster: 'haElectricalMeasurement',
                attribute: 'rmsVoltage',
                minimumReportInterval: 10,
                maximumReportInterval: 600,
                reportableChange: 1,
            });
            await reporting.configureReporting(endpoint, {
                cluster: 'haElectricalMeasurement',
                attribute: 'rmsCurrent',
                minimumReportInterval: 10,
                maximumReportInterval: 600,
                reportableChange: 1,
            });
        } catch (error) {
            logger.error('Failed to configure reporting for haElectricalMeasurement', error);
        }
        
        // Reporting für den seMetering-Cluster (Energie) konfigurieren
        try {
            await reporting.configureReporting(endpoint, {
                cluster: 'seMetering',
                attribute: 'currentSummDelivered',
                minimumReportInterval: 10,
                maximumReportInterval: 600,
                reportableChange: 1,
            });
        } catch (error) {
            logger.error('Failed to configure reporting for seMetering', error);
        }
    },
};

```

```
Scalings (divider and multiplier) have to be requested via json, then it works (But via IObroker Zigbee-adapter which is using zhc)

{
   "read",
   {
      "haEnergyMeasurement":"acCurrentMultiplier"
   },
   {
      "haEnergyMesaurement":"acCurrentDivisor"
   },
   {
      "haEnergyMeasurement":"acPowerMultiplier"
   },
   {
      "haEnergyMesaurement":"acPowerDivisor"
   },
   {
      "haEnergyMeasurement":"acVoltageMultiplier"
   },
   {
      "haEnergyMesaurement":"acVoltageDivisor"
   },
   {
      "haEnergyMeasurement":"acFrequencyMultiplier"
   },
   {
      "haEnergyMesaurement":"acFrequencyDivisor"
   }
}
```

Mosher23, 2025-03-31:
> I have submitted a ticket.
My ticket was closed, shortly after without any comments. Afterwards I tried opening new ticket again, these were auto closed instantly.

[image]

marce14-git, 2025-03-31:
> > I have submitted a ticket.
> > My ticket was closed, shortly after without any comments. Afterwards I tried opening new ticket again, these were auto closed instantly.
> 
> <img alt="Image" width="929" src="https://private-user-images.githubusercontent.com/8256425/428736600-6c011975-08ad-4f14-9ac2-7b4af2d33af6.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDM0NDU5MDQsIm5iZiI6MTc0MzQ0NTYwNCwicGF0aCI6Ii84MjU2NDI1LzQyODczNjYwMC02YzAxMTk3NS0wOGFkLTRmMTQtOWFjMi03YjRhZjJkMzNhZjYucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI1MDMzMSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNTAzMzFUMTgyNjQ0WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ODZiODY2MTc4YzMwYWMxY2UzMDUxOTM4NTJmZGVlZDM3ZGE1ZDkzNTE3ZDFlMjkyN2YyYjdlYjEyZmFlODBjMSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.ojbU4lvlPzO7sFTKbGmauC0jeipM46bnrGSyaVLjCsM">

This is normal behaviour, you got an email confirming your proposal…

MeisterQ, 2025-04-01:
Whats the state now? How does the whole definition file looks like? Will it be addet to the herdsman?

Mosher23, 2025-04-01:
> Whats the state now? How does the whole definition file looks like? Will it be addet to the herdsman?

Its already added

MeisterQ, 2025-04-01:
@Mosher23 i couldnt find it in the devices/shelly.ts.

Mosher23, 2025-04-01:
> @Mosher23 i couldnt find it in the devices/shelly.ts.

It should be in dev branch. There was a PR merged few days ago

MeisterQ, 2025-04-01:
> > [@Mosher23](https://github.com/Mosher23) i couldnt find it in the devices/shelly.ts.
> 
> It should be in dev branch. There was a PR merged few days ago

I dont get the exact relation why it should be in zigbee2mqtt and not in zigbee-herdsman-converter. Because devices wont be addet here in the repo.?

MeisterQ, 2025-04-04:
Shelly just released all informations about zigbee. 
[https://shelly-api-docs.shelly.cloud/gen2/Integrations/Zigbee/DeviceFeatures/?fbclid=IwZXh0bgNhZW0CMTAAAR3u7k9BWm7ohRnixlqcWHPo0sQrHYW8MNwEEI8Ntv7iQecv7imLeO9oYIM_aem_RX7tvhb4XFk95FtKSCAgKQ](url)
[https://shelly-api-docs.shelly.cloud/gen2/Integrations/Zigbee/RPCCluster/?fbclid=IwZXh0bgNhZW0CMTAAAR1-_PHVDMTAlWUtfsqNaS2ch5_dZV5vFRFlPkg5INhONUekH8ccD3BTqAo_aem_vzG2C_uQ_rINhqWjZYPjCQ](url)
[https://shelly-api-docs.shelly.cloud/gen2/Integrations/Zigbee/WiFiSetupCluster/?fbclid=IwZXh0bgNhZW0CMTAAAR3HhSYN9Pcg0iUw4QJzEUitHE4eIN1KbBzpqG_NMrlnCaOCGFHiWuravXY_aem_r8DshCgfxVitJGG35Fvh2A](url)

Here you could see the input state of the switch input etc etc. So you could configure everything and read all informations out of the devices via zigbee.

How can we integrate it?

Mosher23, 2025-04-04:
@MeisterQ I don't see directly Input state in their documentation. Is it going over RPC cluster?

MeisterQ, 2025-04-04:
> [@MeisterQ](https://github.com/MeisterQ) I don't see directly Input state in their documentation. Is it going over RPC cluster?

Yes, its with the RPC cluster. There you can get any informations you need. The question is how to implement this feature

MeisterQ, 2025-04-04:
> > > > > [@MeisterQ](https://github.com/MeisterQ) I don't see directly Input state in their documentation. Is it going over RPC cluster?
> > > > 
> > > > 
> > > > Yes, its with the RPC cluster. There you can get any informations you need. The question is how to implement this feature
> > > 
> > > 
> > > I am not skilled enouigh to make it work, but if i read it correct alle the Data in the RPC and Wifi Cluster are PULLED at the request of user, so we can GET the input state, but only by asking for it, it will not be published automatic when the input state changes, so the feature is pretty useless...
> > > however, maybe it is possible over the reporting tab to specify the exact attribute and get the state automatically published
> > 
> > 
> > Well. Maybe we / someone can extend this converter for like internal temperature or input state. this shouldnt be that complex, right? Dont know if [@Koenkk](https://github.com/Koenkk) can do it or someone else has to do it?
> 
> With the RPC cluster you need to send a request to get the input state. If your shelly is in detached mode and you want to control another device with the input state, you constantly need to request the current state. That’s not a viable option.

I made a suggestion at shelly to send this inputstate automatically when the state changes. Lets see what they will say.

phil-schneider, 2025-04-05:
Thanks guys for all your work. I am really looking forward for the shelly devices to me fully supported.
I think this will be a game changes for Shelly, Z2M and HA.

Trickytree1984, 2025-04-07:
Ive recently received 3 of these. I've just wired the first one up behind an FCU on the load side, to switch a towel rail. Firstly i paired this with Matter and it worked fine. I then switched it to Zigbee mode and paired with Z2M. I have noticed that the power reading is way off on Z2M but accurate on Matter (its an 800w towel rail)

[image]

emarref, 2025-04-09:
I am also using a Shelly 1PM Gen4 with Zigbee2MQTT and am seeing unexpected values.

My hot water cylinder reports using 14A on a 240V supply which I would have expected to show 3,360kW.

[image]

Additionally, on a separate Shelly (identical model), when an oil heater switches on in my office I see negative power.

[image]

Mosher23, 2025-04-10:
> Ive recently received 3 of these. I've just wired the first one up behind an FCU on the load side, to switch a towel rail. Firstly i paired this with Matter and it worked fine. I then switched it to [...]

I see same issue with mines. They do not update often. If you press on refresh button for corresponding values it will update the values to the correct ones. Some time I see values that are 18 hours old.

I think there must be solution for that.

MeisterQ, 2025-04-11:
> With the new beta firmware 1.6.0-beta1, currentSumDelivered is now reportable.

I can agree with that. Its not "null" anymore. But if your energy consumption is below 1kWh like 120Wh, it says 0kWh. Should be floatingpoint or report in Wh because shellys usually deliver energy values in Wh.

Mosher23, 2025-04-11:
> > > With the new beta firmware 1.6.0-beta1, currentSumDelivered is now reportable.
> > 
> > 
> > I can agree with that. Its not "null" anymore. But if your energy consumption is below 1kWh like 120Wh, it says 0kWh. Should be floatingpoint or report in Wh because shellys usually deliver energy values in Wh.
> 
> Have you attempted adjusting the energy precision under the “Settings (Specific)” tab? I am only able to test it myself in a few hours.

There is no energy precision adjustment in that tab available

Mosher23, 2025-04-11:
I do not use any external converter. The one which comes with Z2M

Mosher23, 2025-04-11:
> > I do not use any external converter. The one which comes with Z2M
> 
> That's probably why. Afaik the internal one doesn't expose energy since it wasn't reportable.

Maybe you can make a pull request to add it into a Z2M?

marce14-git, 2025-05-03:
I just made a pull request to activate energy metering with the new shellys [https://github.com/Koenkk/zigbee-herdsman-converters/pull/9278](url)

still waiting on information about the new rpc cluster though

xoatrash, 2025-05-14:
Any update regarding power reporting? Mine is a negative value and way off. Instead off 400W it shows around minus 250W.

MisterJ49, 2025-05-22:
I have the same problems. The power is wrong. I have 30 W for 9 A of intensity...

and I'm waiting for an update to have the energy consumed.

[image]
[image]

MisterJ49, 2025-05-22:
Do you have to go through Matter to update the firmware? Or is the update done automatically on Z2M?
I don't see the firmware version on Z2M

MisterJ49, 2025-06-03:
They added the energy ratio, but the power ratio doesn't seem to have been corrected.

alu-ka, 2025-06-26:
Looks like values above 255 get negative:

[image]

Same device, values going over WIFI to Home Assistant and values going over ZigBee2MQQT to Home Assistant.

Koenkk, 2025-06-27:
@alu-ka could you provide the debug log when less than < 255 power is drawn (correct behaviour) and more than 255 > (incorrect behaviour)?

See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

alu-ka, 2025-06-28:
Is this the information you are looking for:
`[2025-06-28 10:42:50] debug: 	zh:controller:endpoint: ZCL command 0x7c2c67fffe6923f0/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false})
[2025-06-28 10:42:50] debug: 	zh:zstack: sendZclFrameToEndpointInternal 0x7c2c67fffe6923f0:15471/1 (0,0,1)
[2025-06-28 10:42:50] debug: 	zh:zstack:znp: --> SREQ: AF - dataRequest - {"dstaddr":15471,"destendpoint":1,"srcendpoint":1,"clusterid":0,"transid":43,"options":0,"radius":30,"len":5,"data":{"type":"Buffer","data":[16,240,0,0,0]}}
[2025-06-28 10:42:50] debug: 	zh:zstack:unpi:writer: --> frame [254,15,36,1,111,60,1,1,0,0,43,0,30,5,16,240,0,0,0,169]
[2025-06-28 10:42:50] debug: 	zh:zstack:unpi:parser: <-- [254,1,100,1,0,100]
[2025-06-28 10:42:50] debug: 	zh:zstack:unpi:parser: --- parseNext [254,1,100,1,0,100]
[2025-06-28 10:42:50] debug: 	zh:zstack:unpi:parser: --> parsed 1 - 3 - 4 - 1 - [0] - 100
[2025-06-28 10:42:50] debug: 	zh:zstack:znp: <-- SRSP: AF - dataRequest - {"status":0}
[2025-06-28 10:42:50] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2025-06-28 10:42:50] debug: 	zh:zstack:unpi:parser: <-- [254,3,68,128,0,1,43,237]
[2025-06-28 10:42:50] debug: 	zh:zstack:unpi:parser: --- parseNext [254,3,68,128,0,1,43,237]
[2025-06-28 10:42:50] debug: 	zh:zstack:unpi:parser: --> parsed 3 - 2 - 4 - 128 - [0,1,43] - 237
[2025-06-28 10:42:50] debug: 	zh:zstack:znp: <-- AREQ: AF - dataConfirm - {"status":0,"endpoint":1,"transid":43}
[2025-06-28 10:42:50] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2025-06-28 10:42:50] debug: 	zh:zstack:unpi:parser: <-- [254,28,68,129,0,0,0,0,111,60,1,1,0,69,0,17,82,102,0,0,8,24,240,1,0,0,0,32,8,111,60,29,109]
[2025-06-28 10:42:50] debug: 	zh:zstack:unpi:parser: --- parseNext [254,28,68,129,0,0,0,0,111,60,1,1,0,69,0,17,82,102,0,0,8,24,240,1,0,0,0,32,8,111,60,29,109]
[2025-06-28 10:42:50] debug: 	zh:zstack:unpi:parser: --> parsed 28 - 2 - 4 - 129 - [0,0,0,0,111,60,1,1,0,69,0,17,82,102,0,0,8,24,240,1,0,0,0,32,8,111,60,29] - 109
[2025-06-28 10:42:50] debug: 	zh:zstack:znp: <-- AREQ: AF - incomingMsg - {"groupid":0,"clusterid":0,"srcaddr":15471,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":69,"securityuse":0,"timestamp":6705681,"transseqnumber":0,"len":8,"data":{"type":"Buffer","data":[24,240,1,0,0,0,32,8]}}
[2025-06-28 10:42:50] debug: 	zh:controller: Received payload: clusterID=0, address=15471, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=69, frame={"header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":240,"commandIdentifier":1},"payload":[{"attrId":0,"status":0,"dataType":32,"attrData":8}],"command":{"ID":1,"name":"readRsp","parameters":[{"name":"attrId","type":33},{"name":"status","type":32},{"name":"dataType","type":32,"conditions":[{"type":"statusEquals","value":0}]},{"name":"attrData","type":1000,"conditions":[{"type":"statusEquals","value":0}]}]}}
[2025-06-28 10:42:50] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2025-06-28 10:42:50] debug: 	z2m: Received Zigbee message from 'Shelly_1PM_g4', type 'readResponse', cluster 'genBasic', data '{"zclVersion":8}' from endpoint 1 with groupID 0
[2025-06-28 10:42:50] info: 	z2m:mqtt: MQTT publish: topic 'zbdongle-p_zigbee2mqtt/Shelly_1PM_g4', payload '{"ac_frequency":49.92,"current":0.01,"energy":2.2,"last_seen":"2025-06-28T10:42:50+02:00","linkquality":69,"power":2.08,"produced_energy":0,"state":"ON","voltage":229.27}'
[2025-06-28 10:42:50] debug: 	z2m: Successfully pinged 'Shelly_1PM_g4' (attempt 1/2)`
If so, I'm searching for further entries with correct and incorrect values.

Koenkk, 2025-06-29:
I don't see the power going negative in this logging (it's 2.08), please provide the logging when it is positive (like in this one) until it goes negative

alu-ka, 2025-06-29:
@Koenkk here more logging:
```
[2025-06-28 14:09:55] debug: 	zh:controller:endpoint: ZCL command 0x7c2c67fffe6923f0/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false})
[2025-06-28 14:09:55] debug: 	zh:zstack: sendZclFrameToEndpointInternal 0x7c2c67fffe6923f0:15471/1 (0,0,1)
[2025-06-28 14:09:55] debug: 	zh:zstack:znp: --> SREQ: AF - dataRequest - {"dstaddr":15471,"destendpoint":1,"srcendpoint":1,"clusterid":0,"transid":71,"options":0,"radius":30,"len":5,"data":{"type":"Buffer","data":[16,232,0,0,0]}}
[2025-06-28 14:09:55] debug: 	zh:zstack:unpi:writer: --> frame [254,15,36,1,111,60,1,1,0,0,71,0,30,5,16,232,0,0,0,221]
[2025-06-28 14:09:55] debug: 	zh:zstack:unpi:parser: <-- [254,1,100,1,0,100]
[2025-06-28 14:09:55] debug: 	zh:zstack:unpi:parser: --- parseNext [254,1,100,1,0,100]
[2025-06-28 14:09:55] debug: 	zh:zstack:unpi:parser: --> parsed 1 - 3 - 4 - 1 - [0] - 100
[2025-06-28 14:09:55] debug: 	zh:zstack:znp: <-- SRSP: AF - dataRequest - {"status":0}
[2025-06-28 14:09:55] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2025-06-28 14:09:55] debug: 	zh:zstack:unpi:parser: <-- [254,3,68,128,0,1,71,129]
[2025-06-28 14:09:55] debug: 	zh:zstack:unpi:parser: --- parseNext [254,3,68,128,0,1,71,129]
[2025-06-28 14:09:55] debug: 	zh:zstack:unpi:parser: --> parsed 3 - 2 - 4 - 128 - [0,1,71] - 129
[2025-06-28 14:09:55] debug: 	zh:zstack:znp: <-- AREQ: AF - dataConfirm - {"status":0,"endpoint":1,"transid":71}
[2025-06-28 14:09:55] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2025-06-28 14:09:55] debug: 	zh:zstack:unpi:parser: <-- [254,28,68,129,0,0,0,0,111,60,1,1,0,60,0,3,182,191,0,0,8,24,232,1,0,0,0,32,8,252,82,27,216]
[2025-06-28 14:09:55] debug: 	zh:zstack:unpi:parser: --- parseNext [254,28,68,129,0,0,0,0,111,60,1,1,0,60,0,3,182,191,0,0,8,24,232,1,0,0,0,32,8,252,82,27,216]
[2025-06-28 14:09:55] debug: 	zh:zstack:unpi:parser: --> parsed 28 - 2 - 4 - 129 - [0,0,0,0,111,60,1,1,0,60,0,3,182,191,0,0,8,24,232,1,0,0,0,32,8,252,82,27] - 216
[2025-06-28 14:09:55] debug: 	zh:zstack:znp: <-- AREQ: AF - incomingMsg - {"groupid":0,"clusterid":0,"srcaddr":15471,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":60,"securityuse":0,"timestamp":12563971,"transseqnumber":0,"len":8,"data":{"type":"Buffer","data":[24,232,1,0,0,0,32,8]}}
[2025-06-28 14:09:55] debug: 	zh:controller: Received payload: clusterID=0, address=15471, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=60, frame={"header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":232,"commandIdentifier":1},"payload":[{"attrId":0,"status":0,"dataType":32,"attrData":8}],"command":{"ID":1,"name":"readRsp","parameters":[{"name":"attrId","type":33},{"name":"status","type":32},{"name":"dataType","type":32,"conditions":[{"type":"statusEquals","value":0}]},{"name":"attrData","type":1000,"conditions":[{"type":"statusEquals","value":0}]}]}}
[2025-06-28 14:09:55] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2025-06-28 14:09:55] debug: 	z2m: Received Zigbee message from 'Shelly_1PM_g4', type 'readResponse', cluster 'genBasic', data '{"zclVersion":8}' from endpoint 1 with groupID 0
[2025-06-28 14:09:55] info: 	z2m:mqtt: MQTT publish: topic 'zbdongle-p_zigbee2mqtt/Shelly_1PM_g4', payload '{"ac_frequency":49.92,"current":0.01,"energy":2.24,"last_seen":"2025-06-28T14:09:55+02:00","linkquality":60,"power":2.08,"produced_energy":0,"state":"ON","voltage":229.27}'
[2025-06-28 14:09:55] debug: 	z2m: Successfully pinged 'Shelly_1PM_g4' (attempt 1/2)
[2025-06-28 14:11:23] debug: 	zh:controller:endpoint: ZCL command 0x7c2c67fffe6923f0/1 haElectricalMeasurement.defaultRsp({"cmdId":10,"statusCode":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"reservedBits":0,"transactionSequenceNumber":161,"writeUndiv":false})
[2025-06-28 14:11:23] debug: 	zh:zstack: sendZclFrameToEndpointInternal 0x7c2c67fffe6923f0:15471/1 (0,0,1)
[2025-06-28 14:11:23] debug: 	zh:zstack:znp: --> SREQ: AF - dataRequest - {"dstaddr":15471,"destendpoint":1,"srcendpoint":1,"clusterid":2820,"transid":85,"options":0,"radius":30,"len":5,"data":{"type":"Buffer","data":[24,161,11,10,0]}}
[2025-06-28 14:11:23] debug: 	zh:zstack:unpi:writer: --> frame [254,15,36,1,111,60,1,1,4,11,85,0,30,5,24,161,11,10,0,128]
[2025-06-28 14:11:23] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2025-06-28 14:11:23] debug: 	z2m: Received Zigbee message from 'Shelly_1PM_g4', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":6733,"rmsCurrent":31}' from endpoint 1 with groupID 0
[2025-06-28 14:11:23] info: 	z2m:mqtt: MQTT publish: topic 'zbdongle-p_zigbee2mqtt/Shelly_1PM_g4', payload '{"ac_frequency":49.92,"current":0.31,"energy":2.24,"last_seen":"2025-06-28T14:11:23+02:00","linkquality":72,"power":67.33,"produced_energy":0,"state":"ON","voltage":229.27}'
[2025-06-28 14:11:33] debug: 	zh:controller:endpoint: ZCL command 0x7c2c67fffe6923f0/1 haElectricalMeasurement.defaultRsp({"cmdId":10,"statusCode":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"reservedBits":0,"transactionSequenceNumber":162,"writeUndiv":false})
[2025-06-28 14:11:33] debug: 	zh:zstack: sendZclFrameToEndpointInternal 0x7c2c67fffe6923f0:15471/1 (0,0,1)
[... cut ...]
[2025-06-28 14:59:34] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2025-06-28 14:59:34] debug: 	z2m: Received Zigbee message from 'Shelly_1PM_g4', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":-13304}' from endpoint 1 with groupID 0
[2025-06-28 14:59:34] info: 	z2m:mqtt: MQTT publish: topic 'zbdongle-p_zigbee2mqtt/Shelly_1PM_g4', payload '{"ac_frequency":49.92,"current":2.27,"energy":2.54,"last_seen":"2025-06-28T14:59:34+02:00","linkquality":63,"power":-133.04,"produced_energy":0,"state":"ON","voltage":229.27}'
```

Koenkk, 2025-06-29:
@alu-ka do you think that simply dropping the `-` sign would provide the correct values?

xoatrash, 2025-06-29:
> @alu-ka do you think that simply dropping the `-` sign would provide the correct values? I think the firmware of this device is bugged.

No, I dont' think do, because also the provided value is not correct.

alu-ka, 2025-06-29:
@xoatrash @Koenkk from my point of view it looks some byte order or big endian / little endian issue. Is there some basic Information available how the bytes from the zigbee frame are converted to the json property values? 
Perhaps you are right the firmware is buggy... the Shelly guys are new in the zigbee communication.

[32 more replies]