Koenkk/zigbee2mqtt #26857: TS130F _TZ3000_ missing option to change manual switch from toggle to momentary

tsjjBCNBna, 2025-03-25:
### What happened?

TS130F _TZ3000_ devices with missing option to change the external switch from toggle to momentary. It is possible in the Tuya app, but not in Z2M.

### What did you expect to happen?

Option to choose switch type like this: 

[image]

### How to reproduce it (minimal and precise)

Connect TS130F device with Z2M

### Zigbee2MQTT version

2.1.3

### Adapter firmware version

EZSP v12

### Adapter

EZSP v12

### Setup

plain addon on x86-64

c0ldtech, 2025-04-25:
Same issue here.

tsjjBCNBna, 2025-04-25:
Would it be possiible to ping all adresses of the device and check the response. 
I would guess there are a lot of "hidden" settings...

alikhanich, 2025-04-25:
I bought this  [Moes curtain switch](https://moeshouse.com/products/zigbee-smart-curtain-switch-module-mini-design-app-remote-control?srsltid=AfmBOootyQ8RTcqIAG6Q5IuMUUF0Hd7yFgd3yogrgKJjW3vBYlWBfOSV) and it should be very similar to Tuya, and in HA I can't change switch type, and also position reporting is not working good.

closed 2025-08-24 by github-actions[bot]