Koenkk/zigbee2mqtt #26858: Generally hard glide path for people trying to contribute.

tomtom13, 2025-03-25:
### What happened?

Hi,
@Koenkk 
This is generally to you. We've been using z2m for quite some time and I think most of people are pretty happy with it. However, don't know if you noticed but issues opened here are by seasoned engineers, that my bet is would love to contribute / fix the problem them self. Some I think would love to start contributing in general, however it seems that "glide path" is pretty steep here and only some die hards are contributing / fixing (I might be wrong). I know it's your precious project but embracing help might help :D

### What did you expect to happen?

To see some design documentation, howto's, procedures, quick and easy startup to develop a simple patch on their own machine etc etc that will let people get their hand dirty. 

### How to reproduce it (minimal and precise)

Current state :D 

### Zigbee2MQTT version

:D

### Adapter firmware version

xxx

### Adapter

plentiful

### Setup

on my seat writing rubbish.

Koenkk, 2025-03-26:
hi @tomtom13 , I would love to improve this, but having worked so long on the project it is always hard to see where other people get stuck. Would be nice if you could give some concrete example.

tomtom13, 2025-03-26:
@Koenkk It's not about being stuck, it's about diving into the project depending what ones core interest is.

So let's start from the top:
1. I'm an aspiring entrepreneur and would love to create a new zigbee device (I know, dumb idea since market is saturated by cheap stuff) - is there any half decent document describing the zigbee protocol that I could use to get it to work.
2. I'm trying to understand how the messages are structured (payloads) on zigbee network for home stuff including the discovery process etc.
3. Let's say somebody hand me this new awesome thing that will change my life along with full document of how it talks, and I would love to add support for it since everything else is zigbee based arround me, but I don't know how that actually get from the coordinator to actual herdsman deamon - is there any translation layer, any decoding etc etc.
4. how data flows inside of Z2M ? let's say I see a bug staring me in the face - and I've got a choice a) ignore it b) reverse engineer whole project.
5. I know how data is formated by Z2M for MQTT along with automatic discovery, but that took me a lot of time to reverse engineer for other product (which didn't use Z2M, since it was MQTT over TCP based) - but people don't know it. 
6. I want to "build" my very first Z2M and run it locally and plug it to something working like HASS - is there "Tommy my first Z2M set" guide. 
7. Are there any coding standards involved ? (I know boring corporate stuff but worth asking)
8. Are there any reviews ? (I know boring corporate stuff but worth asking)
9. Are there any automated build / tests that go along with somebody's branch ? (I know boring corporate stuff but worth asking)

Look I'm not trying to pick on you, this project is great - but what if you get hit by a bus tomorrow ? any "lieutenants" to step in your shoes ? Any list of maintainers or people to refer to if one get's trully stuck ?

In kernel long time ago there was a joke - "you can predict how long FS will live by how badly it's documented"

Koenkk, 2025-03-27:
- 1 and 2: This is out of the scope of Zigbee2MQTT project, as this is defined in the Zigbee protocol (well documented, search for e.g. ZCL docs)
- 3. How to add support for new devices is documented (e.g. https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html)
- 4. Not documented indeed
- 5. I think you can help writing some docs here
- 6. Yes, it is basically this: https://www.zigbee2mqtt.io/guide/installation/01_linux.html (checks out repo, builds and runs it, you can use the same env to develop)
- 7. Yes, things like linting, formatting and a 100% test coverage is enforced
- 8. Yes (look at the merged PRs)
- 9. Yes (for every PR)

clockbrain, 2025-04-07:
I've posted some guidance for setting up and developing for z2m in WSL2 here https://github.com/Koenkk/zigbee2mqtt/discussions/21544#discussioncomment-12709490

It would need some others to work through the process and include whatever I missed and fix what I got wrong before it would be ready to add to the documentation though.

closed 2025-06-15 by github-actions[bot]