Koenkk/zigbee2mqtt #26870: ZBDongle-E Error while starting zigbee-herdsman

hermannw, 2025-03-26:
### What happened?

After my previous dongle failed, I decided t upgrade Z2M to V2.1.3 and bought a new "Zigbee 3.0 USB dongle plus" (ZBDongle-E).  I upgraded the coordinator firmware to 7.4.4 but Z2M won't start. The error is " z2m: Error while starting zigbee-herdsman".
This is a plain vanilla installation on a Pine64 with latest version Armbian (this has been running for the last three+ years) No docker, no Home Assistant... 

Config.yaml:
```
homeassistant:
  enabled: true
frontend:
  enabled: true
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://192.168.1.120

# Serial settings, only required when Zigbee2MQTT fails to start with:
#   USB adapter discovery error (No valid USB adapter found).
#   Specify valid 'adapter' and 'port' in your configuration.
serial:
    # Location of the adapter
    # USB adapters - use format "port: /dev/serial/by-id/XXX"
    # Ethernet adapters - use format "port: tcp://192.168.1.12:6638"
    port: /dev/serial/by-id/usb-Itead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_V2_1a22a8dd3a53ef11b2ea20e0174bec31-if00-port0
    # Adapter type, allowed values: `zstack`, `ember`, `deconz`, `zigate` or `zboss`
    adapter: ember
    rtscts: false
    baudrate: 115200
```

Any help would be greatly appreciated.

### What did you expect to happen?

Z2M starting

### How to reproduce it (minimal and precise)

Upgrade Z2M to 2.1.3, plus in upgraded dongle, start Z2M with "pnpm start"

### Zigbee2MQTT version

2.1.3

### Adapter firmware version

7.4.4

### Adapter

ZBDongle-E

### Setup

Plain vanilla Armbian on Pine64. No other programs running on this.

### Debug log

Error after starting:
'''
pi@pine64:/opt/zigbee2mqtt$ pnpm start

> zigbee2mqtt@2.1.3 start /opt/zigbee2mqtt
> node index.js

Starting Zigbee2MQTT without watchdog.
[2025-03-26 09:44:20] info:     z2m: Logging to console, file (filename: log.log)
[2025-03-26 09:44:20] info:     z2m: Starting Zigbee2MQTT version 2.1.3 (commit #fc35ac5)
[2025-03-26 09:44:20] info:     z2m: Starting zigbee-herdsman (3.2.7)
[2025-03-26 09:44:21] info:     zh:ember: Using default stack config.
[2025-03-26 09:44:21] info:     zh:ember: ======== Ember Adapter Starting ========
[2025-03-26 09:44:21] info:     zh:ember:ezsp: ======== EZSP starting ========
[2025-03-26 09:44:21] info:     zh:ember:uart:ash: ======== ASH Adapter reset ========
[2025-03-26 09:44:21] info:     zh:ember:uart:ash: RTS/CTS config is off, enabling software flow control.
[2025-03-26 09:44:21] info:     zh:ember:uart:ash: Serial port opened
[2025-03-26 09:44:21] info:     zh:ember:uart:ash: ======== ASH starting ========
[2025-03-26 09:44:22] info:     zh:ember:uart:ash: ======== ASH connected ========
[2025-03-26 09:44:22] info:     zh:ember:uart:ash: ======== ASH started ========
[2025-03-26 09:44:22] info:     zh:ember:ezsp: ======== EZSP started ========
[2025-03-26 09:44:22] info:     zh:ember: Adapter EZSP protocol version (13) lower than Host. Switched.
[2025-03-26 09:44:22] info:     zh:ember: Adapter version info: {"ezsp":13,"revision":"7.4.4 [GA]","build":0,"major":7,"minor":4,"patch":4,"special":0,"type":170}
[2025-03-26 09:44:22] error:    z2m: Error while starting zigbee-herdsman
[2025-03-26 09:44:22] error:    z2m: Failed to start zigbee-herdsman
[2025-03-26 09:44:22] error:    z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-03-26 09:44:22] error:    z2m: Exiting...
[2025-03-26 09:44:23] error:    z2m: Error: [BACKUP] Current backup file is not for EmberZNet stack.
    at EmberAdapter.getStoredBackup (/opt/zigbee2mqtt/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:1152:23)
    at EmberAdapter.initTrustCenter (/opt/zigbee2mqtt/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:928:29)
    at EmberAdapter.initEzsp (/opt/zigbee2mqtt/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:712:18)
    at EmberAdapter.start (/opt/zigbee2mqtt/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:1536:24)
    at Controller.start (/opt/zigbee2mqtt/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/controller/controller.ts:136:29)
    at Zigbee.start (/opt/zigbee2mqtt/lib/zigbee.ts:69:27)
    at Controller.start (/opt/zigbee2mqtt/lib/controller.ts:142:13)
    at start (/opt/zigbee2mqtt/index.js:161:5)
 ELIFECYCLE  Command failed with exit code 1.
'''

hermannw, 2025-03-26:
It looks like the culprit was the coordinator-backup.json file, although I deleted that twice before. I actually deleted everything but the configuration.yaml from the data folder...
Well, it works now.

hermannw, 2025-03-26:
Well, that didn't last very long. I tried to join the first device and that made it crash again.... Now it won't start again.

> zigbee2mqtt@2.1.3 start /opt/zigbee2mqtt
> node index.js

Starting Zigbee2MQTT without watchdog.
[2025-03-26 13:21:19] info:     z2m: Logging to console, file (filename: log.log)
[2025-03-26 13:21:20] info:     z2m: Starting Zigbee2MQTT version 2.1.3 (commit #fc35ac5)
[2025-03-26 13:21:20] info:     z2m: Starting zigbee-herdsman (3.2.7)
[2025-03-26 13:21:20] info:     zh:ember: Using default stack config.
[2025-03-26 13:21:20] info:     zh:ember: ======== Ember Adapter Starting ========
[2025-03-26 13:21:20] info:     zh:ember:ezsp: ======== EZSP starting ========
[2025-03-26 13:21:20] info:     zh:ember:uart:ash: ======== ASH Adapter reset ========
[2025-03-26 13:21:20] info:     zh:ember:uart:ash: RTS/CTS config is off, enabling software flow control.
[2025-03-26 13:21:20] info:     zh:ember:uart:ash: Serial port opened
[2025-03-26 13:21:20] info:     zh:ember:uart:ash: ======== ASH starting ========
[2025-03-26 13:21:21] info:     zh:ember:uart:ash: ======== ASH connected ========
[2025-03-26 13:21:21] info:     zh:ember:uart:ash: ======== ASH started ========
[2025-03-26 13:21:21] info:     zh:ember:ezsp: ======== EZSP started ========
[2025-03-26 13:21:21] info:     zh:ember: Adapter EZSP protocol version (13) lower than Host. Switched.
[2025-03-26 13:21:21] info:     zh:ember: Adapter version info: {"ezsp":13,"revision":"7.4.4 [GA]","build":0,"major":7,"minor":4,"patch":4,"special":0,"type":170}
[2025-03-26 13:21:22] info:     zh:ember: [STACK STATUS] Network up.
[2025-03-26 13:21:22] info:     zh:ember: [INIT TC] Adapter network matches config.
[2025-03-26 13:21:22] info:     zh:ember: [CONCENTRATOR] Started source route discovery. 1247ms until next broadcast.
[2025-03-26 13:21:22] error:    z2m: Failed to start zigbee-herdsman
[2025-03-26 13:21:22] error:    z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-03-26 13:21:22] error:    z2m: Exiting...
[2025-03-26 13:21:22] error:    z2m: /opt/zigbee2mqtt/node_modules/.pnpm/zigbee-herdsman-converters@23.2.0/node_modules/zigbee-herdsman-converters/dist/lib/ewelink.js:2
var __createB�nding } (this && this.O_createBin`ing) || (Object.create ? (function(o, m, k, k2) {

SyntaxError: Invalid or unexpected token
    at wrapSafe (node:internal/modules/cjs/loader:1472:18)
    at Module._compile (node:internal/modules/cjs/loader:1501:20)
    at Object.Module._extensions..js (node:internal/modules/cjs/loader:1613:10)
    at Module.load (node:internal/modules/cjs/loader:1275:32)
    at Function.Module._load (node:internal/modules/cjs/loader:1096:12)
    at Module.require (node:internal/modules/cjs/loader:1298:19)
    at require (node:internal/modules/helpers:182:18)
    at Object.<anonymous> (/opt/zigbee2mqtt/node_modules/.pnpm/zigbee-herdsman-converters@23.2.0/node_modules/zigbee-herdsman-converters/src/devices/sonoff.ts:6:1)
    at Module._compile (node:internal/modules/cjs/loader:1529:14)
    at Object.Module._extensions..js (node:internal/modules/cjs/loader:1613:10)
 ELIFECYCLE  Command failed with exit code 1.

hermannw, 2025-03-26:
After deleting one more time coordinatorßbackup.json and database.db, Y2M started again. But the moment I am trying to join the first device, it stops.
Here is the relevant part of the log:

[2025-03-26 13:30:12] debug:    z2m: Skipping message, still interviewing
[2025-03-26 13:30:12] debug:    zh:controller:device: Interview - IAS - after enrolling state (0): '{"iasCieAddr":"0x08b95ffffe597465","zoneState":1}'
[2025-03-26 13:30:13] debug:    zh:controller:device: Interview - IAS successfully enrolled '0x00124b002228b417' endpoint '1'
[2025-03-26 13:30:13] debug:    zh:controller:device: Interview - completed for device '0x00124b002228b417'
[2025-03-26 13:30:13] debug:    zh:controller:database: Writing database to '/opt/zigbee2mqtt/data/database.db'
[2025-03-26 13:30:13] info:     zh:controller: Succesfully interviewed '0x00124b002228b417'
[2025-03-26 13:30:13] debug:    zhc: Getting definitions for: sonoff.js,11
/opt/zigbee2mqtt/node_modules/.pnpm/zigbee-herdsman-converters@23.2.0/node_modules/zigbee-herdsman-converters/dist/lib/ewelink.js:2
var __createB�nding } (this && this.O_createBin`ing) || (Object.create ? (function(o, m, k, k2) {

SyntaxError: Invalid or unexpected token
    at wrapSafe (node:internal/modules/cjs/loader:1472:18)
    at Module._compile (node:internal/modules/cjs/loader:1501:20)
    at Object.Module._extensions..js (node:internal/modules/cjs/loader:1613:10)
    at Module.load (node:internal/modules/cjs/loader:1275:32)
    at Function.Module._load (node:internal/modules/cjs/loader:1096:12)
    at Module.require (node:internal/modules/cjs/loader:1298:19)
    at require (node:internal/modules/helpers:182:18)
    at Object.<anonymous> (/opt/zigbee2mqtt/node_modules/.pnpm/zigbee-herdsman-converters@23.2.0/node_modules/zigbee-herdsman-converters/src/devices/sonoff.ts:6:1)
    at Module._compile (node:internal/modules/cjs/loader:1529:14)
    at Object.Module._extensions..js (node:internal/modules/cjs/loader:1613:10)
 ELIFECYCLE  Command failed with exit code 1.

[log1.log](https://github.com/user-attachments/files/19473561/log1.log)

Nerivec, 2025-03-26:
First error was indeed because you had a previous `coordinator-backup.json` that was not made for `ember`. Removing/renaming should fix it.

Can you try removing the `node_modules` and re-installing dependencies:
```bash
rm -rf node_modules
pnpm i --frozen-lockfile
pnpm run prepack
```
(that last one will clean and build again)

If it still doesn't work, can you post the output of the following commands on your machine?
```bash
uname -a
uname -m
cat /etc/issue.net
node --version
pnpm --version
```

hermannw, 2025-03-26:
Thanks for the answer. That did not help.

pi@pine64:/opt/zigbee2mqtt$ uname -a
Linux pine64 6.6.75-current-sunxi64 #1 SMP Sat Feb  1 17:37:57 UTC 2025 aarch64 aarch64 aarch64 GNU/Linux

pi@pine64:/opt/zigbee2mqtt$ uname -m
aarch64

pi@pine64:/opt/zigbee2mqtt$ cat /etc/issue.net
Armbian 25.2.3 jammy

pi@pine64:/opt/zigbee2mqtt$ node -v
v20.19.0

pi@pine64:/opt/zigbee2mqtt$ pnpm -v
10.4.1

As I mentioned in the beginning, Z2M has been running for years on this configuration until I updated. Is there a way to go back to the previous version?

Nerivec, 2025-03-26:
There is something wrong with the installed dependencies that Z2M uses, somehow, it got corrupted.

Can you re-do this (in that order):
```bash
rm -rf $(pnpm store path)
rm -rf node_modules
pnpm i --frozen-lockfile
pnpm run prepack
```

Note: Ewelink/Sonoff have common devices (same company more or less).

hermannw, 2025-03-26:
Ok, I think that did the trick! Works now.
Thank you very much for your help!

closed 2025-03-26 by hermannw