Koenkk/zigbee2mqtt #26876: Vallhorn E2134 wrong/missing manufacturer specific cluster and attribute definition

jeffmhastings, 2025-03-26:
### What happened?

Device has cluster 0xfc81 incorrectly identified as "heimanSpecificAirQuality". This cluster contains device-specific attributes that correspond to the buttons on the back of the device (day mode/night mode, off delay).

### What did you expect to happen?

I would expect to see this cluster named something along the lines of the other Ikea manufacturer specific cluster names. Perhaps "manuSpecificIkeaMotionSensor"?

It would also be nice to have the mode and offDelay attributes defined. From information that I found [here](https://community.home-assistant.io/t/ikea-vallhorn-set-off-delay-and-dark-only-mode-with-zha-toolkit/815490) mode is attribute 0 and offDelay is attribute 2. There may be other unknown attributes as well that I have not fully investigated.

And for what it's worth, I am interested in contributing a fix for this. I was able to read and write to these attributes by writing an external converter and loading it. It seems like I would need to submit the changes to the zigbee-herdsman-converters repo.

### How to reproduce it (minimal and precise)

Pair the Ikea Vallhorn E2134 with zigbee2mqtt. Notice the heimanSpecificAirQuality and manuSpecificIkeaUnknown clusters. These are not usable in their current state. manuSpecificIkeaUnknown has no attributes, so nothing can be read or written. heimanSpecificAirQuality has attributes defined, but they are for a completely unrelated air quality sensor.

### Zigbee2MQTT version

2.1.3

### Adapter firmware version

20230507

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus ZBDongle-P

### Setup

Docker container running in Kubernetes

Nerivec, 2025-03-27:
Looks like both manufacturers might be using a conflicting ID for the cluster, so there's a need to move both to custom clusters and off zigbee-herdsman (it's been ongoing for a while, but still some work to do, PRs welcomed!).

jeffmhastings, 2025-03-28:
> Looks like both manufacturers might be using a conflicting ID for the cluster, so there's a need to move both to custom clusters and off zigbee-herdsman (it's been ongoing for a while, but still some work to do, PRs welcomed!).

I can take a stab at that. I guess the process would be to first define the custom clusters in zigbee-herdsman-converters and update the devices to use those. Then a separate PR to remove the definitions from zigbee-herdsman? Are there any dependency or ordering issues between updating each library that I need to be aware of?

Koenkk, 2025-03-30:
> Are there any dependency or ordering issues between updating each library that I need to be aware of?

I'll take care of that, feel free to submit the PRs.

jeffmhastings, 2025-04-01:
OK, I've got a few PRs up now to address moving the heimanSpecificAirQuality cluster over to zigbee-herdsman-converts.

- https://github.com/Koenkk/zigbee-herdsman-converters/pull/9104
- https://github.com/Koenkk/zigbee-herdsman/pull/1379

Let me know if there's anything that should be done differently. I figure that once these changes are in I can go about adding the changes for Ikea E2134 in a separate PR.

closed 2025-06-09 by github-actions[bot]