Koenkk/zigbee2mqtt #26879: Aqara Switch H2 EU display power consumption problem

JoergAdler, 2025-03-27:
### What happened?

Hi,

I use the Sonoff Dongle e with firmware 7.4.5 (The error also occurs in 7.4.4) when configuring the [Aqara H2 EU switch](https://www.zigbee2mqtt.io/devices/WS-K07E.html#aqara-ws-k07e), i get the following error:

`z2m: Failed to configure 'Dusche Wandschalter', attempt 4 (Error: ZCL command 0x54ef441000ef08ee/1 haElectricalMeasurement.read(["acCurrentDivisor","acCurrentMultiplier"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
    at Endpoint.checkStatus (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:349:28)
    at Endpoint.zclCommand (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:954:26)
    at Endpoint.read (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:445:29)
    at result.configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@23.2.0/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:1991:37)
    at Object.configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@23.2.0/node_modules/zigbee-herdsman-converters/src/index.ts:357:21)
    at Configure.configure (/app/lib/extension/configure.ts:124:13)
    at Configure.onMQTTMessage (/app/lib/extension/configure.ts:49:25)`

When i reload the current consumption in the settings page, I get

`z2m: Publish 'get' 'current' to 'Dusche Wandschalter' failed: 'Error: ZCL command 0x54ef441000ef08ee/1 haElectricalMeasurement.read(["rmsCurrent"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'`

This results in a failrly slow update of the power consumption data from the switch, as it seems in the normal updates this data is present. E.g.:

`z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Dusche Wandschalter', payload '{"consumption":0.28369951248168945,"current":0.03,"device_temperature":27,"energy":0.28,"flip_indicator_light":"OFF","led_disabled_night":true,"led_indicator":"ON","linkquality":184,"lock_relay":true,"lock_relay_up":true,"multi_click_down":false,"operation_mode_up":"decoupled","power":0.96,"power_on_behavior":"previous","power_outage_count":3,"power_up":13.5,"state":"ON","state_up":"ON","update":{"installed_version":-1,"latest_version":-1,"state":null}}'`

Seems that somewhere a varaible name is wrong. Could you please correct it?

### How to reproduce it (minimal and precise)

The error alread occurs in the first attempt to pair the device

### Zigbee2MQTT version

2.1.3-1

### Adapter firmware version

7.4.5 [GA]

### Adapter

EmberZNet

### Setup

Add-on in HA on a raspberry pi 4, native os

zeratul021, 2025-06-06:
I have z2m with SLZB-07p7 and Aqara WS-K07E and this error is appearing in the log:
`2025-06-06 02:49:13z2m: Failed to configure 'bedroom_switch_back_2', attempt 3 (Error: Device 0x54ef441000ef2d8c has no input cluster haElectricalMeasurement at getEndpointsWithCluster (/app/node_modules/.pnpm/zigbee-herdsman-converters@23.53.0/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:62:15) at result.configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@23.53.0/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:2085:44) at Object.configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@23.53.0/node_modules/zigbee-herdsman-converters/src/index.ts:363:27) at Configure.configure (/app/lib/extension/configure.ts:124:13) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:204:17))`

JoergAdler, 2025-08-06:
Still valid