Koenkk/zigbee2mqtt #26882: Coordinator no longer functioning after software upgrade. of the coordinator

Itsme-Hans, 2025-03-27:
### What happened?

I have installed a software upgrade for the SLZB coordinator and as a result it was mo longer seen by z2m. downgrading the coordinator to the previous software did not resolve the problem. Troubleshooting with SMLIGHT they concluded that it was caused by a bug in Z2M and needed to report this.

### What did you expect to happen?

I expected that my operational system would not go 100% down after a software upgrade

### How to reproduce it (minimal and precise)

My operational system is running on a HA yellow. However for troubleshooting I installed a fresh HA install on a Pi and connected the SLZB-06 coordinator to the pi via USB. on my HA yellow I'm using the network connection to connect to that coordinator. 

[Both Sonoff and SLZB are connected to the PI but its pointing to the….pdf](https://github.com/user-attachments/files/19485006/Both.Sonoff.and.SLZB.are.connected.to.the.PI.but.its.pointing.to.the.pdf)

### Zigbee2MQTT version

2.1.3-1

### Adapter firmware version

20221226

### Adapter

SLZB-6

### Setup

For troubleshooting a Ri 5 life network a HA yellow

### Debug log

While in the serial I have configured Stack it seems tha tit ignore that for I read ember in the logfile

[45df7312_zigbee2mqtt_2025-03-27T11-16-03.185Z.log](https://github.com/user-attachments/files/19485131/45df7312_zigbee2mqtt_2025-03-27T11-16-03.185Z.log)

Nerivec, 2025-03-27:
You seem to have a conflict with the port used. From your logs, it tries to start with one or the other randomly:

```json
{"path":"/dev/ttyUSB0","manufacturer":"SMLIGHT","serialNumber":"0e4cbd03318bef119b4c20ccef8776e9","pnpId":"usb-SMLIGHT_SMLIGHT_SLZB-06_0e4cbd03318bef119b4c20ccef8776e9-if00-port0","vendorId":"10c4","productId":"ea60"}
```

```json
{"path":"/dev/ttyUSB0","manufacturer":"ITead","serialNumber":"3a1d831477d9eb1180e4158e6fe9f4d9","pnpId":"usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_3a1d831477d9eb1180e4158e6fe9f4d9-if00-port0","vendorId":"10c4","productId":"ea60"}
```

Can you also post your `configuration.yaml` (you can remove sensitive sections)?

Itsme-Hans, 2025-03-27:
what I noticed is that if both sonoff and the SLZB are both connected to the usb and visible in the hardware, regardless what I put in the serial it always startup with the Sonoff. See below my config.yaml. Only when I remove the sonoff it will use the credentials in the serial

[data_path: :config:zigbee2mqtt.pdf](https://github.com/user-attachments/files/19486994/data_path.config.zigbee2mqtt.pdf)

Nerivec, 2025-03-27:
You have two `serial` in that file (also can't be sure the indentation is correct in this format, better to copy/paste directly here on github).

```yaml
data_path: /config/zigbee2mqtt
socat:
  enabled: false
  master: pty,raw,echo=0,link=/tmp/ttyZ2M,mode=777
  slave: tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5
  options: "-d -d"
  log: false
mqtt:
  server: mqtt://core-mosquitto:1883
  user: xxxxxxx
  password: xxxxxxxx
serial:
  port: >-
    /dev/serial/by-id/usb-SMLIGHT_SMLIGHT_SLZB-06_0e4cbd03318bef119b4c20ccef8776e9-if00-port0
  baudrate: 115200
  adapter: zstack
  disable_led: false
```

Itsme-Hans, 2025-03-27:
not sure what you mean with 2 serial. this is the path I copied from the hardware. See attachment.
Please bear in mind that this is my test setup. where I use usb. I did that to investigate if it was a problem with the SLZB-06 or with HA. My route of thinking was that it was the SLZB-06 device for the Sonoff dongle worked flawless. 
smlight however disagreed and point to Z2M. Perhaps it's better to send and use the real log and config that im using on my life network ?

[image]

data_path: /config/zigbee2mqtt
socat:
  enabled: false
  master: pty,raw,echo=0,link=/tmp/ttyZ2M,mode=777
  slave: tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5
  options: "-d -d"
  log: false
mqtt:
  server: mqtt://core-mosquitto:1883
  user: xxxxx
  password: xxxxx
serial:
  serial:
    port: >-
      /dev/serial/by-id/usb-SMLIGHT_SMLIGHT_SLZB-06_0e4cbd03318bef119b4c20ccef8776e9-if00-port0
    baudrate: 115200
    adapter: zstack
    disable_led: false
  advanced:
    transmit_power: 20

Nerivec, 2025-03-27:
```
password: xxxxx
serial: <======= HERE
serial: <======= AND HERE
port: >-
```

Two `serial` in the config means it's not taken into account at all, it's invalid (it will behave like it's not there at all, which means Z2M will try to auto-detect instead, hence your troubles with the multiple adapters).

Itsme-Hans, 2025-03-27:
Is it an idea to move  away from my test setup ?
I used that to troubleshoot with smlight customer support to identify what is going on.  
I wanted to compare the SLZB coordinator with Sonoff and the only way to do that was via usb. 
My conclusion was that if the Sonoff works fine and the ZLZB-06 doesn't  it has to be the SLZB-06 that is causing this ?. 
Take in consideration that my problems started after I upgraded the firmware of the coordinator
smlight however disagrees and based on what they see on my test setup using usb they conclude that that my problem is caused by a bug in Z2M. In the mean time my life system is down. I attach the log and config file of my life configuration for that is for me the real problem.

data_path: /config/zigbee2mqtt
socat:
  enabled: false
  master: pty,raw,echo=0,link=/tmp/ttyZ2M,mode=777
  slave: tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5
  options: "-d -d"
  log: false
mqtt:
  server: mqtt://core-mosquitto:1883
  user: xxxxxx
  password: xxxxx

[45df7312_zigbee2mqtt_2025-03-27T16-11-36.101Z.log](https://github.com/user-attachments/files/19490026/45df7312_zigbee2mqtt_2025-03-27T16-11-36.101Z.log)

  channel: 15
serial:
  port: tcp://192.168.1.17:6638
  baudrate: 115200
  adapter: zstack
  disable_led: false
advanced:
  transmit_power: 20

Itsme-Hans, 2025-03-27:
> ```
> password: xxxxx
> serial: <======= HERE
> serial: <======= AND HERE
> port: >-
> ```
> 
> Two `serial` in the config like you have (based on your previous pdf) means it's not taken into account at all, it's invalid (it will behave like it's not there at all, which means Z2M will try to auto-detect instead, hence your troubles with the multiple adapters). _See the configuration I posted above, that is correct (just replace your username/password for MQTT to the correct values)._
> 
> Once you've cleared this problem, we can re-evaluate the rest with a "clean" startup log.

check. I did remove that one serial and it seems to be  cosmetic. Hence I did not even wanted to update.  
I also let it run again but had the same result. but....please see my comment further down. perhaps we should move away from the USB connection I used to for troubleshooting with smlight and focus on my life network ? as that is not connected via usb but via the network. See below

Nerivec, 2025-03-27:
> it seems to be cosmetic

Based on your previous PDF, it's not, it would result in that configuration not being used at all. Which explains why Z2M was trying to auto-discover, and running into troubles with the two adapters plugged in.
The new logs show a clear error message now.
```
Error: network commissioning timed out - most likely network with the same panId or extendedPanId already exists nearby (Error: AREQ - ZDO - stateChangeInd after 60000ms
```

See the chapter in the docs about this:
https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html#error-coordinator-failed-to-start-probably-the-panid-is-already-in-use-try-a-different-panid-or-channel

For further attempts, you will indeed want to try directly with your "real" setup.

Can you post what was the firmware version you had before, and the version you have now?

Koenkk, 2025-03-27:
Regarding:

> Both Sonoff and SLZB are connected to the PI but its pointing to the SLZB port it however doesn’t boot up on the SLZB but still on the Sonoff

Your serial config is wrong, it should be:

```yaml
 adapter: zstack
 port: ...
```

instead of 

```yaml
serial:
    adapter: zstack
    port: ...
```

Itsme-Hans, 2025-03-27:
> Regarding:
> 
> > Both Sonoff and SLZB are connected to the PI but its pointing to the SLZB port it however doesn’t boot up on the SLZB but still on the Sonoff
> 
> Your serial config is wrong, it should be:
> 
>  adapter: zstack
>  port: ...
> instead of
> 
> serial:
>     adapter: zstack
>     port: ...
> When starting on the sonoff, also make sure the SLZB is now powered on

the setting above are a copy of what is suggested by the vendor of the device. 
That said, I did remove the  word Serial:  as you suggested and saved it. Funny enough it is automatically added again at the same location

Nerivec, 2025-03-27:
If you switch between the full yaml view, and the UI view in the add-on config page, the formatting has to be just right otherwise it breaks the configuration.
In the full yaml view, for your live network, you want this (exact copy/paste, with just changing mqtt user/password back to proper):
```yaml
data_path: /config/zigbee2mqtt
socat:
  enabled: false
  master: pty,raw,echo=0,link=/tmp/ttyZ2M,mode=777
  slave: tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5
  options: "-d -d"
  log: false
mqtt:
  server: mqtt://core-mosquitto:1883
  user: xxxxxx
  password: xxxxx
serial:
  port: tcp://192.168.1.17:6638
  baudrate: 115200
  adapter: zstack
  disable_led: false
```

Nerivec, 2025-03-27:
Other settings have to be configured directly through the `configuration.yaml` file.

Did you check the list on the page I sent?
Do you remember if the error you had initially (right after the firmware update) was the same as this last one? Do you have the logs?

Itsme-Hans, 2025-03-27:
Yep,  I went through the list ( link) you send.  And the one topic that comes into mind is the panID. 
All is related to  " z2m: Failed to start zigbee-herdsman"

That's also what the support team from SMlight told me when looking at the log.
but, let me take you back in time for a moment

A few months ago when installing the coordinator I faced this panId problem. (z2m: Failed to start zigbee-herdsman)
Talking with SMlight support they explained me that likely 1 or more zigbee sensors were transmitting a wrong panID.
Advise was to start with a fresh install and build it up from scratch. 
That way I was making sure that all sensors were transmitting that same panId. 

Now we are a few moths later and I here I am facing the same thing as when I started
and now the only thing that happened was a firmware upgrade of the coordinator
If a firmware upgrade is able to change something with the panID then it's non reliable.
Rolling back my daily backup doesn't work, I tried that.

SMlight support concluded that this is a Z2M  issue ( bug) and pointed me to this page to report it.

I'm still hoping that I can get the coordinator working again. ?  It would be a real nightmare if I need to Start from scratch again

Nerivec, 2025-03-28:
Z2M shows this error after the coordinator did not reply to the request to start a _new_ network for over 60 seconds.
This means your network configuration was reset at some point (during the previous debugging when you switched machines?).
If you lost the `coordinator_backup.json` & `configuration.yaml` during the shuffle, you may have no other choice than to start anew.
I suggest you check the presence of these two files in your zigbee2mqtt data folder and verify their content (in your "live" machine). If you don't have them anymore, or they don't match your previous network (where your devices currently reside), check if you have any previous backup of them, and put these back in their place.

Itsme-Hans, 2025-03-28:
I was afraid that I end up starting all over again. 
That brings me to the point of the reliability using the SLZB-06 as coordinator for Z2M.
I hope you agree with me that the coordinator has to be the most stable part in the chain.  Now in my system it has become the weakest link
I'm  not blind for the arguments from SMlight. 
That said I'm also not convinced that it is caused by Z2M. Let's keep it at that it's just not a happy marriage.
What to do ? 
Stick with Z2M ( I like it a lot) and move back to my Sonoff as coordinator ?  Or ...move to ZHA and keep the SLZB-06 in ?

Itsme-Hans, 2025-03-30:
Just some background info. I believe that the latest software upgrade 2.8.2 dev 3 is not working correctly for my hardware. 
I dropped a message to SMlight support on this.
FYI, you will see a message to upgrade in HA and it's just easy click Y 
What drives me to come to this conclusion? 
I decided to convert the device from coordinator into a router function. 
It however just didn't work as expected . Z2M didn't see the device and the device web page was slow in response. 
Report didn't help Factory restart, didn't help.
I downgraded the firmware to version 2.8.0 and then it  came back to life.
I therefore also presume that my firmware upgrade to 2.8.2 caused the issue with this stick in Coordinator mode.

closed 2025-06-08 by github-actions[bot]