Koenkk/zigbee2mqtt #26884: failed with status '(0x11: BUFFER_FULL)' (expected '(0x00: SUCCESS)'))'

litecross91, 2025-03-27:
### What happened?

After 2 days of server uptime, all my zigbee devices are now stopped working. I get these error messages. Everything is on the latest version.
It has been months since the zigbee coordinator has been updated. I would rule out coordinator fw mistake.

SLZB-06
Firmware version: | 20240710 | v2.7.1
Zigbee2MQTT version
[2.1.3]
Coordinator type
zStack3x0
Coordinator revision
20240710
Coordinator IEEE Address
0xe0798dfffed1ba31
Frontend version
0.9.4
zigbee-herdsman-converters version
23.2.0
zigbee-herdsman version
3.2.7

```
error 2025-03-27 14:49:53z2m: Publish 'get' 'state' to 'Desk bulb' failed: 'Error: ZCL command 0xa4c138679316a2c3/1 genOnOff.read(["onOff"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (--> 'SREQ: AF - dataRequest - {"dstaddr":29001,"destendpoint":1,"srcendpoint":1,"clusterid":6,"transid":10,"options":0,"radius":30,"len":5,"data":{"type":"Buffer","data":[16,212,0,0,0]}}' failed with status '(0x11: BUFFER_FULL)' (expected '(0x00: SUCCESS)'))'
error 2025-03-27 14:55:25z2m: Publish 'get' 'state' to 'Floor bulb' failed: 'Error: ZCL command 0xa4c1384a50524530/1 genOnOff.read(["onOff"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (--> 'SREQ: AF - dataRequest - {"dstaddr":9799,"destendpoint":1,"srcendpoint":1,"clusterid":6,"transid":23,"options":0,"radius":30,"len":5,"data":{"type":"Buffer","data":[16,225,0,0,0]}}' failed with status '(0x11: BUFFER_FULL)' (expected '(0x00: SUCCESS)'))'
error 2025-03-27 14:55:58z2m: Publish 'get' 'occupancy_timeout' to 'Kitchen Presence Sensor' failed: 'Error: ZCL command 0x0ceff6fffe6c0d69/1 msOccupancySensing.read([32], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (--> 'SREQ: AF - dataRequest - {"dstaddr":19294,"destendpoint":1,"srcendpoint":1,"clusterid":1030,"transid":27,"options":0,"radius":30,"len":5,"data":{"type":"Buffer","data":[16,229,0,32,0]}}' failed with status '(0x11: BUFFER_FULL)' (expected '(0x00: SUCCESS)'))'
```

### Zigbee2MQTT version

2.1.3

### Adapter firmware version

20240710

### Adapter

SLZB-06

### Setup

Proxmox, HASSOS

litecross91, 2025-03-27:
Ok, I just read now the coordinator fw update description. There is a new version.
It may be useful for others.
[image]

Koenkk, 2025-03-27:
Let's continue in https://github.com/Koenkk/Z-Stack-firmware/discussions/545

closed 2025-03-27 by Koenkk