Koenkk/zigbee2mqtt #26892: Cant pair new devices

Gibbz, 2025-03-28:
### What happened?

I can no longer pair new devices on my setup.
Ive tried rebooting the instance multiple times. There is nothing showing in the logs.

running z2m version 2.1.3

I have run through the guide here:
https://www.zigbee2mqtt.io/guide/faq/#my-network-is-unstable-and-or-performs-poorly-what-can-i-do

My device is a zzh running a firmware from 2022 which has been no problems in the past few months.
Has there been any known issues lately with pairing new devices?
I'm kind of stuck how else I can debug the issue?

### What did you expect to happen?

pairing to log and work

### How to reproduce it (minimal and precise)

try pair any device

### Zigbee2MQTT version

2.1.3

### Adapter firmware version

20220219

### Adapter

zzh!

### Setup

x86 docker

### Debug log

n/a

Nerivec, 2025-03-28:
How many devices do you have currently on the network?
Is the issue showing when permit joining on specific devices?
If not the above, have you tried permit joining on specific devices instead of "all"? 
Can you share some `debug` logs of enabling permit join and trying to join a device (leave it running a few minutes even if you don't see anything)?

Gibbz, 2025-03-28:
i have 24 devices currently.
Tried updating to the latest firmware for my zzh: 20240710
i have permit join all as my default mode of operation ive tried turning it on and off.
The devices i've tried to pair are different brands/devices.

[log.log](https://github.com/user-attachments/files/19513403/log.log)

Gibbz, 2025-03-28:
i removed an old device, and managed to get that to re-pair. When i pairs it now says its an unsupported device.... But the new devices still will not pair... I have lots of errors in the log with one of my garage devices, ill try removing that.

Gibbz, 2025-03-28:
okay got it working! 
it was the option to permit join. It seems my default it no longer working and i needed to do it manually a few times to get it to kick in!

closed 2025-03-28 by Gibbz