Koenkk/zigbee2mqtt #26893: LYWSD03MMC-z Failed to configure

magic-happenz, 2025-03-28:
### What happened?

Hello,

trying to follow up on issue https://github.com/Koenkk/zigbee2mqtt/issues/24035 which was unfortunately closed. I have the same issue right now and tried all recommendation without luck. 

Apparently this issue also sucks the devices batteries empty. Please help.

### What did you expect to happen?

The device being configured.

### How to reproduce it (minimal and precise)

1. delete the device
2. ensure all its entries are removed (MQTT Explorer)
3. permit devices to join
4. reset the device (re-insert battery)
5. watch Zigbee2MQTT adding the device and immediately failing to configure it

### Zigbee2MQTT version

2.1.3 commit: unknown

### Adapter firmware version

7.4.4 [GA]

### Adapter

Sonoff Zigbee 3.0 USB Dongle Plus V2 E (ember)

### Setup

HA Add-On

### Debug log

Please note that device names have been replaced for privacy reasons.

[log.log](https://github.com/user-attachments/files/19503861/log.log)

Koenkk, 2025-03-30:
@robvanoostenrijk are all bindings needed? Maybe we can remove some to get rid of the `TABLE_FULL` error.

magic-happenz, 2025-03-30:
It seems not. From 6 devices I have 3 with and 3 without bindings. They currently all work but it's not very stable. OTA updates do not work either. Some of them run v 124 and some v125 firmware.

robvanoostenrijk, 2025-04-01:
The needed bindings for ZigbeeTlc are mentionedhere:
https://github.com/pvvx/ZigbeeTLc/issues/154#issuecomment-2614457931

> It is enough to receive temperature, humidity and battery data. That is, only 3 bindings, and the device has a buffer of 8.

I'm not well-versed in Z2M enough though to make the changes to limit to these bindings. If you point me in the right direction I will.

Koenkk, 2025-04-01:
@magic-happenz could you provide the data/database.db entry of your device?

magic-happenz, 2025-04-01:
Sure can, took one of the devices that kinda work now after tons of resets and adding but still no binding.

```
{"id":13,"type":"EndDevice","ieeeAddr":"0xa4c13807dad42a99","nwkAddr":29266,"manufId":4417,"manufName":"Xiaomi","powerSource":"Battery","modelId":"LYWSD03MMC-z","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":770,"inClusterList":[0,1,3,32,516,1026,1029],"outClusterList":[25],"clusters":{"msTemperatureMeasurement":{"attributes":{"measuredValue":2021}},"msRelativeHumidity":{"attributes":{"measuredValue":5369}},"genPowerCfg":{"attributes":{"batteryVoltage":28,"batteryPercentageRemaining":141}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":36,"stackVersion":49,"hwVersion":15,"dateCode":"20250301","swBuildId":"3001-0124","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1743537621949,"checkinInterval":3600}
```

Koenkk, 2025-04-03:
Which firmware did you use? This should be fixed in the firmware.

magic-happenz, 2025-04-03:
You mean of the sensor? I use v124 and v125. Seems not yet fixed.

robvanoostenrijk, 2025-04-04:
@Koenkk nothing we can do in the z2m converter to limit the bindings regardless of the firmware?

Koenkk, 2025-04-05:
There are only 3 bindings, seems the device doesn't allow for any bindings? Otherwise we can supress the errors.

magic-happenz, 2025-04-07:
The worst thing is that those devices at some point just stop reporting but instead of marking them offline, z2m just keeps reporting the same temperature and humidity. As you can imagine that's pretty bad in case any automation relies on the data. I wonder if this is something that I can at least monitor.

magic-happenz, 2025-04-16:
Another follow up issue, when trying to run OTA updates with some of the devices the whole add-on crashes. 
`[2025-04-16 18:48:24] error: 	zh:ember:uart:ash: Received ERROR from adapter, with code=ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT.
[2025-04-16 18:48:24] error: 	zh:ember:uart:ash: ASH disconnected | Adapter status: ASH_NCP_FATAL_ERROR
[2025-04-16 18:48:24] error: 	zh:ember:uart:ash: Error while parsing received frame, status=ASH_NCP_FATAL_ERROR.
[2025-04-16 18:48:24] error: 	zh:ember: Adapter fatal error: HOST_FATAL_ERROR`

magic-happenz, 2025-05-15:
I am still having this problem with no idea how to solve. Any help would be much appreciated.

P4uLT, 2025-05-25:
Hello

I'm getting quite the same.

I had a device but due to an empty cell, I need to get it back on track.
Cell changed but as the value didnt changed, I decided to remove it to reappair it

`[2025-05-25 22:37:11] info: 	z2m: Configuring '0xa4c138a70204c533'
[2025-05-25 22:37:11] debug: 	zh:controller:endpoint: Bind 0xa4c138a70204c533/1 msTemperatureMeasurement from '0xe0798dfffea68b39/1'
[2025-05-25 22:37:11] debug: 	zh:ember: ~~~> [ZDO BIND_REQUEST UNICAST to=0xa4c138a70204c533:4453 messageTag=20 payload=1433c50402a738c1a401020403398ba6feff8d79e001]
[2025-05-25 22:37:11] debug: 	zh:ember:ezsp: ===> [FRAME: ID=52:"SEND_UNICAST" Seq=7 Len=44]
[2025-05-25 22:37:11] debug: 	zh:ember:uart:ash: ---> [FRAME type=DATA frmTx=7 frmRx=1](ackRx=7)
[2025-05-25 22:37:11] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2025-05-25 22:37:11] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=0](ackRx=7 frmTx=0)
[2025-05-25 22:37:11] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=0 frmNum=1](frmRx=1) Added to rxQueue
[2025-05-25 22:37:11] debug: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=2](ackRx=0)
[2025-05-25 22:37:11] debug: 	zh:ember:ezsp: <=== [FRAME: ID=52:"SEND_UNICAST" Seq=7 Len=10]
[2025-05-25 22:37:11] debug: 	zh:ember: ~~~> [SENT ZDO UNICAST messageTag=20 apsSequence=76 status=OK]
[2025-05-25 22:37:11] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2025-05-25 22:37:11] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=0](ackRx=0 frmTx=0)
[2025-05-25 22:37:11] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=0 frmNum=2](frmRx=2) Added to rxQueue
[2025-05-25 22:37:11] debug: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=3](ackRx=0)
[2025-05-25 22:37:11] debug: 	zh:ember:ezsp: <=== [CBFRAME: ID=69:"INCOMING_MESSAGE_HANDLER" Seq=7 Len=45]
[2025-05-25 22:37:11] debug: 	zh:ember:ezsp: ezspIncomingMessageHandler: type=UNICAST apsFrame={"profileId":260,"clusterId":1026,"sourceEndpoint":2,"destinationEndpoint":1,"options":256,"groupId":0,"sequence":5} packetInfo:{"senderShortId":4519,"senderLongId":"0x0000000000000000","bindingIndex":255,"addressIndex":255,"lastHopLqi":212,"lastHopRssi":-58,"lastHopTimestamp":716542075} messageContents=08e60a0000293e08
[2025-05-25 22:37:11] debug: 	zh:controller: Received payload: clusterID=1026, address=4519, groupID=0, endpoint=2, destinationEndpoint=1, wasBroadcast=false, linkQuality=212, frame={"header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":230,"commandIdentifier":10},"payload":[{"attrId":0,"dataType":41,"attrData":2110}],"command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}}
[2025-05-25 22:37:11] debug: 	zh:controller:endpoint: ZCL command 0xa4c1386000c538ae/2 msTemperatureMeasurement.defaultRsp({"cmdId":10,"statusCode":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"reservedBits":0,"transactionSequenceNumber":230,"writeUndiv":false})
[2025-05-25 22:37:11] debug: 	zh:ember: ~~~> [ZCL to=0xa4c1386000c538ae:4519 apsFrame={"profileId":260,"clusterId":1026,"sourceEndpoint":1,"destinationEndpoint":2,"options":4352,"groupId":0,"sequence":0} header={"frameControl":{"reservedBits":0,"frameType":0,"direction":1,"disableDefaultResponse":true,"manufacturerSpecific":false},"transactionSequenceNumber":230,"commandIdentifier":11}]
[2025-05-25 22:37:11] debug: 	zh:ember:ezsp: ===> [FRAME: ID=52:"SEND_UNICAST" Seq=8 Len=27]
[2025-05-25 22:37:11] debug: 	zh:ember:uart:ash: ---> [FRAME type=DATA frmTx=0 frmRx=3](ackRx=0)
[2025-05-25 22:37:11] debug: 	z2m: Received Zigbee message from 'Thermometre Imprimante', type 'attributeReport', cluster 'msTemperatureMeasurement', data '{"measuredValue":2110}' from endpoint 2 with groupID 0
[2025-05-25 22:37:11] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Thermometre Imprimante', payload '{"battery":60,"humidity":58.7,"linkquality":212,"temperature":21.1,"update":{"installed_version":68,"latest_version":68,"state":"idle"},"voltage":2900}'
[2025-05-25 22:37:11] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2025-05-25 22:37:11] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=1](ackRx=0 frmTx=1)
[2025-05-25 22:37:11] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=1 frmNum=3](frmRx=3) Added to rxQueue
[2025-05-25 22:37:11] debug: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=4](ackRx=1)
[2025-05-25 22:37:11] debug: 	zh:ember:ezsp: <=== [FRAME: ID=52:"SEND_UNICAST" Seq=8 Len=10]
[2025-05-25 22:37:11] debug: 	zh:ember:ezsp: ~~~> [SENT type=DIRECT apsSequence=77 messageTag=80 status=OK]
[2025-05-25 22:37:11] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2025-05-25 22:37:11] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=1](ackRx=1 frmTx=1)
[2025-05-25 22:37:11] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=1 frmNum=4](frmRx=4) Added to rxQueue
[2025-05-25 22:37:11] debug: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=5](ackRx=1)
[2025-05-25 22:37:11] debug: 	zh:ember:ezsp: <=== [CBFRAME: ID=63:"MESSAGE_SENT_HANDLER" Seq=8 Len=26]
[2025-05-25 22:37:11] debug: 	zh:ember:ezsp: ezspMessageSentHandler: status=OK type=DIRECT indexOrDestination=4519 apsFrame={"profileId":260,"clusterId":1026,"sourceEndpoint":1,"destinationEndpoint":2,"options":4352,"groupId":0,"sequence":77} messageTag=80
[... cut ...]
    at Immediate.<anonymous> (/app/node_modules/.pnpm/zigbee-herdsman@4.0.0/node_modules/zigbee-herdsman/src/adapter/ember/uart/ash.ts:1192:45)
    at processImmediate (node:internal/timers:491:21))
[2025-05-25 22:37:15] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/configure', payload '{"data":{},"error":"Failed to configure (Bind 0xa4c138a70204c533/1 msTemperatureMeasurement from '0xe0798dfffea68b39/1' failed (Delivery failed for '4453'.))","status":"error","transaction":"tn27k-15"}'
`

krombel, 2025-07-25:
Not stale

jonnycastaway, 2025-08-14:
Hi, get these error in every Update from the Sensor (Temp., Hum., Batt).
Is spamming the log:

`z2m: Failed to configure '0xa4c1385c775ad4d2', attempt 2 (Error: Bind 0xa4c1385c775ad4d2/1 msTemperatureMeasurement from '0x00124b002c3fc217/1' failed (Status 'TABLE_FULL') at Endpoint.bind (/opt/zigbee2mqtt/node_modules/.pnpm/zigbee-herdsman@6.0.0/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:602:23) at processTicksAndRejections (node:internal/process/task_queues:105:5) at Object.bind (/opt/zigbee2mqtt/node_modules/.pnpm/zigbee-herdsman-converters@25.1.0/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:40:9) at configure (/opt/zigbee2mqtt/node_modules/.pnpm/zigbee-herdsman-converters@25.1.0/node_modules/zigbee-herdsman-converters/src/devices/custom_devices_diy.ts:872:13) at Object.configure (/opt/zigbee2mqtt/node_modules/.pnpm/zigbee-herdsman-converters@25.1.0/node_modules/zigbee-herdsman-converters/src/index.ts:366:21) at Configure.configure (/opt/zigbee2mqtt/lib/extension/configure.ts:123:13) at EventEmitter.wrappedCallback (/opt/zigbee2mqtt/lib/eventBus.ts:252:17))`

Its always the msTemperatureMeasurement, also when humidity is updates.
But the correct Data is incomming.

closed 2025-10-21 by github-actions[bot]