Koenkk/zigbee2mqtt #26897: To Zigbee processing of MQTT messages

jophar, 2025-03-28:
### What happened?

Hello,

I've been trying to look into this but haven't had any success.

When defining a custom cluster, I need to specify the body of the payload being sent to the device. The issue I'm encountering is that the payload contains non-primitive elements, such as an array of objects.

I'm considering processing this message in toZigbee, but I have no control over the payload processing when the message comes through MQTT.

Is there anything I can do?

Thank you in advance.

### What did you expect to happen?

On message read on mqtt, should parse info through toZigbee (even if custom) before sending to device. Message has to be manipulated before sending to device.

### Zigbee2MQTT version

2.1.3

### Adapter firmware version

20190425

### Adapter

Texas_Instruments_TI_CC2531_USB_CDC

### Setup

Linux baremetal install

Koenkk, 2025-03-30:
> I'm considering processing this message in toZigbee, but I have no control over the payload processing when the message comes through MQTT.

Could you provide an example?

jophar, 2025-03-30:
Yes, absolutely.

external converter:

`const customCluster = {
    ID: <an ID>,
    manufacturerCode: <manufCode>,
    attributes: {
        gpsStatus: { ID: 0x1000, type: Zcl.DataType.UINT8 },
        gpsLatitude: { ID: 0x1001, type: Zcl.DataType.SINGLE_PREC },
        gpsLongitude: { ID: 0x1002, type: Zcl.DataType.SINGLE_PREC }
        ...
    },
    commands: {
        setLocationRequest: {
            ID: 0x23, //35
            parameters: [
                { name: 'LocationLongitude', type: Zcl.DataType.SINGLE_PREC },
                { name: 'LocationLatitude', type: Zcl.DataType.SINGLE_PREC },
            ]
        }
    }
};
`

in this example of custom cluster, if I write on the mqtt server, on "baseTopic/deviceId/endpoint/set" the payload:

`command : {
    cluster : <cluster>
    command: <command>
    payload: {
        LocationLongitude: <longitude>,
        LocationLatitude: <latitude>
    }
}`

This payload is immediatly send to the device. But maybe I could want to remove precision to this call.

How can I edit this payload, in z2m before it gets sent to the device?

Thank you!

Koenkk, 2025-03-30:
This indeed directly sends the command, to convert you have to write a custom toZigbee converter, you can find many examples here; https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/converters/toZigbee.ts

jophar, 2025-03-30:
Thanks for your help and reply.

Although I tried the custom toZigbee, even if importing it in the definition, the command read on mqtt says that the cluster X does not have the command Y (this if I remove the cluester definition to the command).
Also, and this is a question that I have, if I send the command by mqtt and I don't have the cluster definition for that command, how does z2m knows what TZ to fire?

Koenkk, 2025-03-31:
Could you provide your complete external converter?

jophar, 2025-04-01:
```
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const Zcl = require('zigbee-herdsman').Zcl;
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const constants = require('zigbee-herdsman-converters/lib/constants');
const ota = require('zigbee-herdsman-converters/lib/ota');
const utils = require('zigbee-herdsman-converters/lib/utils');
const types = require('zigbee-herdsman-converters/lib/types');
const globalStore = require('zigbee-herdsman-converters/lib/store');

const ea = exposes.access;
const e = exposes.presets;

const customCluster = {
    ID: "customId",
    manufacturerCode: "customCode",
    attributes: {
        gpsStatus: { ID: "customId", type: Zcl.DataType.UINT8 },
        gpsLatitude: { ID: "customId", type: Zcl.DataType.SINGLE_PREC },
        gpsLongitude: { ID: "customId", type: Zcl.DataType.SINGLE_PREC },
    },
    commands: {
        getLocationRequest: {
            ID: "customId",
            parameters: []
        },
        setLocationRequest: {
            ID: "customId",
            parameters: [
                { name: 'LocationLongitude', type: Zcl.DataType.SINGLE_PREC },
                { name: 'LocationLatitude', type: Zcl.DataType.SINGLE_PREC },
            ]
        },
    },
    commandsResponse: {
        getLocationResponse: {
            ID: "customId",
            parameters: [
                { name: 'gpsLatitude', type: Zcl.DataType.SINGLE_PREC },
                { name: 'gpsLongitude', type: Zcl.DataType.SINGLE_PREC },
                { name: 'gpsStatus', type: Zcl.DataType.UINT8 }
            ]
        },
        setLocationResponse: {
            ID: "customId",
            parameters: [
                { name: 'ExecutionResult', type: Zcl.DataType.UINT8 }
            ]
        }
    }
}

const fromZigbee = {
    GetLocationResponse : { 
        cluster: 'customCluster',
        type: 'commandGetLocationResponse',
        convert: async (model, msg, publish, options, meta) => { 
            if(msg.data === undefined) {
                return {};
            }
            return {
                gpsLatitude: msg.data.gpsLatitude,
                gpsLongitude: msg.data.gpsLongitude,
                gpsStatus: msg.data.gpsStatus
            };
        }
    },
    SetLocationResponse : { 
        cluster: 'customCluster',
        type: 'commandSetLocationResponse',
        convert: async (model, msg, publish, options, meta) => { 
            if(msg.data === undefined) {
                return {};
            }
            return { 
                ExecutionResult : msg.data.ExecutionResult 
            };
        }
    }
}

const toZigbee = {
    get_gps_location: {
        convertSet: async (entity, key, meta, value) => {
            return await entity.command('customCluster', 'getLocationRequest', 0);
        }
    },
    set_gps_location: { 
        key: ['LocationLongitude', 'LocationLatitude'],
        convertSet: async (entity, key, value, meta) => {
            let payload = meta.payload;

            // Here is where I should be able to modify the payload
            // but it is sent to the device straight away

            await entity.command('customCluster', 'setLocationRequest', { payload });
        }
    }
}

const definition = {
    zigbeeModel: ['110'],
    model: 'LIGHT',
    vendor: 'Proprietary',
    description: 'Proprietary Smart Light Controller',
    fromZigbee: [
        fromZigbee.GetLocationResponse,
        fromZigbee.SetLocationResponse,
    ],
    toZigbee: [
        toZigbee.get_gps_location,
        toZigbee.set_gps_locatoin
    ],
    exposes: [e.voltage(), e.power_factor(), e.current(), e.energy(), e.switch() ],
    extend: [modernExtend.deviceAddCustomCluster('customCluster', customCluster)],
    ota: true,
    meta: {disableDefaultResponse: true},
    configure: async (device, coordinatorEndpoint, definition) => {
        const endpoint = device.getEndpoint(10);
        const endpoint_OTA = device.getEndpoint(15);
        const utcTime = Math.round(new Date().getTime() / 1000);
        const localTime = utcTime - new Date().getTimezoneOffset() * 60;
        const timeValues = { time: localTime, timeStatus: 2 }
        await endpoint_OTA.write('genTime', timeValues);
        const binds = ['customCluster'];
        await reporting.bind(endpoint, coordinatorEndpoint, binds);
        
    },
};

module.exports = definition;
```

Please ignore the "customId" and "customCode", I have to mask them due to proprietary reasons.
What I think it should happen is, when a toZigbee.set_gps_location is called via mqtt by posting the payload:

```
{
  "command": {
    "cluster": "customCluster",
    "command": "customCommand",
    "payload": {
      "LocationLongitude": 0,
      "LocationLatitude": 0
    }
  }
}
```
Z2m reacts acoardingly by sending the command to the device with the paylod but, it doesn't go through the toZigbee configuration but (my wild guess) goes straight through the cluster definition.

So, my question is, if I want to integrate something that publishes in the mqtt server and for some reason the payload that's published needs to be modified before being sent to the device, how can I acheive that.

Thank you for your work and help!
Cheers

Koenkk, 2025-04-01:
Try with:

```js
    set_gps_location: { 
        key: ['location'],
        convertSet: async (entity, key, value, meta) => {
            let payload = meta.payload;

            // Here is where I should be able to modify the payload
            // but it is sent to the device straight away

            await entity.command('customCluster', 'setLocationRequest', { "LocationLongitude": payload.longitude,   "LocationLatitude": payload.latitude });
        }
    }
```

Then publish: `zigbee2mqtt/device_friendly_name/set` payload `{"location": {"longitude": 1, "latitude": 2}}`

jophar, 2025-04-02:
@Koenkk 
Thank you for your reply.
In fact I am able to enter custom toZigbee to process the message before sending.
Now, I am facing other challenge which is the DataType beeing sent.
As I posted above, there is an hardware limitation where the value of the property send is impossible on Javascript Object definition.
```"this_property" : ['lat' : 'value_1', 'lon' : 'value_2', 'lat' : 'value_3', 'lon' : 'value_4'] ```
How can I define a DataType for this? Or, is there any way to bypass the type restriction and just send the payload to the device? (maybe I'm missing some Zigbee low level concepts and this is driving me crazy)

Koenkk, 2025-04-02:
What you do in the toZigbee converter is basically convert the javascript object to an object matching your custom defined command, e.g. 

```
        setLocationRequest: {
            ID: "customId",
            parameters: [
                { name: 'LocationLongitude', type: Zcl.DataType.SINGLE_PREC },
                { name: 'LocationLatitude', type: Zcl.DataType.SINGLE_PREC },
            ]
        },
```

(this is what the example converter shows)

jophar, 2025-04-03:
Hello @Koenkk 
Thanks your all your help
I was able to send the custom data to the device!

Closing

closed 2025-04-03 by jophar