Koenkk/zigbee2mqtt #26898: Awox 33952 remote control problem with action command

OUARZA, 2025-03-28:
### What happened?

Hello,
There's a problem with this remote control when pressing the buttons.
In fact, the payload shows the information about the button pressed, but in the same second, the payload is "on," which poses a problem when creating scenarios.
Can you fix this?
Thank you.

https://www.zigbee2mqtt.io/devices/33952.html#awox-33952

### What did you expect to happen?

When a button is pressed, the button pressed should be displayed in the payload, but only the information about the button.

### How to reproduce it (minimal and precise)

`info 2025-03-28 19:37:56z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xa4c13809e93e2d35', payload '{"action":"yellow","action_group":32776,"device":{"applicationVersion":2,"friendlyName":"0xa4c13809e93e2d35","hardwareVersion":60,"ieeeAddr":"0xa4c13809e93e2d35","manufacturerID":4417,"manufacturerName":"AwoX","model":"33952","networkAddress":46979,"powerSource":"Battery","softwareBuildID":"2.7.6_1477","stackVersion":1,"type":"EndDevice","zclVersion":3},"last_seen":"2025-03-28T18:37:56.310Z","linkquality":255}'
info 2025-03-28 19:37:57z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xa4c13809e93e2d35', payload '{"action":"on","action_group":32776,"device":{"applicationVersion":2,"friendlyName":"0xa4c13809e93e2d35","hardwareVersion":60,"ieeeAddr":"0xa4c13809e93e2d35","manufacturerID":4417,"manufacturerName":"AwoX","model":"33952","networkAddress":46979,"powerSource":"Battery","softwareBuildID":"2.7.6_1477","stackVersion":1,"type":"EndDevice","zclVersion":3},"last_seen":"2025-03-28T18:37:56.358Z","linkquality":248}'
info 2025-03-28 19:37:57z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xa4c13809e93e2d35', payload '{"device":{"applicationVersion":2,"friendlyName":"0xa4c13809e93e2d35","hardwareVersion":60,"ieeeAddr":"0xa4c13809e93e2d35","manufacturerID":4417,"manufacturerName":"AwoX","model":"33952","networkAddress":46979,"powerSource":"Battery","softwareBuildID":"2.7.6_1477","stackVersion":1,"type":"EndDevice","zclVersion":3},"last_seen":"2025-03-28T18:37:56.358Z","linkquality":248}'
info 2025-03-28 19:37:59z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xa4c13809e93e2d35', payload '{"action":"refresh","action_group":32776,"device":{"applicationVersion":2,"friendlyName":"0xa4c13809e93e2d35","hardwareVersion":60,"ieeeAddr":"0xa4c13809e93e2d35","manufacturerID":4417,"manufacturerName":"AwoX","model":"33952","networkAddress":46979,"powerSource":"Battery","softwareBuildID":"2.7.6_1477","stackVersion":1,"type":"EndDevice","zclVersion":3},"last_seen":"2025-03-28T18:37:58.340Z","linkquality":255}'
info 2025-03-28 19:37:59z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xa4c13809e93e2d35', payload '{"action":"on","action_group":32776,"device":{"applicationVersion":2,"friendlyName":"0xa4c13809e93e2d35","hardwareVersion":60,"ieeeAddr":"0xa4c13809e93e2d35","manufacturerID":4417,"manufacturerName":"AwoX","model":"33952","networkAddress":46979,"powerSource":"Battery","softwareBuildID":"2.7.6_1477","stackVersion":1,"type":"EndDevice","zclVersion":3},"last_seen":"2025-03-28T18:37:58.386Z","linkquality":255}'
info 2025-03-28 19:37:59z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xa4c13809e93e2d35', payload '{"device":{"applicationVersion":2,"friendlyName":"0xa4c13809e93e2d35","hardwareVersion":60,"ieeeAddr":"0xa4c13809e93e2d35","manufacturerID":4417,"manufacturerName":"AwoX","model":"33952","networkAddress":46979,"powerSource":"Battery","softwareBuildID":"2.7.6_1477","stackVersion":1,"type":"EndDevice","zclVersion":3},"last_seen":"2025-03-28T18:37:58.386Z","linkquality":255}'
info 2025-03-28 19:38:00z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xa4c13809e93e2d35', payload '{"action":"brightness_step_up","action_brightness_delta":18,"action_group":32776,"action_step_size":18,"action_transition_time":0.02,"brightness":255,"device":{"applicationVersion":2,"friendlyName":"0xa4c13809e93e2d35","hardwareVersion":60,"ieeeAddr":"0xa4c13809e93e2d35","manufacturerID":4417,"manufacturerName":"AwoX","model":"33952","networkAddress":46979,"powerSource":"Battery","softwareBuildID":"2.7.6_1477","stackVersion":1,"type":"EndDevice","zclVersion":3},"last_seen":"2025-03-28T18:37:59.889Z","linkquality":255}'
info 2025-03-28 19:38:00z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xa4c13809e93e2d35', payload '{"brightness":255,"device":{"applicationVersion":2,"friendlyName":"0xa4c13809e93e2d35","hardwareVersion":60,"ieeeAddr":"0xa4c13809e93e2d35","manufacturerID":4417,"manufacturerName":"AwoX","model":"33952","networkAddress":46979,"powerSource":"Battery","softwareBuildID":"2.7.6_1477","stackVersion":1,"type":"EndDevice","zclVersion":3},"last_seen":"2025-03-28T18:37:59.889Z","linkquality":255}'`

### Zigbee2MQTT version

 1.42.0 commit: 861cba63

### Adapter firmware version

0x26780700

### Adapter

ConBee2/RaspBee2

### Setup

Intel NUC

### Debug log

[2025-03-26.21-00-24-3.zip](https://github.com/user-attachments/files/19510642/2025-03-26.21-00-24-3.zip)

OUARZA, 2025-04-03:
up :)

OUARZA, 2025-04-07:
Hello @Koenkk  can you help me?

Koenkk, 2025-04-10:
Could you provide the debug log of only the duplicate `on` payload?

See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

OUARZA, 2025-04-11:
@Koenkk 
Hello, Here are the logs in debug mode.
I've put the double values ​​of 0xa4c13809e93e2d35 at the very bottom.

[log (2).log](https://github.com/user-attachments/files/19712742/log.2.log)

Koenkk, 2025-04-12:
It seems the remote indeed send the double commands, but we cannot filter them as the `transactionSequenceNumber` for both differ. Maybe as a workaround you can `debounce` the `action` property, docs: https://www.zigbee2mqtt.io/guide/configuration/devices-groups.html

OUARZA, 2025-04-13:
Hello @Koenkk , is this it?
It seems to work with this configuration. :)

[image]

Koenkk, 2025-04-14:
Yes, great!

closed 2025-04-14 by Koenkk

OUARZA, 2025-05-23:
Hello @Koenkk 
I looked into creating an external_converters file.
I think my file isn't being used. What should I do?
html/plugins/z2m/core/config/converters/AwoX/AwoX_33952.js
`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');

const eglo_custom_converter = {
    fingerprint: [
        { modelID: '33952', manufacturerName: 'AwoX' },
    ],
    fromZigbee: [
        // Power
        {
            cluster: 'genOnOff',
            type: ['commandOn'],
            convert: () => ({ action: 'power_on' }),
        },
        {
            cluster: 'genOnOff',
            type: ['commandOff'],
            convert: () => ({ action: 'power_off' }),
        },

        // Luminosité
        {
            cluster: 'genLevelCtrl',
            type: ['commandStepWithOnOff'],
            convert: () => ({ action: 'brightness_up' }),
        },
        {
            cluster: 'genLevelCtrl',
            type: ['commandStep'],
            convert: () => ({ action: 'brightness_down' }),
        },

        // Couleur chaude / froide
        {
            cluster: 'lightingColorCtrl',
            type: ['commandStepColorTemp'],
            convert: (model, msg) => {
                const stepmode = msg.data.stepmode;
                return {
                    action: stepmode === 1 ? 'color_temp_warmer' : 'color_temp_cooler'
                };
            },
        },

        // Mouvement lumière circulaire
        {
            cluster: 'lightingColorCtrl',
            type: ['commandEnhancedMoveHue'],
            convert: () => ({ action: 'cycle_light' }),
        },

        // Favoris
        {
            cluster: 'genScenes',
            type: ['commandRecall'],
            convert: (model, msg) => {
                return { action: `favorite_${msg.data.sceneid}` };
            },
        },

        // Boutons couleur en RAW
        {
            cluster: 'lightingColorCtrl',
            type: ['raw'],
            convert: (model, msg) => {
                const buffer = msg.data.data;
                const hex = buffer.map(b => b.toString(16).padStart(2, '0')).join('');

                const colorMap = {
                    '11d03000fe0800': 'red',
                    '11d230002a0800': 'yellow',
                    '11d43000540800': 'green',
                    '11d63000a90800': 'blue',
                };

                const color = colorMap[hex.toLowerCase()];
                if (color) {
                    return { action: `color_${color}` };
                }
                return;
            },
        },

        // Refresh (bouton sans nom ?)
        {
            cluster: 'genLevelCtrl',
            type: ['raw'],
            convert: (model, msg) => {
                const hex = msg.data.data.map(b => b.toString(16).padStart(2, '0')).join('');
                if (hex === '11df100101') {
                    return { action: 'refresh' };
                }
                return;
            },
        },
    ],
    toZigbee: [],
    exposes: [
        {
            type: 'enum',
            name: 'action',
            values: [
                'power_on', 'power_off',
                'brightness_up', 'brightness_down',
                'color_temp_warmer', 'color_temp_cooler',
                'cycle_light',
                'color_red', 'color_green', 'color_blue', 'color_yellow',
                'favorite_1', 'favorite_2',
                'refresh'
            ],
        },
    ],
};

module.exports = eglo_custom_converter;`

configuration.yaml :

[image]

OUARZA, 2025-05-23:
Subject: [New device support] AwoX Remote Controller (Model: 33952, ERCU_Zm)

---

### Description of the device

This PR adds full support for the **AwoX Remote Controller, Model 33952 (Zigbee Model ID: ERCU_Zm)**.

This is a multi-function remote that supports on/off, dimming, scene control, and color/color temperature adjustments.

### Device Interview Details

* **Model:** 33952
* **Vendor:** AwoX
* **Zigbee Model ID:** ERCU_Zm
* **Manufacturer Name:** AwoX
* **Endpoint:** 1
* **Power Source:** Battery (confirmed by device type "EndDevice" and typical remote behavior)

---

### Device Functionality (from logs)

I have captured detailed debug logs for each button press on the remote. The remote sends specific cluster commands and `raw` data payloads for various functions, alongside a generic `commandOn` on `genOnOff` for many actions. The proposed converter prioritizes specific actions to avoid redundant `on` commands.

Here's a breakdown of the observed functionality and corresponding Zigbee messages:

* **On:** `commandOn` on `genOnOff`
* **Off:** `commandOff` on `genOnOff`
* **Color Buttons (Blue, Green, Yellow, Red):** `raw` messages on `lightingColorCtrl` cluster. The 5th byte (`data[4]`) in the payload differentiates the colors:
    * Blue: `{"data":[17,214,48,0,169,8,0],"type":"Buffer"}` (0xD6)
    * Green: `{"data":[17,212,48,0,84,8,0],"type":"Buffer"}` (0xD4)
    * Yellow: `{"data":[17,210,48,0,42,8,0],"type":"Buffer"}` (0xD2)
    * Red: `{"data":[17,208,48,0,254,8,0],"type":"Buffer"}` (0xD0)
* **Brightness+:** `commandStepWithOnOff` on `genLevelCtrl` with `{"stepmode":0,"stepsize":18,"transtime":2}`
* **Brightness-:** `commandStep` on `genLevelCtrl` with `{"stepmode":1,"stepsize":18,"transtime":2}`
* **Light Movement (e.g., color cycle):** `commandEnhancedMoveHue` on `lightingColorCtrl` with `{"movemode":1,"rate":3072}`
* **"Refresh" Button (unlabeled):** `raw` message on `genLevelCtrl` with `{"data":[17,223,16,1,1],"type":"Buffer"}` (identified by `data[1]` being `0xDF`)
* **Favorite 1:** `commandRecall` on `genScenes` with `{"groupid":0,"sceneid":1}`
* **Favorite 2:** `commandRecall` on `genScenes` with `{"groupid":0,"sceneid":2}`
* **Warm Color:** `commandStepColorTemp` on `lightingColorCtrl` with `{"maximum":0,"minimum":0,"stepmode":1,"stepsize":32,"transtime":2}`
* **Cold Color:** `commandStepColorTemp` on `lightingColorCtrl` with `{"maximum":0,"minimum":0,"stepmode":3,"stepsize":32,"transtime":2}`

---

### Proposed Converter Code

I have developed a custom `fromZigbee` converter that correctly parses these messages and exposes appropriate actions.

```javascript
// Add this definition to a new file or an appropriate existing file in the Zigbee2MQTT device definitions.
// Example: devices/awox.js or a new file like devices/awox_33952.js

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['ERCU_Zm'],
    model: '33952',
    vendor: 'AwoX',
    description: 'AwoX Remote Controller',
    supports: 'on/off, dimming, scene control, color control, color temperature control',
    fromZigbee: [fz.legacy_action_rate, fz.on_off, fz.awox_remote_actions], // Using a unique name for the custom converter
    toZigbee: [],
[... 52 lines cut ...]
                if (payload.sceneid === 1) {
                    action = 'scene_1';
                } else if (payload.sceneid === 2) {
                    action = 'scene_2';
                }
            }
        }
        // Handle generic On/Off last if no specific action is found
        else if (msg.cluster === 'genOnOff') {
            if (msg.type === 'commandOn') {
                action = 'on';
            } else if (msg.type === 'commandOff') {
                action = 'off';
            }
        }

        if (action) {
            return {action: action};
        }
    },
};

module.exports = [
    definition,
];