Koenkk/zigbee2mqtt #26899: Tuya TS011F_plug_2 UNSUPPORTED_ATTRIBUTE

OUARZA, 2025-03-28:
### What happened?

Hello,
I can't configure this outlet.
I'm getting errors.

https://www.zigbee2mqtt.io/devices/TS011F_plug_2.html#tuya-ts011f_plug_2

### What did you expect to happen?

`info 2025-03-28 19:51:01z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x70b3d52b6000778d', payload '{"device":{"applicationVersion":1,"dateCode":"20210720","friendlyName":"0x70b3d52b6000778d","hardwareVersion":2,"ieeeAddr":"0x70b3d52b6000778d","manufacturerID":0,"manufacturerName":"_TZ3000_bfn1w0mm","model":"TS011F_plug_2","networkAddress":45743,"powerSource":"Mains (single phase)","softwareBuildID":"20B-TZSKT11BS3.2","stackVersion":1,"type":"Router","zclVersion":2},"last_seen":"2025-03-28T18:51:00.770Z","linkquality":255,"state":"ON"}'
info 2025-03-28 19:51:01z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x70b3d52b6000778d', payload '{"device":{"applicationVersion":1,"dateCode":"20210720","friendlyName":"0x70b3d52b6000778d","hardwareVersion":2,"ieeeAddr":"0x70b3d52b6000778d","manufacturerID":0,"manufacturerName":"_TZ3000_bfn1w0mm","model":"TS011F_plug_2","networkAddress":45743,"powerSource":"Mains (single phase)","softwareBuildID":"20B-TZSKT11BS3.2","stackVersion":1,"type":"Router","zclVersion":2},"last_seen":"2025-03-28T18:51:00.772Z","linkquality":255,"state":"ON"}'
error 2025-03-28 19:51:01z2m: Publish 'set' 'indicator_mode' to '0x70b3d52b6000778d' failed: 'Error: ZCL command 0x70b3d52b6000778d/1 genOnOff.write({"tuyaBacklightMode":2}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
info 2025-03-28 19:51:01z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'indicator_mode' to '0x70b3d52b6000778d' failed: 'Error: ZCL command 0x70b3d52b6000778d/1 genOnOff.write({\"tuyaBacklightMode\":2}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"reservedBits\":0,\"writeUndiv\":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'","meta":{"friendly_name":"0x70b3d52b6000778d"},"type":"zigbee_publish_error"`

### Zigbee2MQTT version

1.42.0 commit: 861cba63

### Adapter firmware version

ConBee2/RaspBee2

### Adapter

ConBee2/RaspBee2

### Setup

Intel NUC

### Debug log

[2025-03-26.21-00-24-3.zip](https://github.com/user-attachments/files/19510779/2025-03-26.21-00-24-3.zip)

OUARZA, 2025-04-03:
up :)

OUARZA, 2025-04-07:
Hello @Koenkk  can you help me?

OUARZA, 2025-04-08:
@Koenkk What can I do to integrate this function?

Koenkk, 2025-04-09:
@OUARZA I don't think you cannot do it, the firmware of your device doesn't support it.

closed 2025-06-17 by github-actions[bot]