Koenkk/zigbee2mqtt #26903: EGLO 900091 Lamp – Separate Control of CCW Front Panel and RGB Ambient Lighting via mqtt2zigbee

LupusR3x, 2025-03-29:
### What happened?

I am using an EGLO Lamp (model 900091) which includes two distinct lighting components:

- CCW (Cold/Warm White) front panel
- RGB ambient lighting

Currently, when using mqtt2zigbee, it's not possible to separately control these two lighting units. However, the lamp itself supports independent operation of the CCW panel and RGB ambient lighting (verified functionality). Attached is an image showing the device's capabilities.
[image]

### What did you expect to happen?

It should be possible to independently control the CCW front panel and the RGB ambient lighting using mqtt2zigbee.

### How to reproduce it (minimal and precise)

Both lighting components cannot be separately addressed; commands sent via mqtt2zigbee affect both units simultaneously.

### Zigbee2MQTT version

2.1.3

### Adapter firmware version

6.10.3.0 build 297

### Adapter

Sonoff Zigbee 3.0 USB-Dongle Plus E

### Setup

Add-on on HAOS on HP EliteDesk

TheAndi, 2025-04-16:
Any progress on this problem? I have a similar kind of EGLO lamp and face exactly the same issue. 

In these lamps, the RGB backlight is a **separate** light source to the main CCT one. 
Would it be possible to change the lamp's Z2M config so that the Z2M provides two light entities to HA? 

Any help is appreciated!

LupusR3x, 2025-06-23:
Issue Remains. Does anybody has a solution?

closed 2025-09-01 by github-actions[bot]

dann032, 2025-09-26:
Any news about this problem? 
Separate control of the 2 lights would be great!

Koenkk, 2025-10-01:
Could you check if the issue is fixed with the following [external converter](https://gist.github.com/Koenkk/02a91d57937d897e23234bbcfdab37b8): 

- save this as file next to `configuration.yaml` in the `external_converters` directory (create if it doesn't exist yet) as `ext_converter.js` (so `external_converters/ext_converter.mjs`)
- start z2m, 
- in the frontend, check if the description of this device now ends with `CUSTOM` (this indicates the external converter has been loaded correctly)
- check if issue is fixed

Koenkk, 2025-10-02:
Updated https://gist.github.com/Koenkk/02a91d57937d897e23234bbcfdab37b8, can you try again?

Koenkk, 2025-10-04:
It seems they use a non-standard way, can someone sniff the traffic when controlling it via the original hub? https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html