Koenkk/zigbee2mqtt #26905: Fix converter for Tuya SDM01 3-phase power meter after upgrading to firmware v1.0.6

CodeFinder2, 2025-03-29:
### What happened?

After upgrading [this device](https://www.zigbee2mqtt.io/devices/SDM01.html) from v1.0.3 to v1.0.6 (via Smart Life app), the reported values where incorrect as already reported in #24151 .

I can confirm that the following external converter provided by @rp2019433 in #24151 is working fine with device firmware v1.0.6:
```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE204_ugekduaj'}, {modelID: 'TS0601', manufacturerName: '_TZE200_ugekduaj'},],
    model: 'SDM01',
    vendor: 'TuYa',
    description: 'Smart energy monitor for 3P+N system (with firmware v1.0.6)',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    configure: tuya.configureMagicPacket,
    exposes:  [tuya.exposes.voltageWithPhase('A'), tuya.exposes.voltageWithPhase('B'), tuya.exposes.voltageWithPhase('C'),
        tuya.exposes.powerWithPhase('A'), tuya.exposes.powerWithPhase('B'), tuya.exposes.powerWithPhase('C'),
        tuya.exposes.currentWithPhase('A'), tuya.exposes.currentWithPhase('B'), tuya.exposes.currentWithPhase('C'),
        e.energy().withDescription('Total forward active energy'),
        e.produced_energy().withDescription('Total reverse active energy'),
        e.power_factor().withUnit('%').withDescription('Total power factor'), e.power().withDescription('Total active power'),
        e.ac_frequency(),
        tuya.exposes.energyWithPhase('A'), tuya.exposes.energyWithPhase('B'), tuya.exposes.energyWithPhase('C'),
        tuya.exposes.energyProducedWithPhase('A'), tuya.exposes.energyProducedWithPhase('B'), tuya.exposes.energyProducedWithPhase('C'),
        tuya.exposes.powerFactorWithPhase('A'), tuya.exposes.powerFactorWithPhase('B'), tuya.exposes.powerFactorWithPhase('C')
    ],
    meta: {
        tuyaDatapoints: [
            [1, 'energy', tuya.valueConverter.divideBy100],
            [23, 'produced_energy', tuya.valueConverter.divideBy100],
            [29, 'power', tuya.valueConverter.raw],
            [32, 'ac_frequency', tuya.valueConverter.divideBy100],
            [50, 'power_factor', tuya.valueConverter.raw],
            [103, 'voltage_A', tuya.valueConverter.divideBy10],
            [104, 'current_A', tuya.valueConverter.divideBy1000],
            [105, 'power_A', tuya.valueConverter.raw],
            [108, 'power_factor_A', tuya.valueConverter.raw],
            [109, 'energy_A', tuya.valueConverter.divideBy100],
            [110, 'energy_produced_A', tuya.valueConverter.divideBy100],
            [112, 'voltage_B', tuya.valueConverter.divideBy10],
            [113, 'current_B', tuya.valueConverter.divideBy1000],
            [114, 'power_B', tuya.valueConverter.raw],
            [117, 'power_factor_B', tuya.valueConverter.raw],
            [118, 'energy_B', tuya.valueConverter.divideBy100],
            [119, 'energy_produced_B', tuya.valueConverter.divideBy100],
            [121, 'voltage_C', tuya.valueConverter.divideBy10],
            [122, 'current_C', tuya.valueConverter.divideBy1000],
            [123, 'power_C', tuya.valueConverter.raw],
            [126, 'power_factor_C', tuya.valueConverter.raw],
            [127, 'energy_C', tuya.valueConverter.divideBy100],
            [128, 'energy_produced_C', tuya.valueConverter.divideBy100],
        ],
    },
};

module.exports = definition;
```

Would it be

### What did you expect to happen?

Would it be possible to distinguish the firmware versions and integrate this into Z2M?

### How to reproduce it (minimal and precise)

Upgrade to v1.0.6 and continue using the default converter.

### Zigbee2MQTT version

2.1.3

### Adapter firmware version

20240710

### Adapter

TI LAUNCHXL-CC1352P-2

### Setup

HAOS von VM (x86-64) using Z2M add-on

### Debug log

No error output but incorrect values (e.g., voltage of about 0.5V where is should be around 230V, etc.).

Some more (minor) details can be found in #24151 , if relevant.

Koenkk, 2025-05-08:
Could you make a PR by clicking [here](https://github.com/Koenkk/zigbee-herdsman-converters/edit/master/src/devices/tuya.ts)?

PiotrRaszkowski, 2025-05-08:
I think somehow it should be possible... there is ts011f plug (Tuya) recognized as TS011F_plug1 if old firmware and _plug3 if upgraded firmware. I guess the same should be possible with SPM01.

PiotrRaszkowski, 2025-05-08:
For example for TS011F_plug3 we have the following:
```
 fingerprint: [
            {modelID: "TS011F", applicationVersion: 160, priority: -1},
            {modelID: "TS011F", applicationVersion: 100, priority: -1},
            {modelID: "TS011F", applicationVersion: 69, priority: -1},
            {modelID: "TS011F", applicationVersion: 68, priority: -1},
            {modelID: "TS011F", applicationVersion: 65, priority: -1},
            {modelID: "TS011F", applicationVersion: 64, priority: -1},
            {modelID: "TS011F", softwareBuildID: "1.0.5\u0000", priority: -1},
        ],
        model: "TS011F_plug_3",
        description: "Smart plug (with power monitoring by polling)",
        vendor: "Tuya",
        whiteLabel: [
            {vendor: "VIKEFON", model: "TS011F"},
            {vendor: "BlitzWolf", model: "BW-SHP15"},
            {vendor: "AVATTO", model: "MIUCOT10Z"},
            {vendor: "Neo", model: "NAS-WR01B"},
            {vendor: "Neo", model: "PLUG-001SPB2"},
        ],
```

So the fingerprint can be used to match the correct version (firmware).

I don't know if it is valid for SPM01... especially "softwareBuildID".

igorsamoylenko, 2026-11-06:
Hi @Koenkk,

I've been working on this issue and already have a working adaptive external converter for the SDM01.

I've investigated the problem with the two different Tuya DP mappings sharing the same device fingerprint. Unfortunately, applicationVersion cannot reliably distinguish between them.

I've implemented and tested a solution that handles both mappings. The newer DP mapping has been confirmed using raw reports from my actual device, and both mappings are covered by tests.

Before submitting a PR, I'd like to ask for permission because a significant part of the implementation was developed with AI assistance.

I want to be transparent about this: I've personally investigated the issue, reviewed the implementation, verified the behavior and tests, and I understand the changes being proposed. I take full responsibility for the code and will handle any review feedback myself.

Would you be willing to accept an AI-assisted PR under these conditions?

I'd be happy to submit it if that's acceptable.

Koenkk, 2026-11-06:
> Would you be willing to accept an AI-assisted PR under these conditions?

Yes, no problem!