Koenkk/zigbee2mqtt #26909: SONOFF ZBDongle-E automatically detected as zstack

kthamm, 2025-03-29:
### What happened?

I've freshly set up zigbee2mqtt on a raspberry pi 4 and connected a Sonoff ZBDongle-**E**. I did not configure anything regarding the adapter in the serial section. Zigbee2mqtt fails to start because it's automatically detected as zstack adater.

When I manually setup the serial section to adapter: ember and provide the port in the configuration, zigbee2mqtt starts as expected.

### What did you expect to happen?

I expected zigbee2mqtt to detect the adapter correctly as **ember** and to startup correctly.

### How to reproduce it (minimal and precise)

1. Fresh zigbee2mqtt installation
2. Connect flashed Sonoff ZBDongle-E
3. Startup z2m

### Zigbee2MQTT version

2.1.3

### Adapter firmware version

ncp-uart-sw_EZNet7.4.4_V1.0.0

### Adapter

Itead Sonoff Zigbee 3.0 USB Dongle Plus V2 (ZBDongle-E) ember

### Setup

Raspberry Pi 4, followed instructions from documentation

### Debug log

```
[2025-03-29 21:25:35] debug:    z2m: Removing old log directory '/opt/zigbee2mqtt/data/log/2025-03-29.21-20-07'
[2025-03-29 21:25:35] info:     z2m: Logging to console, file (filename: log.log)
[2025-03-29 21:25:35] debug:    z2m: Loaded state from file /opt/zigbee2mqtt/data/state.json
[2025-03-29 21:25:35] info:     z2m: Starting Zigbee2MQTT version 2.1.3 (commit #fc35ac5)
[2025-03-29 21:25:35] debug:    z2m: sd-notify is not installed
[2025-03-29 21:25:35] info:     z2m: Starting zigbee-herdsman (3.2.7)
[2025-03-29 21:25:35] debug:    z2m: Using zigbee-herdsman with settings: '"{\"network\":{\"panID\":33460,\"extendedPanID\":[73,249,17,195,94,172,78,36],\"channelList\":[11],\"networkKey\":\"HIDDEN\"},\"databasePath\":\"/opt/zigbee2mqtt/data/database.db\",\"databaseBackupPath\":\"/opt/zigbee2mqtt/data/database.db.backup\",\"backupPath\":\"/opt/zigbee2mqtt/data/coordinator_backup.json\",\"serialPort\":{},\"adapter\":{\"disableLED\":false}}"'
[2025-03-29 21:25:35] debug:    zh:adapter:discovery: Connected devices: [{"path":"/dev/ttyAMA5"},{"path":"/dev/ttyS0"},{"path":"/dev/ttyUSB0","manufacturer":"Itead","serialNumber":"a6ee897e4d1fef11aa004ad0639e525b","pnpId":"usb-Itead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_V2_a6ee897e4d1fef11aa004ad0639e525b-if00-port0","vendorId":"10c4","productId":"ea60"}]
[2025-03-29 21:25:35] info:     zh:adapter:discovery: Matched adapter: {"path":"/dev/ttyUSB0","manufacturer":"Itead","serialNumber":"a6ee897e4d1fef11aa004ad0639e525b","pnpId":"usb-Itead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_V2_a6ee897e4d1fef11aa004ad0639e525b-if00-port0","vendorId":"10c4","productId":"ea60"} => zstack: path=/dev/ttyUSB0, score=4
[2025-03-29 21:25:35] debug:    zh:controller: Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":"HIDDEN","panID":33460,"extendedPanID":[73,249,17,195,94,172,78,36],"channelList":[11]},"serialPort":{"path":"/dev/ttyUSB0","adapter":"zstack"},"adapter":{"disableLED":false},"databasePath":"/opt/zigbee2mqtt/data/database.db","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json"}'
[2025-03-29 21:25:35] info:     zh:zstack:znp: Opening SerialPort with {"path":"/dev/ttyUSB0","baudRate":115200,"rtscts":false,"autoOpen":false}
[2025-03-29 21:25:35] info:     zh:zstack:znp: Serialport opened
[2025-03-29 21:25:35] debug:    zh:zstack:znp: --> SREQ: SYS - ping - {"capabilities":1}
[2025-03-29 21:25:35] debug:    zh:zstack:unpi:writer: --> frame [254,0,33,1,32]
[2025-03-29 21:25:35] info:     zh:zstack:znp: Writing CC2530/CC2531 skip bootloader payload
[2025-03-29 21:25:35] debug:    zh:zstack:unpi:writer: --> buffer [239]
[2025-03-29 21:25:36] debug:    zh:zstack:znp: --> SREQ: SYS - ping - {"capabilities":1}
[2025-03-29 21:25:36] debug:    zh:zstack:unpi:writer: --> frame [254,0,33,1,32]
[2025-03-29 21:25:37] info:     zh:zstack:znp: Skip bootloader for CC2652/CC1352
[2025-03-29 21:25:37] debug:    zh:zstack:znp: --> SREQ: SYS - ping - {"capabilities":1}
[2025-03-29 21:25:37] debug:    zh:zstack:unpi:writer: --> frame [254,0,33,1,32]
[2025-03-29 21:25:38] debug:    zh:zstack:unpi:parser: <-- [26,193,2,2,155,123,126]
[2025-03-29 21:25:38] debug:    zh:zstack:unpi:parser: --- parseNext [26,193,2,2,155,123,126]
[2025-03-29 21:25:43] debug:    zh:zstack:znp: --> SREQ: SYS - ping - {"capabilities":1}
[2025-03-29 21:25:43] debug:    zh:zstack:unpi:writer: --> frame [254,0,33,1,32]
[2025-03-29 21:25:49] debug:    zh:zstack:znp: --> SREQ: SYS - ping - {"capabilities":1}
[2025-03-29 21:25:49] debug:    zh:zstack:unpi:writer: --> frame [254,0,33,1,32]
[2025-03-29 21:25:55] error:    z2m: Error while starting zigbee-herdsman
[2025-03-29 21:25:55] error:    z2m: Failed to start zigbee-herdsman
[2025-03-29 21:25:55] error:    z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-03-29 21:25:55] error:    z2m: Exiting...
[2025-03-29 21:25:55] error:    z2m: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.start (/opt/zigbee2mqtt/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:113:27)
    at Controller.start (/opt/zigbee2mqtt/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/controller/controller.ts:136:29)
    at Zigbee.start (/opt/zigbee2mqtt/lib/zigbee.ts:69:27)
    at Controller.start (/opt/zigbee2mqtt/lib/controller.ts:142:13)
    at start (/opt/zigbee2mqtt/index.js:161:5)
```

kthamm, 2025-03-29:
My assumption is that due to the switch by ITEAD from the CH9102F to the CP2102N chip on the ZBDongle-E, the adapter-discovery by zigbee-herdsman detects the adapter as ZBDongle-**P** instead of **E**.

Nerivec, 2025-03-30:
Yes the Sonoff dongles have so many iterations, it's very hard to match them properly. The serial metadata is already slim to begin with, but between errors in config from Sonoff (some E have the metadata of the P...), and conflicts between versions, it's almost impossible to detect all variants properly...

closed 2025-03-31 by Koenkk

kthamm, 2025-03-31:
Thanks for the quick reply @Nerivec. I've tested changing the regex in my zigbee-herdsman implementation to the one of the PR an can confirm it's working with my stick. Is there any timeline when this fix could be available in a Zigbee2Mqtt release?

Nerivec, 2025-03-31:
New release every 1st, so... tomorrow 😉
_Note that if you know the adapter/path, best to put it in the config directly, it avoids having to run the full discovery on every start. The new onboarding in next release will do just that (discover during onboarding, then save details in config)._