Koenkk/zigbee2mqtt #26912: after Daylight Savings Time transition at 4am - Failed to send request with status=NETWORK_DOWN

10der, 2025-03-30:
### What happened?

boot loop

### Zigbee2MQTT version

2.1.3

### Adapter firmware version

7.4.3 [GA]

### Adapter

EmberZNet

### Setup

HASS (proxmox)

### Debug log

```
[2025-03-30 04:17:09] info: 	z2m: Starting zigbee-herdsman (3.2.7)
[2025-03-30 04:17:09] debug: 	z2m: Using zigbee-herdsman with settings: '"{\"network\":{\"panID\":6754,\"extendedPanID\":[221,221,221,221,221,221,221,221],\"channelList\":[11],\"networkKey\":\"HIDDEN\"},\"databasePath\":\"/config/zigbee2mqtt/database.db\",\"databaseBackupPath\":\"/config/zigbee2mqtt/database.db.backup\",\"backupPath\":\"/config/zigbee2mqtt/coordinator_backup.json\",\"serialPort\":{\"path\":\"/dev/ttyACM0\",\"adapter\":\"ember\"},\"adapter\":{\"disableLED\":false,\"transmitPower\":5}}"'
[2025-03-30 04:17:09] debug: 	zh:adapter:discovery: Connected devices: [{"path":"/dev/ttyACM0","manufacturer":"ITEAD","serialNumber":"20230505110743","pnpId":"usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20230505110743-if00","vendorId":"1a86","productId":"55d4"},{"path":"/dev/ttyS0"},{"path":"/dev/ttyS1"},{"path":"/dev/ttyS2"},{"path":"/dev/ttyS3"}]
[2025-03-30 04:17:09] info: 	zh:adapter:discovery: Matched adapter: {"path":"/dev/ttyACM0","manufacturer":"ITEAD","serialNumber":"20230505110743","pnpId":"usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20230505110743-if00","vendorId":"1a86","productId":"55d4"} => ember: 4
[2025-03-30 04:17:10] info: 	zh:ember: Using default stack config.
[2025-03-30 04:17:10] debug: 	zh:controller: Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":"HIDDEN","panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"path":"/dev/ttyACM0","adapter":"ember"},"adapter":{"disableLED":false,"transmitPower":5},"databasePath":"/config/zigbee2mqtt/database.db","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","backupPath":"/config/zigbee2mqtt/coordinator_backup.json"}'
[2025-03-30 04:17:10] info: 	zh:ember: ======== Ember Adapter Starting ========
[2025-03-30 04:17:10] info: 	zh:ember:ezsp: ======== EZSP starting ========
[2025-03-30 04:17:10] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2025-03-30 04:17:10] info: 	zh:ember:uart:ash: RTS/CTS config is off, enabling software flow control.
[2025-03-30 04:17:10] debug: 	zh:ember:uart:ash: Opening serial port with {"path":"/dev/ttyACM0","baudRate":115200,"rtscts":false,"autoOpen":false,"parity":"none","stopBits":1,"xon":true,"xoff":true}
[2025-03-30 04:17:10] info: 	zh:ember:uart:ash: Serial port opened
```

10der, 2025-03-30:
UPD: only start / stop via 

[image]

solve my issue.

Nerivec, 2025-03-30:
Have you tried rebooting HA completely?

10der, 2025-03-30:
> Have you tried rebooting HA completely?
i can `rm -rf /` as well :) sorry.
I'm already said what stop-start by hassio/addon solved my issue.
I'm asking from root cause issue for the future, Sir.

closed 2025-06-07 by github-actions[bot]