Koenkk/zigbee2mqtt #26913: Lost Connection to Device

Daywalker91, 2025-03-30:
### What happened?

Hey,

after a few Days i lost the connection to an Device.

It's the SNZB-04 by Sonoff. My Coordinator is an Sonoff Zigbee 3.0 USB Dongle Plus (ZBDongle-P).

### What did you expect to happen?

I dont know. All is working properly and then are ther no messages more after a few days. And 12 h after the last message, the avability go to offline.

After repairing it works. But in the Log i get this message `info 2025-03-30 10:31:17zh:controller:device: Device '0x00124b002fbad62e' is only compliant to revision 'pre-21' of the ZigBee specification (current revision: 23).`

And when i start an manual interview i get `error 2025-03-30 10:32:12z2m: Request 'zigbee2mqtt/bridge/request/device/interview' failed with error: 'interview of 'Schlafzimmer/Fensterkontakt' (0x00124b002fbad62e) failed: Error: ZCL command 0x00124b002fbad62e/1 ssIasZone.read(["iasCieAddr","zoneState"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false,"sendPolicy":"immediate"}) failed (Data request failed with error: 'MAC transaction expired' (240))'` and `info 2025-03-30 10:32:12z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/interview', payload '{"data":{},"error":"interview of 'Schlafzimmer/Fensterkontakt' (0x00124b002fbad62e) failed: Error: ZCL command 0x00124b002fbad62e/1 ssIasZone.read([\"iasCieAddr\",\"zoneState\"], {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"reservedBits\":0,\"writeUndiv\":false,\"sendPolicy\":\"immediate\"}) failed (Data request failed with error: 'MAC transaction expired' (240))","status":"error","transaction":"1btr8-3"}'`

### How to reproduce it (minimal and precise)

I dont know.

### Zigbee2MQTT version

2.1.3

### Adapter firmware version

20240710

### Adapter

Sonoff Zigbee 3.0 USB Dongle Plus (ZBDongle-P)

### Setup

Plain on K3s via Docker Image on an TuringPI Cluster with 4 TuringPi RK1

Nerivec, 2025-03-30:
> only compliant to revision 'pre-21' of the ZigBee specification

Means the device has a very old firmware (R21 was in 2015). `pre-21` devices have been known to have pairing issues (among others) because they don't conform to ZigBee 3.0 (security mostly).

As for the interview, it's likely the device is sleeping, you'd have to wake it up and keep it awake by pressing the button on it every 1-2sec until the interview is over.

Some door sensors have a tendency to completely drop off the network until they are re-activated (open/close), so things like availability and network map don't work properly with them.
Can you confirm _after the availability has gone offline_ that if you open or close the sensor, the state is updated?
You might also want to try changing the battery, just in case.

Daywalker91, 2025-03-30:
Hey, transparent for the fast replay. I will see if i can change the batterie. An See the next time it ist offline, if the Status change. But i think not because the rule wasnt triggered.

Daywalker91, 2025-03-31:
So, the device ist offline again and it dosen't update the state. I had
tested it.

Tomorow i will change the batterie.

closed 2025-06-07 by github-actions[bot]